MarlinFirmware/Marlin #5497: [Question] Hardcoded Home Offset and PID for second extruder

Kaibob2, 2016-12-13:
Is there a place where i can hardcode the Home Offset and the PID values for the second extruder?
As my printer is quite good calibrated it would be nice to have **all** values hardcoded to the firmware.

thinkyhead, 2017-02-11:
Yes, it should be possible.

Kaibob2, 2017-02-13:
This would be nice if someone could do so :)

thinkyhead, 2017-02-15:
I'll mark it as a task for the 1.1.1 update.

boelle, 2019-02-18:
did this make it in to 1.1.1?

boelle, 2019-06-22:
@thinkyhead Another one that needs to be closed. assuming it made it in to 1.1.1

sjasonsmith, 2020-11-10:
I believe this is now addressed.

Default PIDs can now be specified as a list, with unique values for each hotend.
The offset is defined using the `HOTEND_OFFSET_` constants.

closed 2020-11-10 by sjasonsmith