MarlinFirmware/Marlin #5498: Missing Z-Probe option/type?

yonkiman, 2016-12-13:
My DeltaMaker was designed with a solenoid that is extended to set up a NO switch that closes on contact with the build plate.  So I just need one GPIO output pin to go high to extend the solenoid (and low to release it), and one one GPIO input pin to read the switch state.  But none of the current probe options seem to support that:
- Fix-Mounted Probe doesn't engage a solenoid
- BLTouch uses a servo, not a solenoid
- Z-Servo Probe uses a servo, not a solenoid
- Z-Sled doesn't work for delta printers
- Allen Key Probes - all mechanical extend/retract (clever!)

My best bet seems to be to modify one of the servo routines so it outputs a static high or low instead of a PWM signal, but I suspect I may be missing a simpler solution...

Blue-Marlin, 2016-12-13:
Take the Z-Sled probe and throw out the moves.

yonkiman, 2016-12-14:
That did it - just removing that line and the sanity check line not allowing deltas to use sleds.

Thanks!

nathan-rightscale, 2016-12-29:
I have a similar z-probe on a Prusa style printer. In addition to commenting out the moves I had to add a delay or I ran into a race condition where the endstop would trigger before the solenoid could fully deploy. Drove me nuts trying to figure that one out.

```c
  static void dock_sled(bool stow) {
    #if ENABLED(DEBUG_LEVELING_FEATURE)
      if (DEBUGGING(LEVELING)) {
        SERIAL_ECHOPAIR("dock_sled(", stow);
        SERIAL_CHAR(')');
        SERIAL_EOL;
      }
    #endif

    // Dock sled a bit closer to ensure proper capturing                                                                                                                
    //do_blocking_move_to_x(X_MAX_POS + SLED_DOCKING_OFFSET - ((stow) ? 1 : 0));                                                                                        

    #if PIN_EXISTS(SLED)
      digitalWrite(SLED_PIN, !stow); // switch solenoid                                                                                                                 
      delay(SERVO_DELAY);
    #endif
  }
```
I don't use any servos on my machine so decided to use SERVO_DELAY so it's more configurable.

closed 2017-02-26 by thinkyhead