MarlinFirmware/Marlin #5499: EEPROM checksum mismatch

MoonshineSG, 2016-12-14:
Today, dec 14, after updating to latest codes (I was behind a couple of weeks) i now get error on M501

```
Send: M501
Recv: Error:EEPROM checksum mismatch
Recv: echo:Hardcoded Default Settings Loaded
```

I tried to save, reset, save again etc, nothing works.

Kaibob2, 2016-12-14:
I can confirm. Do you have set `#define EXTRUDERS 2` or any other value than 1?
If so, try setting `#define EXTRUDERS 1` this fixed the whole EEPROM behaviour in my test right now.
It would be nice if you could confirm this.
I suppose this happened somewhere here #5448
Maybe @thinkyhead can take a look at it.

This in configuration.h 
```
// This defines the number of extruders
// :[1, 2, 3, 4]
#define EXTRUDERS 2

// Enable if your E steppers or extruder gear ratios are not identical
//#define DISTINCT_E_FACTORS
```
results in 

[image]

on connection with Pronterface.

This in configuration.h 
```
// This defines the number of extruders
// :[1, 2, 3, 4]
#define EXTRUDERS 1

// Enable if your E steppers or extruder gear ratios are not identical
//#define DISTINCT_E_FACTORS
```
results in 
[image]
on connection with Pronterface.

MoonshineSG, 2016-12-15:
@Kaibob2 yes, the error occurs with 2 extruders and once I change to `#define EXTRUDERS 1` the EEPROM is loaded without error.

Kaibob2, 2016-12-15:
This is what I expected 
@thinkyhead Would you mind taking a look at your EEPROM refinement. Like it is now it's a PITA to print with any other extruder than E0 :(

Kaibob2, 2016-12-15:
It would also be nice to have the ability to hardcode the PID values for each extruder. 
And the home offset would also be a candidate for hardcoding it.

Kaibob2, 2016-12-15:
Until this is fixed i reverted to [configuration_store.cpp V27](https://github.com/MarlinFirmware/Marlin/blob/3391c785c6af1fe7bbcfd9a3eadf4ce3c279b850/Marlin/configuration_store.cpp)

esenapaj, 2016-12-15:
@MoonshineSG @Kaibob2 
I also confirmed when `EXTRUDERS 2` and disabled `DISTINCT_E_FACTORS`.
And this is quick fix.
https://github.com/esenapaj/Marlin/commit/7f7162005b7d377a74cf25aca9f0da16e8500972

Kaibob2, 2016-12-15:
@esenapaj Tested right now. Works perfect. All values are stored and recovered after reset.
Good job!! Please make a PR out of it that it get's merged asap.

esenapaj, 2016-12-16:
@MoonshineSG @Kaibob2
@thinkyhead has prepared a different solution for resolving it.
https://github.com/MarlinFirmware/Marlin/pull/5524 (Report EEPROM data size, not final index)

MoonshineSG, 2016-12-19:
updated and the fix works.
Thanks!

closed 2016-12-19 by MoonshineSG