MarlinFirmware/Marlin #5501: Language files - new system

AnHardt, 2016-12-14:
If i understand it right,
with the new language system proposed in #5441, every time a new symbol is needed the 
'mappers' and 'fonts' have to be regenerated. At least on windows systems this is a very inconvenient
process. Setting it up is a pain. 
At least for German i know - there are several users having their own variant of `language_de.h`. To make the changes convenient for them - omitting the regeneration process - only a small chang is needed. 

For the languages with a relative small amount of different symbols we could make a complete list
 of the needed symbols.
For example 'German' currently doesn't have a 'Q', But a future version may need it. If we don't want to regenerate the files every time we change something, 'German' would need to have the complete US ASCII symbols plus a relative small amount of additions "ÄäÖöÜüß".
```
#define ASCII_20TO126_SYMBOLS               _UxGT(" !\"#$%&'()*+,-./0123456789:;<=>?@ABCDEFGHIJKLMNOPQRSTUVWXYZ[\]^_`abcdefghijklmnopqrstuvwxyz{|}~")
#define THIS_LANGUAGES_SPECIAL_SYMBOLS      _UxGT("ÄäÖöÜüß")
```

I guess US_ASCII is needed in all languages to make the fallback to English possible. So let's define that at a central place, For example in `language_en.h`. That should be always included.
`THIS_LANGUAGES_SPECIAL_SYMBOLS` should be defined in the languages files.

For the languages with a 'unlimited' set of symbols (cn, tw, kanji) `THIS_LANGUAGES_SPECIAL_SYMBOLS` could be omitted - but the 'mappers' and 'fonts' have the to be regenerated much more frequent then, for this languages only.

Languages with Greek or Cyrillic will have a relative big THIS_LANGUAGES_SPECIAL_SYMBOLS-string. Regardless of that the resulting fonts should be smaller then now.

thinkyhead, 2016-12-19:
Q: Is the benefit of a unified UTF-8 system worth the extra effort involved? I suppose it's good to have more control over the mapping and font-generation for languages with many symbols. Otherwise, what else does #5441 bring to the table?

AnHardt, 2016-12-19:
> Q: Is the benefit of a unified UTF-8 system worth the extra effort involved?

No. - But don't ask me - i'm biased - the current system is mine - i'm not from Asia. ;-)

Some arguments:

* '-'  The new system needs more work. Currently it's not usable. 
* '+' With the current system, languages with many symbols are difficult to maintain. Fonts with more than 256 symbols are impossible.
* '+' Adding new languages to the current system seems to be too difficult. If I don't do it - nobody else does it. (If you know what to do - it's simple)
* '-' Having only one font (or better a lot of font stubs) makes it difficult to estimate the space a text needs. Not only the height can differ a lot, but also the width. The new font is not 'fixed width'.
* '-' If the new system has any benefit for char based displays has to be investigated. I don't think so. 
* '-' The new system currently needs lots of memory.  (Lots of font headers. 127 unused symbols in the main font.)
* '+-' Setting up a new language is now a one time task. When its done - its done. You never have to touch it again - except the scripts with too many symbols. 

Don't ask me. I'm biased.
Maybe __I__'m the only one who sees the time it still needs to make it work. Time, __I__'d prefer to spent on other things.

yhfudev, 2017-01-04:
> '-' Having only one font (or better a lot of font stubs) makes it difficult to estimate the space a text needs. Not only the height can differ a lot, but also the width. The new font is not 'fixed width'.

> '-'  The new system currently needs lots of memory. (Lots of font headers. 127 unused symbols in the main font.)

I don't think the size of the rom is not controllable, the translated string is limited and the expanded size is a small ratio of the strings.
Under the new updated version, it supports unicode value 0-255 now, and a lot of languages don't need generate font data now: (the language_data_xx.h of 398 bytes contains no font data)
[image]

And the compiled ROM files listed here:
[image]

Most of language that require extra font data need extra 400-3k bytes space compared to "en" version, even the Chinese only need extra ~10K bytes space (4% of tatal ROM space).

thinkyhead, 2017-02-12:
I think we've moved on from this issue. Are we feeling better about the new system today?

closed 2017-10-31 by thinkyhead