MarlinFirmware/Marlin #5503: G0 on Z without specified Feedrate

martindb, 2016-12-14:
The mpcnc project use marlin as the firmware to run this cnc multitool machine. 
There are users that generate their gcode with estlcam software, and this gcode use G0 without feedrate specified in the first lines of the programs. We where using marlin 1.1.0RC7 without problems.
With RC8, arduino start to reset when execute the G0 over Z. If you put a feedrate on those first moves, all is working.
Does marlin has to limit these movevements as specified in default_max_feedrate for each axis, in particular for Z?

thinkyhead, 2016-12-19:
The reset may be due to the Z axis failing to move, or trying to move too quickly and eating up all the CPU.

The speed of the Z axis _should_ be constrained by the max feedrate (`DEFAULT_MAX_FEEDRATE`) and max acceleration (`DEFAULT_MAX_ACCELERATION`) values for the Z axis, which prevent it from moving or accelerating at excessive rates. But it's possible the acceleration and jerk behave differently enough in RC8 that your settings need to be adjusted.

I suggest you start by cutting the `DEFAULT_ZJERK` value in half (use `M205 Z` or the LCD controller). If that doesn't fix the issue, try setting lower max feedrate (`M203 Z` or the LCD) and max acceleration (`M201 Z` or the LCD) until the problem goes away.

If you find values that work, save them to EEPROM with `M500` and set them as your `DEFAULT_MAX_FEEDRATE` and `DEFAULT_MAX_ACCELERATION` values for Z.

HTH

martindb, 2016-12-19:
Thanks! I've lowered DEFAULT_MAX_FEEDRATE for Z and no more resets with G0 without explicit feedrate.

closed 2017-02-11 by thinkyhead