MarlinFirmware/Marlin #5507: How does ENSURE_SMOOTH_MOVES behave with SLOWDOWN?

cfelicio, 2016-12-15:
Hi All,

I noticed #define SLOWDOWN is enabled by default, and if I enable ENSURE_SMOOTH_MOVES, what happens in a scenario where the buffer is getting empty? Would both work at the same time? Should I disable one option if I enable the other?

Or are we talking about 2 different buffers / issues here?

Cheers! :-)

Blue-Marlin, 2016-12-15:
`ENSURE_SMOOTH_MOVES` is already history in RCBugFix.
Since the planner buffer isn't drained that suddenly as before, `SLOWDOWN` has a realistic chance to work.

If you want to stay with RC8 - both of the options do about the same, but `ENSURE_SMOOTH_MOVES` is a. adjustable and b. a bit more powerful, but therefore less smooth. With `ENSURE_SMOOTH_MOVES` you can adjust the minimal time for all planed moves. With `SLOWDOWN` the moves time gets gradually longer, depending on how empty the buffer is.

Sebastianv650, 2016-12-15:
Additionally to what has already been said, it's save to enable both at the same time.

cfelicio, 2016-12-15:
thanks for clarifying! :)

closed 2016-12-15 by cfelicio