MarlinFirmware/Marlin #5508: G29's parameters not working correctly

yonkiman, 2016-12-15:
I'm able to perform a bed calibration on my delta with a simple G29 and I get a mesh report at the end.  Everything seems to be working perfectly.

But if I try G29 with any of the switches (G29 S0 or G29 S4 Z0.05 or G29 S1), the motors start singing and the carriage starts moving across the Z plane incredibly slowly until I reset the CPU.  Am I perhaps misunderstanding G29 usage?

yonkiman, 2016-12-15:
OK, I have a little more clarity on this.  In the ~2-3 year old version of Marlin I had been using, I would adjust Z offset with "G29 R0.05", "G29 R-0.05", etc.  That doesn't work with Marlin 1.1.0.  The only Z offset adjustment I could find in 1.1.0 was "G29 S4 xxx", which doesn't work - just sends the carriage slowly off into the sunset (that's the bug).

Is there a different command I should be using to adjust Z offset?

yonkiman, 2016-12-15:
Duh, it's M851.  

There's still a bug with M29, though: My delta, with AUTO_BED_LEVELING_BILINEAR enabled and apparently working, responds incorrectly to an "M29 Sx" command.  I believe that command should be ignored if MESH_BED_LEVELING is not enabled.

If it's useful, here are my config files (for some reason you can't attach .h files?)
`/**
 * Marlin 3D Printer Firmware
 * Copyright (C) 2016 MarlinFirmware [https://github.com/MarlinFirmware/Marlin]
 *
 * Based on Sprinter and grbl.
 * Copyright (C) 2011 Camiel Gubbels / Erik van der Zalm
 *
 * This program is free software: you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation, either version 3 of the License, or
 * (at your option) any later version.
 *
 * This program is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this program.  If not, see <http://www.gnu.org/licenses/>.
 *
 */

/**
 * Configuration.h
 *
 * Basic settings such as:
 *
 * - Type of electronics
 * - Type of temperature sensor
 * - Printer geometry
 * - Endstop configuration
 * - LCD controller
 * - Extra features
 *
 * Advanced settings can be found in Configuration_adv.h
 *
 */
#ifndef CONFIGURATION_H
#define CONFIGURATION_H

/**
 *
 *  ***********************************
 *  **  ATTENTION TO ALL DEVELOPERS  **
 *  ***********************************
 *
 * You must increment this version number for every significant change such as,
 * but not limited to: ADD, DELETE RENAME OR REPURPOSE any directive/option.
 *
 * Note: Update also Version.h !
 */
#define CONFIGURATION_H_VERSION 010100

//===========================================================================
//============================= Getting Started =============================
//===========================================================================

/**
 * Here are some standard links for getting your machine calibrated:
 *
 * http://reprap.org/wiki/Calibration
 * http://youtu.be/wAL9d7FgInk
 * http://calculator.josefprusa.cz
 * http://reprap.org/wiki/Triffid_Hunter%27s_Calibration_Guide
 * http://www.thingiverse.com/thing:5573
[... 2587 lines cut ...]
 * echo:i2c-reply: from:99 bytes:5 data:hello
 */

// @section i2cbus

//#define EXPERIMENTAL_I2CBUS
#define I2C_SLAVE_ADDRESS  0 // Set a value from 8 to 127 to act as a slave

/**
 * Add M43 command for pins info and testing
 */
//#define PINS_DEBUGGING

/**
 * Auto-report temperatures with M155 S<seconds>
 */
//#define AUTO_REPORT_TEMPERATURES

/**
 * Include capabilities in M115 output
 */
//#define EXTENDED_CAPABILITIES_REPORT

#endif // CONFIGURATION_ADV_H
`

fiveangle, 2016-12-16:
G29 when ABL enabled is:

[S<rate>]	
Set the XY travel speed between probe points

http://marlinfw.org/docs/gcode/G29.html

fiveangle, 2016-12-16:
If you submit G29 S**n** where **n**=XY_PROBE_SPEED in your Configuration.h, does it work correctly ?  (also, I assumed you didn't mean "M29  - Stop SD write."

yonkiman, 2016-12-16:
Thanks for the pointer to the more complete documentation.  And the laugh: I was setting movement speed to 1mm/min and it was doing exactly what I asked it to do!

So there's no ABL equivalent of MBL's G29 S4?  I was expecting to be able to do an ABL and then do a single "Adjust Z so a piece of paper can just pass under the nozzle" (from the MBL docs) to account for any variation in probe-to-nozzle distance.  Is the assumption with ABL that the probe-to-nozzle distance will be perfect in Configuration.h and never change over time?  Not sure how ABL is supposed to get that right...

Nomenclature-wise, I think it's a little confusing that the choices are Auto BL or Mesh BL, yet ABL_BILINEAR produces a mesh.  Also is there a difference between Manual Bed Leveling and Mesh Bed Leveling (they seem a little conflated in http://marlinfw.org/docs/gcode/G29-mbl.html), and does the MBL acronym apply consistently to one or the other?

> If you submit G29 Sn where n=XY_PROBE_SPEED in your Configuration.h, does it work correctly?
Will try tomorrow.  But ABL seems to work perfectly (at what seems like XY_PROBE_SPEED) when I submit G29 with no arguments, so I suspect G29 Sn where n=XY_PROBE_SPEED will work the same.

yonkiman, 2016-12-16:
Yes, G29 S4000 started a normal ABL.  So there's no apparent issues with the code, just my understanding of the G29 command and exactly how ABL is supposed to be set up...

fiveangle, 2016-12-16:
I suspect once UBL gets fleshed out, code will collapse to UBL with an auto and manual method for inputs. All in good time…

To answer your nozzle height question: review M851

closed 2017-02-11 by thinkyhead