MarlinFirmware/Marlin #5509: Disabling #define ENDSTOPS_ONLY_FOR_HOMING causes that printer wont move after G28

alewir, 2016-12-15:
I'd like to use 2 endstops on each axis for hardware safety sake and so it would have to be respected while executing G0/G1 move commands...

However when I comment out #define ENDSTOPS_ONLY_FOR_HOMING - this causes that printer won't move after G28, since all home endstops are activate and that prevents any further movements.  It seems like a deadlock... S1/S2 parameters in G0/G1 does not seem to have an effect in runtime.

How to use this Marlin feature or using-endstops-not-only-for-homing without self-blocking the device?

Blue-Marlin, 2016-12-15:
ENDSTOPS_ONLY_FOR_HOMING is not an option newer versions of Marlin have.
It now is called ENDSTOPS_ALWAYS_ON_DEFAULT.

Please update your Marlin and retry.

alewir, 2016-12-15:
Ok. Thanks for quick reply.

Blue-Marlin, 2016-12-15:
In 1.0.2-2 the described behaviour could be caused by always on endstops. Check with M119.

alewir, 2016-12-15:
Yes. All connected endstops are in TRIGGERED. (Not connected too of course.)
After homing and M119 I get:

Reporting endstop statusEOLx_min: TRIGGEREDEOLx_max: TRIGGEREDEOLy_min: TRIGGEREDEOLy_max: TRIGGEREDEOLz_min: TRIGGEREDEOLz_max: TRIGGEREDEOLokEOL

closed 2016-12-15 by alewir