MarlinFirmware/Marlin #5510: HOME_RETRACT_MM does not have effect

alewir, 2016-12-15:
Setting any of:
-  #define X_HOME_RETRACT_MM 10
-  #define Y_HOME_RETRACT_MM 10
-  #define Z_HOME_RETRACT_MM 10
does not have any effect. G28 for each axis just finishes immediately after related endstop is activated. I.e. no "double touch" is happening.

Using version: 1.0.2-2

Blue-Marlin, 2016-12-15:
His is fixed in later versions.
Bugs in 1.0.2-2 will not be fixed any more.

thinkyhead, 2016-12-16:
I think those are supposed to be:
```cpp
#define X_HOME_BUMP_MM 10
#define Y_HOME_BUMP_MM 10
#define Z_HOME_BUMP_MM 10
```

alewir, 2016-12-16:
After commenting out:
//#define ENDSTOPS_ONLY_FOR_HOMING // If defined the endstops will only be used for homing
and adding additional 3 endstops, this double bump started to work.

thinkyhead, 2016-12-21:
I see. In the newer code, that option is now called `ENDSTOPS_ALWAYS_ON_DEFAULT`. And it is off by default.

closed 2016-12-21 by thinkyhead