MarlinFirmware/Marlin #5511: 1.1.0 RC8: constexpr error in compile

duisenberg, 2016-12-15:
Compiling 1.1.0 RC8 using Arduino 1.6.11 I have several constexpr errors in compile.  (Marlin_main.cpp)

Changing them to const (what is somewhat more familiar to me) gets the code compiled as expected.

thinkyhead, 2016-12-16:
The `constexpr` keyword is a C++11 feature. As far as we know, Arduino 1.6.11 has no problem with C++11 constructs in the application distribution version. Are you using a Linux system and possibly an older gcc version?

duisenberg, 2016-12-16:
I run the Windows version of 1.6.11, updated from prior version some time ago. It didnt install a new gcc version, so yes, its still c98, avr-gcc 4.8.1. Just learned from their release notes that the update for some reason missed to install avr-gcc4.9.2 that was introduced with 1.6.10. 

But good to see that Marlin still compiles with c98 ;)

nbmorgan, 2017-01-30:
I am having the same issue with Arduino IDE 1.8.1

My temp solution was to change constexpr to BROKEN_constexpr where it errored and then I added #define BROKEN_constexpr to config.h

What other compilers are bing used to build the binary other than the Arduino IDE?

nbmorgan, 2017-01-30:
When I added my second extruder, I encountered the following error as well:

sketch/configuration_store.cpp: In function 'void Config_ResetDefault()':
configuration_store.cpp:605: error: 'static_assert' was not declared in this scope
     );
     ^
exit status 1

This would seem to indicate another compiler miss/match somehow.

thinkyhead, 2017-02-06:
> I am having the same issue with Arduino IDE 1.8.1

We do not support Arduino.org only Arduino.cc.

thinkyhead, 2017-02-06:
Marlin now checks the gcc version and throws an error if it doesn't support C++11.

closed 2017-02-06 by thinkyhead

Blue-Marlin, 2017-02-06:
Arduino.cc and Arduino.org IDE are unified again. Their latest common version is 1.8.1!

nbmorgan, 2017-02-11:
Yes, I just diffed the two and they are identical.  The about says visit Arduino.cc

nbmorgan, 2017-02-11:
Here are the setting on my mac
compiler.cpp.cmd=avr-g++
compiler.cpp.flags=-c -g -Os {compiler.warning_flags} -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -flto

nbmorgan, 2017-02-11:
Ok, I had to delete the directory $HOME/Library/Arduino15

It seems that 1.8.1 and 1.6.2 (I needed for an older project) confuse each other. And it was using the older gcc from 1.6.2 from the 1.8.1 IDE.

Thanks for adding the compiler check.