MarlinFirmware/Marlin #5513: [Question] Auto Bed Leveling - Bilinear

cbitterfield, 2016-12-15:
I have everything configured and working. The G29 adjustment seems to be a one point calculation/correction between the points. 

I ran G29 and it responded with this matrix:
<-      0     1     2     3     4
<- 0 -0.50 -0.24 -0.02 +0.12 +0.22
<- 1 -0.39 -0.20 -0.04 +0.10 +0.17
<- 2 -0.37 -0.17 +0.00 +0.13 +0.17
<- 3 -0.40 -0.19 -0.02 +0.11 +0.14
<- 4 -0.42 -0.20 -0.03 +0.08 +0.13

Why is the left side higher than the right side during the print? 
I read this as the right side is higher than the left by ~600 microns.

Is the correction simply re-aligning the Z0 or does it allow for a bed plane that is tilted ?
During the printing with Debug flags on; I do not see Z being adjusted during the print.

[Debug_leveling_print_log.txt](https://github.com/MarlinFirmware/Marlin/files/654832/Debug_leveling_print_log.txt)

FHeilmann, 2016-12-15:
G29 probes the bed,

The output shows you where (relative to the center where G28 determined Z=0) the probe triggered. This means that "on the right" (note this depends on where your 0,0 is) your bed is .2mm higher than your center, whereas "on the left" it is .5mm lower.

Looking at your output your bed flat but tilted slightly. The left side is lower than the right side. When you're doing bilinear leveling, the printer will move the Z-Axis up and down so the nozzle follows the bed surface.

You can either let the leveling go its course, or lift the left side of your bed up slightly and re-run G29 until you get near-0 values for all probe points.

cbitterfield, 2016-12-16:
I have tried both. I still get high and low spots on the bed for print. The variation appears to be about 250 microns. I used a set of feeler gauges to measure the height. Even if the bed is almost perfect, I get high and low spots. Is there a way I can see the dynamic Z correction to determine the issue. Does it make a difference if I print through the LCD (printer) or use a computer?(MatterControl).