MarlinFirmware/Marlin #5518: Manual bed leveling hang with RCBugfix

alexxy, 2016-12-15:
Hi all!

I'm using manual bed leveling, and i see that printer hang (printing busy processing to console) when i try to do manual bed leveling (latest RCBugfix). However it works fine with RC8

playlet, 2016-12-15:
I can confirm that, it hangs on the 1st point and the encoder is unresponsive. Thanks for posting this problem, I was just about to do it too :)

wookie666, 2016-12-18:
RC7 here. I have just done this and mine works fine. No hanging

playlet, 2016-12-18:
We are not using RC7, title says RCBugFix latest version from 3 days ago :)

Tannoo, 2016-12-19:
I ran into the same issue. I reverted back to RC8 and no issues.

Fixx, 2016-12-26:
I experience the same issue with RCBugfix dated 15/12/2016.

thinkyhead, 2017-04-02:
Any better now? I used MBL recently and found it working.

closed 2017-10-31 by thinkyhead