MarlinFirmware/Marlin #5521: fillament runout sensor not workrign

pmbroth, 2016-12-15:
I have configured the firmware and connected my sensor to the mks gen 1.4.  The sensor has power and is connect to the d4.  I have uncommented out the lines in config.g and the config av .h   When I start a print, and then pull the filament out of the sensor (for testing, I just have the filament though the sensor, not to the extruder)  and it continues to print.  

I What else has to be done?  I have attached my configuration.  
[marlin 8 with custom configs.zip](https://github.com/MarlinFirmware/Marlin/files/655859/marlin.8.with.custom.configs.zip)
Thank you

pmbroth, 2016-12-17:
Anybody have any advice?

Gaben5224, 2016-12-20:
Hi!

So ... You are in trouble! It won't working. In the Marlin RC has a bug. You should to have an ULTIPANEL LCD screen that you will be able to use this feature. But you can fix it, if you uncomment this line in Configuration.h -> // # define ULTIPANEL. After that it will work, theoretically. But just theoretically because you have a makerbase screen. This kind of screen communicate with the motherboard a different way than the others and it is not ready for the ronout sensor.

Sorry! And sorry for my bad English too!

pmbroth, 2016-12-20:
Thank you!   Your english was great!

thinkyhead, 2016-12-24:
I've added more sanity checking in #5589 that will give proper warnings.

closed 2016-12-24 by thinkyhead