MarlinFirmware/Marlin #5523: build 7ca4f16 crashing and other questions

ruggb, 2016-12-15:
RE: 7ca4f16
CoreXY

What is required to be interupt-capable?
I have an Arduino mega2560 and Ramps 1.4.
Endstops are active low.

What is the "MENU_ITEM_CASE_LIGHT"? I am assuming it is a function to turn on LEDs in the printer. Maybe there is a better term for it, like just "printer light" as I am thinking it is the LCD backlight.

BIGGER PROBLEM:
one print completes the skirt and bombs. Another gets to 3 layers and bombs.
Seems to get totally lost sending Z to max and jambing into stops. Build fa6bf12 is working OK. Same print, same slice, only thing changed was build.

Did I miss something or is there an issue??

Tannoo, 2016-12-16:
Case light is for an external printer xase light controlled by the printer's firmware. It utilizes a free pin on the ramps board (normally one of the servo pins).
The MENU_ITEM_CASE_LIGHT is the option to control the case light via the LCD screen.

thinkyhead, 2016-12-23:
> What is required to be interrupt-capable?

Use `PINS_DEBUGGING` and the `M43` command. Marlin will report which pins are interruptible and available.

> one print completes the skirt and bombs. Another gets to 3 layers and bombs.

- If you disable `USE_WATCHDOG` does it still bomb?
- Are you using 32X microstepping on any of your axes?

ruggb, 2016-12-23:
I have watchdog disabled and  have deleted watchdog.h from the folder -- same on previous versions.
I use 32x microstepping. Motors are capable. I guess it is implemented in Slic3r. Works in previous version. I have not changed anything RE microstepping in Marlin.

thinkyhead, 2017-03-04:
Please post a new issue if the problem still exists in the latest `RCBugFix` code.

closed 2017-03-04 by thinkyhead