MarlinFirmware/Marlin #5528: Auto Bed Leveling with nozzle

drackmoon, 2016-12-16:
Hi everyone,

I'm building a Delta printer, and i would like to use the nozzle as Z-prob, to align the bed.

I will have a switch that will be always trigger and i will use the Z-min endstop, i know that i will have to invert Z-min so this can work correctly once the nozzle touch the bed.
If i do M119 i'm sure Z-Min will appear a trigger, so i must invert this end switch. 

I don't know what more do i need to considerate to make this idea to work, i have take a quick look to this new Marlin version and i see a lot of configuration for Auto Bed Leveling, any suggestion what i would have to configure to get this idea to work correctly. 

Thanks,

yonkiman, 2016-12-16:
I just did exactly this with my delta.  Have ABL working, though I haven't figured out the best way to do the final Z-height adjustment (that's what I'm working on today).  You can get them from my fork at https://github.com/yonkiman/MarlinDM, but I'd wait an hour or so I can clean up the files and check in ones that are easier to follow.  Most of my changes are prefaced with 'FMH' in the comments.

yonkiman, 2016-12-16:
OK, here's my current understanding.  Marlin 1.1.0 doesn't have a probe mode for only a nozzle probe switch.  I asked about this in #5498 and Blue-Marlin suggested "Take the Z-Sled probe and throw out the moves."  So that's what I did.

I added a new identifier SOLENOID_PROBE to Configuration.h.  When SOLENOID_PROBE is #defined, two things happen: 

1. The "You cannot use Z_PROBE_SLED with DELTA." error in SanityCheck.h is ignored, and
2. The do_blocking_move_to_x() function is bypassed in Marlin_main.cpp so the printhead won't try to move to the side of the bed to pick up a nonexistent sled.

I kept the code to extend/retract a solenoid during probing (hence the SOLENOID_PROBE name), because that keeps it compatible with the original DeltaMaker hardware.  If you don't have a solenoid you should be able to just ignore it.

yonkiman, 2016-12-16:
Have it working fairly well now.  Here's how I'm doing it: 

1. Send G28 to home and G29 to start an ABL.  
2. After ABL finishes, gradually move the nozzle down with G1 Z10 F400 (then Z5, Z2, etc.) until it's just rubbing against one sheet of paper.  For my printer that was at Z0.75.  This means the printer is thinking Z0 is 0.75mm below where it is in reality.
3. Send M851 Z0.75 to send the Z error to Marlin. 
4. Send M500 to store the Z error in EEPROM.
5. The next ABL should end with Z0 = the bottom of the bed.

Note that the height will not be adjusted until you perform another G29.  If you simply M851 Z0.75 and then start trying to print, your nozzle will be forced against the build plate.

Also read somewhere that that the mesh generated by the ABL is not stored in EEPROM.  If that's still the case you'll have to do this at least once after every reset/power cycle.

adamfilip, 2016-12-17:
I have this on my CoreXY machine. its awesome.. no offset needed.

yonkiman, 2016-12-17:
@adamfilip - do you follow the same procedure I described above?

drackmoon, 2016-12-20:
@yonkiman - I will try your FW version once i get home, is nice to have another person that is trying to do the same thing i'm trying to do ;)  , ty for your reply.
For what i did understand, you have this working but, we will have to run G29 every time we power off the printer correct.
I let you know once i try your firmware version.

yonkiman, 2016-12-20:
It's actually more often than after a power cycle - a G28 Home apparently clears the G29 results, so you basically have to do it before every print.

The good news is I just heard that it's being addressed (has been addressed?) in RCBugFix.  Mesh array should be stored in EEPROM soon.

thinkyhead, 2016-12-20:
`RCBugFix` now has the ability to save `G29` results to EEPROM.

When using the nozzle tip as the probe, the probe type should be selected as `FIX_MOUNTED_PROBE`. Both `X_PROBE_OFFSET_FROM_EXTRUDER` and `Y_PROBE_OFFSET_FROM_EXTRUDER` should be zero. The `Z_PROBE_OFFSET_FROM_EXTRUDER` value (i.e., `M851 Z`) will be positive. On my delta it is `0.2`. For the leveling procedure use `AUTO_BED_LEVELING_BILINEAR`.

yonkiman, 2016-12-20:
That is GREAT news.  One last question, though: while FIX_MOUNTED_PROBE is exactly what most probes of this type need, for backwards compatibility with the original print head, my delta needs to "extend a solenoid" (though in this case it's actually just flip the state of a pin to connect the other end of the nozzle switch to the right voltage).  If this pin is not set, the Z-probe switch input will read 5V regardless of whether it is open or closed.  I'm currently doing it by using the sled probe method and disabling sled movement and removing the SanityCheck that sleds can't be used with deltas.  

Ideally there would be a probe type (perhaps SOLENOID_PROBE) that flips a bit (energizes a solenoid) before every probe measurement and flips it back (retracts the solenoid) after the measurement.  With the x, y, and z offsets set appropriately, it would work with the probe that originally came with my DeltaMaker.  With x=0 and y=0, and z set appropriately, it will work with a fixed nozzle probe that still needs that bit set for the z-probe switch to register.

You can see how I made it work here: https://github.com/yonkiman/MarlinDM/tree/RC/Marlin
Search for SOLENOID_PROBE in Configuration.h, SanityCheck.h, and Marlin_main.cpp (with pin assignment in pins_RAMPS.h).  I'm using SOLENOID_PROBE as a modifier for the Z_PROBE_SLED probe type.  Ideally SOLENOID_PROBE or just SOLENOID would be its own probe type.

yonkiman, 2016-12-20:
> The RCBugFix branch has the ability to save G29 results to EEPROM with M500.
I'm running the latest RCBugFix (`EEPROM_VERSION "V29"`), but it's not saving my ABL data - G28 seems to clear it (or at least where it thinks Z0 is).  Could that be because I'm still using (have to, for now) `Z_PROBE_SLED` instead of `FIX_MOUNTED_PROBE`?

manianac, 2016-12-24:
`G28` disables the bed leveling by default.  If you've already saved your grid to the EEPROM, issuing an `M420 S1` should re-enable it.  To verify issue an `M420 V` and it will state if ABL is enabled and print your leveling grid.

yonkiman, 2016-12-24:
Thanks manianac, that's a huge help.  Is G28 disabling bed leveling a new thing?  I don't believe my old 1.0.x Marlin did that (or maybe it did and I just have a flat enough bed that I didn't notice).

I don't understand why one would want the home command to disable bed leveling by default (shouldn't everything stay as dialed-in as possible unless explicitly cleared?), but as long as there's a work-around I'm OK...  Thanks again.

manianac, 2016-12-24:
It has always disabled leveling,  in the past a G29 was issued after to remeasure.  Now we can just retrieve saved measurements,  but only if we request to.  I think this is the right way to do it,  just replace G29 with M420 S1 in your start script

Joetripp, 2017-01-18:
I think this is the problem I am having - G28 clearing out my G29 data.  I just finished setting up ABL with a servo/switch setup.  I get G29 properly and it reports back the data fine.  I've set up OctoPrint to do a G28 and G29 before a print.  However when I print from OctoPrint it does the G28, G29, then ANOTHER G28 then starts the print.  I can tell it's not using the ABL data.  I don't know where this extra G28 is coming from - reading through OctoPrint docs now.  I'm guessing this is a OctoPrint problem and not Marlin.  I'm running 1.1.0-RC8.

Joetripp, 2017-01-18:
Actually this extra G28 may be coming from Cura.  the .gcode file has a G28 before print.  I'll try commenting that out.  Yep, that was it!!! Hours of looking at this for something so simple.

thinkyhead, 2017-04-02:
If needed you can also add `M420 S1` to your Start GCode, and it will ensure your leveling (if available) is enabled.

closed 2017-04-02 by thinkyhead