MarlinFirmware/Marlin #5529: V1.1.0-RC8 won't compile

ChrisCarragher, 2016-12-16:
I was upgrading to RC8 as I was interested in trying the Bilinear levelling. I download the files modified the config file and got the error below. I narrowed it down to being the line "#define Z_SAFE_HOMING" once enabled the code won't compile.
Any help is much appreciated.
Regards,
Christopher.

`Arduino: 1.6.13 (Windows 10), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

sketch\temperature.cpp: In function 'Temperature::manage_heater() [clone .part.4]':

temperature.cpp:797: error: unable to find a register to spill in class 'POINTER_REGS'

 }

 ^

temperature.cpp:797: error: this is the insn:

(insn 100 97 102 4 (set (reg:SF 109 [ D.2343 ])

        (mem:SF (post_inc:HI (reg:HI 136 [ ivtmp.126 ])) [7 MEM[base: _157, offset: 0B]+0 S4 A8])) sketch\temperature.cpp:684 100 {*movsf}

     (expr_list:REG_INC (reg:HI 136 [ ivtmp.126 ])

        (nil)))

sketch\temperature.cpp:797: confused by earlier errors, bailing out

lto-wrapper: C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-gcc returned 1 exit status

c:/program files (x86)/arduino/hardware/tools/avr/bin/../lib/gcc/avr/4.9.2/../../../../avr/bin/ld.exe: lto-wrapper failed

collect2.exe: error: ld returned 1 exit status

exit status 1
unable to find a register to spill in class 'POINTER_REGS'

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
`

Kaibob2, 2016-12-16:
I read something similar a few days ago. Take a look at #5459

thinkyhead, 2016-12-17:
This is a spurious compiler issue that we can't get any control on. The only option is to go back to earlier versions of Arduino IDE until you find one that compiles successfully. As the codebase changes here, periodically try the latest Marlin with the latest Arduino IDE, and it may have changed enough to compile.

ChrisCarragher, 2016-12-17:
Thank yous both for help I really appreciated it. I went back to Arduino 1.6.8 but then had to upgrade the AVR Boards to 1.6.9.
Now it is working perfectly.
Thanks.

closed 2016-12-17 by ChrisCarragher