MarlinFirmware/Marlin #5531: 1.1.0-RC8 | G28 Z Homing issue | moves 10 mm two times in + direction then stops?

dvseliteus, 2016-12-16:
Just upgraded to Marlin 1.1.0-RC8

Trying to setup autobed leveling with a fixed inductive probe.
When the machine tries to home Z the bed moves away from the probe about 10mm two times and then stops. I can't seem to get it to move towards the probe so it can trigger home?

http://pastebin.com/UZmsgwFx

brainscan, 2016-12-17:
Have you checked the probe is wired correctly? Sounds like it's triggered when it should be open

http://pastebin.com/UZmsgwFx

dvseliteus, 2016-12-17:
@brainscan I'll have to double check on Monday when I am back in the office with the printer.
Because it worked fine in RC7 and the trigger light turns on when close to bed I didn't even think test the trigger status in pronterface. Thanks for the insight! I'll followup with the findings.

wookie666, 2016-12-18:
get it to trigger then type in M119 and see if it says triggered. Then move it away and type that again, should say open.

dvseliteus, 2016-12-19:
@brainscan @wookie666 
Here is what I am getting when running M119 in Pronterface.
**At first connection pre-homing I run M119;**
SENDING:M119
Reporting endstop status
x_min: open
y_min: open
z_min: open

**If I hit home X only;**
SENDING:M119
Reporting endstop status
x_min: TRIGGERED
y_min: open
z_min: TRIGGERED

**If I hit home y only;**
SENDING:M119
Reporting endstop status
x_min: open
y_min: TRIGGERED
z_min: TRIGGERED

**Back to bed center;**
SENDING:M119
Reporting endstop status
x_min: open
y_min: open
z_min: TRIGGERED

Any idea why the Z-Probe is open on first boot but then as soon as the x or y is homed it goes into static trigger? I tried inverting the probe but then it does not trigger at all?

Blue-Marlin, 2016-12-19:
Play with the pull-up resistors..

dvseliteus, 2016-12-19:
@Blue-Marlin I was afraid you'd say that.
// coarse Endstop Settings
//#define ENDSTOPPULLUPS // Comment this out (using // at the start of the line) to disable the endstop pullup resistors

#if DISABLED(ENDSTOPPULLUPS)
  // fine endstop settings: Individual pullups. will be ignored if ENDSTOPPULLUPS is defined
  #define ENDSTOPPULLUP_XMAX
  #define ENDSTOPPULLUP_YMAX
  #define ENDSTOPPULLUP_ZMAX
  #define ENDSTOPPULLUP_XMIN
  #define ENDSTOPPULLUP_YMIN
  //#define ENDSTOPPULLUP_ZMIN
  #define ENDSTOPPULLUP_ZMIN_PROBE
#endif

These always give me trouble. I guess the trick here will be running the M119 command after homing x or y to make sure it is working correctly.

JohnOCFII, 2016-12-19:
> Play with the pull-up resistors. For a inductive probe it ideally should be off.

But if it worked in RC7, should that have changed in RC8?  (Asking because I'm wondering if this is related to my Z-homing issue on a Delta https://github.com/MarlinFirmware/Marlin/issues/5537)

dvseliteus, 2016-12-19:
@JohnOCFII I'm still trying every combo of pull-up resistor comment/uncomment.
So far I have not been able to get the sensor to trigger.

Yes this sensor was working in RC7 version with no issues.

dvseliteus, 2016-12-19:
@Blue-Marlin 
I have tried every combination of pullups and I am still not able to get the trigger to work on RC8.

Here are the combinations I have tried: http://pastebin.com/903iq9JQ

The strange thing is on the current configuration I have;
// coarse Endstop Settings
//#define ENDSTOPPULLUPS // Comment this out (using // at the start of the line) to disable the endstop pullup resistors

#if DISABLED(ENDSTOPPULLUPS)
  // fine endstop settings: Individual pullups. will be ignored if ENDSTOPPULLUPS is defined
  #define ENDSTOPPULLUP_XMAX
  #define ENDSTOPPULLUP_YMAX
  //#define ENDSTOPPULLUP_ZMAX
  #define ENDSTOPPULLUP_XMIN
  #define ENDSTOPPULLUP_YMIN
  //#define ENDSTOPPULLUP_ZMIN
  #define ENDSTOPPULLUP_ZMIN_PROBE
#endif

On first boot I send the M119 command and receive;
RESULTS;
SENDING:M119
Reporting endstop status
x_min: open
y_min: open
z_min: open

If I home either just x only and then enter the M119 command again I get a false trigger on Z probe?
SENDING:M119
Reporting endstop status
x_min: TRIGGERED
y_min: open
z_min: TRIGGERED

Any idea if I am doing something wrong with my pull-up settings?

wookie666, 2016-12-19:
if you load RC7 back up, does it work again?

dvseliteus, 2016-12-19:
Interesting. I just loaded RC7 backup and I am having similar results. Something may be faulty with the probe. I will try another switch and see what's going on here.

wookie666, 2016-12-19:
I was going to say that. Unplug the z switch and test again. If it's still
triggering when it's unplugged it could be a faulty board. If not it could
be the z switch or cabing

dvseliteus, 2016-12-19:
Okay it's definitely a faulty sensor.
I hooked up a mechanical end stop and it works as it should.
Thanks for your help all I appreciate you walking me through this troubleshoot.
I'll close this issue as it's user error.

closed 2016-12-19 by dvseliteus