MarlinFirmware/Marlin #5533: Bed heating problem

sofos, 2016-12-17:
Symptoms:
1. If I start bed preheating without turning the hotend on, after some time the machine reloads. Was the preheating started from the menu or from the file does not matter.
2. If I start the preheating for both the bed and the hotend, everything goes right until the temperature of the hotend reaches the target, and in short time after this moment the machine reloads. Again it does not matter if the preheating was started from the menu or from the file.
Workaround on (2):
Increasing the PID_FUNCTIONAL_RANGE (now it is 20 cels.) for the hotend eliminates the problem in case the preheating was started with the hotend having the temperature at least 20 cels. lower than the target.
Consideration:
It looks like my hotend have relatively low thermal conductivity between the heater and the probe. So, if the PID_FUNCTIONAL_RANGE is about 10 cels., the temperature of the sensor (when it is near the taeget) will be rising (for some time) even when the heater if completely off. At this moment the heater LED on RAMPS stops blinking and shortly after the machine reloads.
So the common features for cases (1) and (2), which lead to the reload are:
a. the bed is preheating
b. the hotend heater is not powered

Will someone suggest which place in the code should I examine? Any other advice will of course be appreciated.

The firmware is Marlin-1.1.0-RC8 (Marlin-1.1.0-RC5 does have this problem too).
MEGA 2560
BOARD_RAMPS_14_EEB (but only one extruder for now)
RepRapDiscount FULL GRAPHIC Smart Controller
The bed has 220AC heater controlled by relay module on SERVO0.

P.S.: Finally after the preheating is done and the machine is in thermostatic mode, the print goes without problem.

Tannoo, 2016-12-19:
"It looks like my hotend have relatively low thermal conductivity between the heater and the probe."

Remove the 'probe', apply some thermal conductive compound (Artic Silver or similar), reinstall. Coding shouldn't be an issue then.

Tannoo, 2016-12-19:
The bed probe has no conductivity to the heater. Do you have anything on the heater? Glass? The glass will transfer the heat. Aluminum -- even better. Wood....not so good.

I have had issues with the 'machine' getting reset due to power issues (mostly shorts). Check your power connections to the heater and from the power supply to the ramps board. Make sure your probe is not shorting to the 12v. That would be bad.

sofos, 2016-12-19:
The bed is an aluminum plate 10mm thick with glass plate on the top and 220v AC silicon heater stuck on the bottom, the probe is integrated into the heater. I do not see any problem here.

I have read about  power issues, but such things would occur at random moment. While my problem take place under rather predictable circumstances - either when I try to preheat the bed without the hotend or when the hotend temperature reaches the target too fast while the bed is being preheated.

Tannoo, 2016-12-19:
I have had the controller reset due to shorts. Also due to a bad 5v regulator on the mega. You can monitor the 5v on the ramps board while it is heating to check for regulator issues.

Also can pull off any and all accessories to the printer and try again.

Tannoo, 2016-12-19:
Is there a consistant temp that the system restarts?

Blue-Marlin, 2016-12-19:
@Tannoo 
Your problems may be caused by 'Arctic Silver' - that's a very good conductor for current.
Better use a non (current) conductive thermal paste or glue.

sofos, 2016-12-19:
A bad 5v - you mean just voltage drop or more complicated fluctuations?

If you mean the hotend temperature when the problem occurs then it is not absolute constant, it is the target temperature, i.e. when I change the target temperature the absolute temperature for the effect changes accordingly.

sofos, 2016-12-19:
For now I do not use 'Arctic Silver'

Tannoo, 2016-12-19:
I have not used any conductive thermal paste on my hotend as it was assembled already when I got the printer. I was mearly offering it as a suggestion for a bad thermal connection. And being that the thermistors are electrically isolated from the system, Artic Silver would work just fine.

Tannoo, 2016-12-19:
@sofos, you have tested multiple times with just heating the hotend and no issues?

Then heating ONLY the bed gives issues?

If this is the case, then I would forget about the hotend for now.

Monitor the 5v on the controller while the bed is heating. Keep monitoring until after the controller resets.

Was there a voltage drop? More than 3v? No drop? Etc.

Blue-Marlin, 2016-12-19:
[image]
Any current conductive thermal paste/glue is crap here. It will short the legs with each other or to the alu bed or to a copper trace on the heatbed - if you are not __very__ careful with it.

Tannoo, 2016-12-19:
I agree, but this reference was for the hotend.

sofos, 2016-12-19:
I never did preheating of only the hotend without the bed, but tested this now for several times and got no issue.
While preheating of only the bed leads to reload, as always. No I monitored the 5v with multimeter and definitely noticed instant voltage drop right at the moment of the reload. Before and after the reload the voltage is 5.01v to 5.02v, while at moment of the reload it drops down to 3.1v to 4v for very short time (hardly noticeable).
Now I am really puzzled. What could be a reason for this?

Tannoo, 2016-12-20:
How is your bed heater driven? MOSFETs can go bad. Is it possible the fets are getting overloaded.
Did you notice the temp the bed got to at the point of the 'reload'? What temp are you trying to get to?

Are you driving the bed using a 5v output?
Is the power supply for the bed having issues and drawing too much from the control signal?

If you overload the regulator on the Arduino, It will shut down until the load is removed. Which is exactly what sounds like is happening. If the Arduino's regulator has been overloaded for too long, it will shorten it's life and will not deliver as much current as it once did.

Try another Arduino board.

sofos, 2016-12-20:
The bed heater is driven by relay module on SERVO0, the relay itself is driven by transistor (the coil is connected to the collector in the module), which base is driven by SERVO0 through 1KOhm resistor.

At the point of the reload the temperature of the bed could be different. The target for the bed is 150 oC.

The relay module is powered with 5v (from SERVO0 pin group).  I monitored the voltage on the module connector. Is it possible that this voltage differs from 5v on Arduino board?

It is difficult for me to imagine how such a schematics could overload SERVO0 in case it works at all.

Is it possible that this voltage drop is not a cause but a consequence of the reload?

Tannoo, 2016-12-20:
Servo outputs are directly from the Arduino board. (On Ramps 1.4, anyway).
Try Servo4's output.

Good question, maybe. What about the 12v supply powering the board itself? Any major drops in that?

lrpirlet, 2016-12-20:
@sofos This looks a lot like what I have when my servo is blocked because t is prevented to deploy or retract.... 
I have a RAMPS 1.4 with a jumper to get the 5V from the arduino board...
Solution is to insure that the servo never gets blocked OR to feed the servo from an external 5V Power Supply...

is your relay a mecanical one? If yes, it MAY draw too much power AND have an adverse effect on the arduino regulator...

Tannoo, 2016-12-20:
He says that relay is driven by a transistor. That transistor may not be playing fair though.

lrpirlet, 2016-12-20:
Well, even if a transistor is driving the relay, if the power surge at switch-on time is too high for the arduino regulator, the net result will be reset...

So, as you say, a short could explain the overload and the rest, but too high an inrush amperage due to the 'relay module' could also explain all (provided the arduino regulator is driving it).

A possible test here would be to use an external 5 volt to power the relay module.

Tannoo, 2016-12-20:
Ah...I missed the part where he says that the 'module' is powered by the servo0 pin.

So, yes.. I can see this overloading the arduino's regulator. I believe the max load is 100mA, correct?

sofos, 2016-12-21:
Can not do tests on the hardware today, will do tomorrow. But meanwhile ... you focus on possible hardware problems connected with the relay module and my question is Why those problems appear only during the preheating stage? Because after the successful preheating the system works for hours, days and nights without failure of any kind. 
@Tannoo , I use Ramps 1.4. What is the difference between Servo4 and Servo0?
@lrpirlet , what do you mean by 'servo gets blocked' what kind of blocking do you imply?
The relay is mechanical, but at the point of the reload it is not supposed to switch - it must be (and is) just ON (i.e. no transient process). Of course right after the reload it switches OFF.

Tannoo, 2016-12-21:
During the preheat....the transistor and the relay are ON all the time. During this time, current is being drawn. If it is enough current too much time, the Arduino regulator will overheat and shut down....hence, the 'reload'.
    Once the bed has gotten to temp, the output will not be ON all the time. It will be on only long enough to keep the temp happy. The current required to keep the bed at temp is no where near what it takes to 'preheat'.
    'Servo gets blocked' means if a servo is commanded to move to a given position and the servo is mechanically bound to where it cannot move that given position, the servo will start to draw the maximum current to try and get to the given position. Depending on the controller or driver, this can cause an over-current condition that may or may not be protected against.

There is no Servo0 on Ramps 1.4.

Servo1 = Arduino Pin 11
Servo2 = Arduino Pin 6
Servo3 = Arduino Pin 5
Servo4 = Arduino Pin 4

The reason of trying a different servo output is to see if the output has gotten weak due to the load put on it.

lrpirlet, 2016-12-21:
@sofos. I just can talk for what I saw and I share my experience... @Tannoo did explain my meaning... FYI, I have NO IDEA how many amps (millliamps in fact :-) ) will overheat the regulator in what time...

If the power supply was to provide, say, 9 volts instead of 12 V, the dissipation in the regulator would be a lot less and more current will be available before to drive the regulator too hot... 

In my case, I had a power supply delivering 13.1 volt... everything was ok, but the servo could be blocked (or rather the travel of the servo could be blocked) for a maximum of 2 secs before reset... 
I lowered the input voltage to 12.2 and now the servo travel can be blocked for 3-4 sec without reset giving me enough time to make an intervention and clear the path...

Please understand that this rambling is valid FOR ME but may be invalid for you... Thermal condition VARY from saison to saison, from how the fan blowing the electronic is installed if it is installed, from the tolerances on the componants of the RAMPS and of the arduino, etc., etc.

Tannoo, 2016-12-22:
Take a look at this:

http://playground.arduino.cc/Main/ArduinoPinCurrentLimitations

lrpirlet, 2016-12-22:
Hi @Tannoo, @sofos 
It seems we are talking parallel here (meaning that we will never met)... 

This link that you give, do talk about the SIGNAL pins... The regulated power available from the MEGA 2560 module is a different matter... 

Please have a read at this article http://www.open-electronics.org/the-power-of-arduino-this-unknown/ 
paragraph titled **THE REGULATOR’S LIMITS** and **POWER SOURCES THAT ARE EXTERNAL TO ARDUINO** did explain what I was faced with...

Tannoo, 2016-12-22:
I am not just trying to talk. I am trying to help you find the issue. If we are just talking. I am done.

sofos, 2016-12-23:
Hi, @Tannoo, @lrpirlet 
I am sorry for the delay. And I am not just talking. At first I considered this problem to be a software one, maybe this caused a kind of 'parallel talk', but now you really changed my mind, thank you :-) To say the truth, I did not pay much attention to the electronics - just put together stuff from Aliexpress and - bingo! - it works.  Now I need to understand it more deeply, but this requires some time plus I became busy at work. But I am really going to solve this issue, please give me some time to work on it. I really appreciate your help.
Meanwhile one more question: I power the system from ATX PSU, so it has 5v output, what is the best way to connect it to the system?

Tannoo, 2016-12-23:
No harm done. I am not upset...it's just that I was not in this for any debate. I am only trying to help.

There are so many different places that a 5vdc supply can be connected, unprotected. But, I have not done this, so I will look into the best spot for that or someone else can chime in.

lrpirlet, 2016-12-23:
@sofos 
If you have a genuine RAMPS 1.4, if you have a working servo from the servo outpost, then you have a jumper between the "5V" and "VCC"... VCC is the 5V generated by the arduino.

See this picture extracted from https://github.com/euphy/polargraph/wiki/Using-the-Polargraph-software-with-RAMPS-motorshield

[image]

Please examine the RAMPS 1.4 printed circuit board. Note that the "5V" connects the central pins of the 4 servos outpost... See http://reprap.org/mediawiki/images/4/49/Ramps_v1.4_shield_bothsides.png

You MAY want to **remove the jumper** and connect the 5V of your ATX PSU to the servo"5V"... (The return is common with the 12v).
**DOUBLE CHECK WHAT YOU ARE DOING, ANY MISTAKE WILL FREE THE MAGIC SMOKE..** 

Last but not least, please excuse the use of that "talk" word... and thanks, I learned that "talk" implies debate... I do share my experience, I am not trying to debate.

sofos, 2016-12-26:
Hi, everybody! And thank you very much :-)
Cause we did it!
External 5V from ATX PSU connected to the servos’ “5V” pin solved the problem.
And before this I found out that
a.	If the system is just powered On and do nothing the input 12v voltage really is 11.4v
b.	Heating  the hotend drops it to 11.1v
c.	Heating the bed does not affect the input voltage 
d.	During the printing process the input voltage really is 10.8v to 11v
e.	If I preheat both the hotend and the bed, and the temperature of the hotend exceeds the target in a way that the controller practically shuts the hotend heater off, the input  voltage increases and the system reloads
So it looks like you were right and the cause of the problem is the overload of the regulator on the Arduino.  And the solution with external 5v supports.
Surprisingly the regulator practically has no reserve. How do people power additional modules?

Tannoo, 2016-12-26:
Yeah.. There are some things that software just can't fix. Lol

lrpirlet, 2016-12-27:
@sofos, nice to hear... Please close this thread if you are happy.

A drop of .4 volt when heating lead me to believe that the size of the wires to feed the bed heater is maybe not appropriate... As a general rule, you should try to keep a DC power line as short as possible and/or have a big section (I use a section of about 2,5 mm²)... I would advice to double your cabling for the bed heater...

A drop of .6 volt lead me to believe that the ATX power supply is on the low powered side... I would recommend you to use an ATX supply that can deliver about 400 watt that will provide enough Amps over the 12V line you use ( you do **NOT** want Multiple +12 V rails for your 12v... see https://en.wikipedia.org/wiki/Power_supply_unit_(computer) for insight)

Last but not least, if you use a beefier ATX PS, make sure to blow a fan on your electronic... You want to cool down the FET that drive the bed heater... (A heat sink over that Field Effect Transistor MAY be a good idea!)

Last but not least enjoy...

lrpirlet, 2016-12-28:
@sofos 

In Short: do NOT try to feed the arduino and the RAMP **logic** with an external 5v... You would gain nothing and you would increase the risk of failure dramatically... You would put at risk the arduino, the USB port of your computer and you would end up with, at best, an unstable setup...

In long:
The fact that your setup did _nearly_ work before and that it work after powering the servo via an external 5v shows that you have now enough power for the arduino AND the display... 
The regulator on the arduino is NOT able to power the bed heater, is not able to power the stepper motors, is _nearly_ able to power the servo... those need _high_ current to work... The display do NOT need high current...
I proposed to power the servo from an external 5v because, by design, the servo allows it AND the RAMPS allows it... This is NOT the case for other low current logic.

BTW, the arduino regulator only power the arduino itself and some logic on the shield... My remark about the ATX power supply was because those _high_ current devices that ARE powered externaly do pull the tension low...

In French I use to say: **le mieux est l'ennemis du bien**, or in English: **if it ain't broke, don't fix it** 

Last but not least... This discussion should have taken place in the REPRAP forum, it is not a Marlin issue at all...

sofos, 2016-12-28:
@lrpirlet , Thank you for your consideration and advice.

closed 2016-12-28 by sofos