MarlinFirmware/Marlin #5537: RC8 Migration question - Delta Z endstop seems to be ignored

JohnOCFII, 2016-12-18:
This is likely a configuration oversight, but I've reviewed my changes a couple of times and can't see it.

I'm migrating my Delta from RC3 to RC8.  After uploading, the first test was to issue G28 to home all axis.  All carriages move upwards, as they should.  X and Y stop once, back off, then slowly approach the end stop and stop.  Z appears to ignore the endstop and tries to keep going until I hit the software "Emergency Stop" button.

Suggestions?  Should I move beyond RC8 to RCBugFix to check?

My compressed c
[Configuration.h.zip](https://github.com/MarlinFirmware/Marlin/files/659255/Configuration.h.zip)
onfiguration.h is attached

Thanks,

John

JohnOCFII, 2016-12-19:
It must be something different with the configuration.  I'll have to look in more detail. 

-- This is with Z-axis tripped on RC8
```
SENDING:M119
Reporting endstop status
x_max: open
y_max: open
z_min: TRIGGERED
z_probe: TRIGGERED
```

-- This is with Z-axis tripped on RC7
```
>>>M119
SENDING:M119
Reporting endstop status
x_max: open
y_max: open
z_min: TRIGGERED
z_max: TRIGGERED
```

jwcrawley, 2016-12-20:
I'm seeing something similar. I have an issue, in which I am noticing on a Delta, the Z motor tower snaps to home, but does not do the slow away, slow approach. It just SNAPS. 

I have a video of my machine showing the erroneous behavior at the bottom https://github.com/MarlinFirmware/Marlin/issues/5467

Blue-Marlin, 2016-12-20:
As long as z-min is triggered the z-carriage will not move down. A delta should not have a z-min-endstop.
It may have a z-min-probe. The difference is - the probe can be switched off when not needed.

jwcrawley, 2016-12-20:
I think I know what's happening. Since I do NOT have a Zmin sensor (inductance, Bltouch, etc) I manually calibrate.  So, I went and commented out 

#define USE_ZMIN_PLUG

And my homing now works correctly.

JohnOCFII, 2016-12-20:
Thanks @jwcrawley I'll check this later tonight.  Right now I use a manually deployed "allen key" for auto bed leveling.  I have purchased (and received) a BLtouch, but was trying to do a like-for-like RC7- RC8 transition first.

jwcrawley, 2016-12-20:
Gladly, @JohnOCFII . I would also look into possibly inverting Z_MIN and see if that solves your issue. Since I just don't have a sensor, no reason for that chunk of code.

If your endstops are reporting z_min:TRIGGERED when just hanging around doing nothing, it's a good sign to invert that endstop in software.

JohnOCFII, 2016-12-20:
I tried commenting in and out various options.  What I don't understand is that z_max: never shows up in the list.  Is it now called z_probe?

jwcrawley, 2016-12-20:
Can you post your configuration.h here? (copy it to desktop, rename to .txt, and drag/drop here)

JohnOCFII, 2016-12-20:
@jwcrawley It was in the first note above.  I have made some minor changes in my recent troubleshooting attempts, so here is a fresh copy:  
[Configuration.h.zip](https://github.com/MarlinFirmware/Marlin/files/664600/Configuration.h.zip)

jwcrawley, 2016-12-20:
I'm unsure of the ramifications of having a # before // , with regards to parsing.  I'm noticing on line 604, you have

`#//define FIX_MOUNTED_PROBE - commented out 12/19/2016 for testing (was commented out in RC7)`

_______________________________

I also notice that #define Z_PROBE_ALLEN_KEY is enabled, thus converting  z_min -> z_probe . However, it states on Line 484 the following:

// Specify here all the endstop connectors that are connected to any endstop or probe.
// Almost all printers will be using one per axis. Probes will use one or more of the
// extra connectors. Leave undefined any used for non-endstop and non-probe purposes.

So if it's an endstop or a probe, you have to have it listed. So, what you want is:

```
//#define USE_XMIN_PLUG
//#define USE_YMIN_PLUG
#define USE_ZMIN_PLUG // a Z probe
#define USE_XMAX_PLUG
#define USE_YMAX_PLUG
#define USE_ZMAX_PLUG
```

Now, you need to plug in USB to your printer and load up your appropriate interface (Pronterface, Repetier, etc). Now you need to manually move the effector around to manually trigger endstops while sending ["M119"](http://reprap.org/wiki/G-code#M119:_Get_Endstop_Status). Hold the Efffector away from any endstop and do M119. Do you see any triggered? It **shouldn't** be. Now, check the top switches. They will show as "TRIGGERED".

I'm guessing that you have an inverted issue with the endstop for z_min/z_probe. After looking at Line 509, I see the following

```
#define X_MIN_ENDSTOP_INVERTING false  // set to true to invert the logic of the endstop.
#define Y_MIN_ENDSTOP_INVERTING false  // set to true to invert the logic of the endstop.
#define Z_MIN_ENDSTOP_INVERTING true  // set to true to invert the logic of the endstop.
#define X_MAX_ENDSTOP_INVERTING false  // set to true to invert the logic of the endstop.
#define Y_MAX_ENDSTOP_INVERTING false  // set to true to invert the logic of the endstop.
#define Z_MAX_ENDSTOP_INVERTING false  // set to true to invert the logic of the endstop.
#define Z_MIN_PROBE_ENDSTOP_INVERTING false  // set to true to invert the logic of the endstop.
```

I would wager that the problem is, if you were running a z_min endstop switch, then INVERTING=TRUE , but you note z_probe.  So, I have a 'feeling' your issue here is the following needs to be changed:

`
#define Z_MIN_PROBE_ENDSTOP_INVERTING false // CHANGEME to true`

jwcrawley, 2016-12-20:
With suggested edits:  
[Configuration.h.zip](https://github.com/MarlinFirmware/Marlin/files/664648/Configuration.h.zip)

JohnOCFII, 2016-12-20:
Thanks, Josh.
It'll be a few hours before I can test this, but I'm looking forward to success.

So after removing my typo, and one of the troubleshooting comments, I'm left with the two lines that have changed between RC7 and RC8 regarding endstops:

new (RC8)
#define Z_MIN_ENDSTOP_INVERTING false  // set to true to invert the logic of the endstop.
#define Z_MIN_PROBE_ENDSTOP_INVERTING true  // set to true to invert the logic of the endstop.

old (RC7)
#define Z_MIN_ENDSTOP_INVERTING true  // set to true to invert the logic of the endstop.
#define Z_MIN_PROBE_ENDSTOP_INVERTING false  // set to true to invert the logic of the endstop.

Assuming this works, I'll be back to wonder why the behavior seems to have changed since RC7.

John

jwcrawley, 2016-12-20:
It looks like they separated out the idea of actual min-max limit switches and a bed leveler using Z_min. That would be my semi-informed guess, having not looked studiously at the code.

JohnOCFII, 2016-12-20:
Sadly - no change in behavior on the Z-axis:

Results after fresh firmware load:

SENDING:M119
Reporting endstop status
x_max: open
y_max: open
z_min: open
z_probe: TRIGGERED

Results with Z-axis endstop manually triggered

SENDING:M119
Reporting endstop status
x_max: open
y_max: open
z_min: open
z_probe: open

Again - I'm surprised to see no listing for x_max.  Still wondering if the endstop probe logic for X-min and X-max is somehow backwards in RC8.

Thanks,

John

JohnOCFII, 2016-12-20:
I figured it out!

Here was the culprit: 

#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN

This was failing when it was commented out.  It was active in RC7 - I'm not sure why I missed it in my configuration review.  

Thanks for your help, @jwcrawley!

closed 2016-12-20 by JohnOCFII