MarlinFirmware/Marlin #5539: more like a request

locutus40, 2016-12-18:
not sure where to put this.. but ill ask anyway

for the next rc i know its 8 now or even the bug fix..

can someone add a pause feature and a resume for the lcd.. and or for repiter host type it in manually etc..

see im using a laser with the fan output... its a on off for now .. anyway i want to be able to pause and shut off the fan output at the same time then resume and turn it on.. or turn it off for said amount of minutes.. to keep the diode cool

even for 3d printing a pause function and resume where it left off.. would be nice...

as it stands you cant stop it mid print ....

i have a 30 minute window with the laser i cant print longer or it gets to hot..

anyway both lasers and printing will benefit from this.. thanks...

Roxy-3D, 2016-12-18:
Perhaps it would be good to shift the discussion to what would be needed for a general purpose command.   For a laser, it might be good to be able to specify an MCode command in the Slicer's startup Gcode that says "You can print for 5 minutes, but then you need to pause and cool down the laser for 10 minutes"    But to extend this type of a pause command to printers, you would also need to specify retraction and priming of the filament.

Blue-Marlin, 2016-12-18:
How about using/extending M=/M1 for your purpose?

locutus40, 2016-12-18:
marlin what is m0 and m1?   see i also need to stop the fan.. my laser is connected on d9 with a dc to dc step down so if all the axix stopped the laser is still firing in one spot would m0 and m1 stop that?

thats my main thing is over heating and shutting off the laser i could issue 2 diffrent commands or combine then in one command line like m0 m107???? is that possible

other wise it takes me 2 seconds if not 3 by that time its burning the place where it stopped..

now i see other machines on the market that use a lcd one lcd that isnt these 2004's or the reprap discount one.. and they have a pause button when pressed it stops moves away from the print so it dont ooze on your part and then when you hit resume it goes right back to printing..

if its possible with a arduino and ramps and some funky lcd i never seen before it should be inplemented into our lcds or even marlin no one not other front ends that send the gcode have a pause and resume button i know i looked..

thinkyhead, 2016-12-20:
> marlin what is m0 and m1?

May I introduce you to http://reprap.org/wiki/GCode …?

> i want to be able to pause and shut off the fan output at the same time then resume and turn it on.. or turn it off for said amount of minutes.. to keep the diode cool

A G-Code based solution exists. Just insert this into your GCode file where you want to pause:

```gcode
M5 ; Laser off
M107 ; fan off
G4 S2 ; Wait 2s
M0 Click when ready
M106 ; fan on
G4 S2 ; Wait 2s
M3 ; Laser on
```

> a pause button when pressed it stops moves away from the print

We have received this feature request many times. Also, the ability to stop a print and resume hours later. For example, #3930, #4131, #4627, #4762, #5266. So these are on the to-do list.

thinkyhead, 2017-04-02:
> a pause button when pressed it stops moves away from the print

We now have this. `PARK_HEAD_ON_PAUSE`. It requires also enabling `M600`.

closed 2017-04-02 by thinkyhead