MarlinFirmware/Marlin #5540: Quick home Core xy grinding

wookie666, 2016-12-18:
Enabling quick home in RC7 on a COREXY machine. The diagonal move works, But when it hits the x stop, it grinds for a bit before stopping x movements and then moves to the y stop.

thinkyhead, 2016-12-20:
Of course, how does it behave in `RCBugFix`? is the real question.

brendanjerwin, 2016-12-24:
Core XY, RAMPS 1.4

I had a similar experience with the RC8 (and then the RCBugFix) branches. I never let it grind long enough to see if it would move on to the other axis. Double, triple, checked that the end stops were behaving correctly with M119. This happened with all homing routines. Even single axis.

thinkyhead, 2017-04-02:
And where do we stand now? I know we've done a bunch of stuff to the Core code since Xmas.

molak34, 2017-06-10:
I'm having troubles with quick home on a smartcore XY ..

When the X is closer to home than Y it works OK. 

But if Y is closer to home than X  it grinds ...

Some how the firmware moves in diagonal and listen to the X endstop only.

molak34, 2017-06-10:
https://www.youtube.com/watch?v=rgs83ao3UdU

closed 2018-02-06 by thinkyhead