MarlinFirmware/Marlin #5541: RC7 Change filament doesn't unload or load filament. Only purges

wookie666, 2016-12-18:
I have set all settings in marlin to the following

// Add support for experimental filament exchange support M600; requires display
#if ENABLED(ULTIPANEL)
  #define FILAMENT_CHANGE_FEATURE             // Enable filament exchange menu and M600 g-code (used for runout sensor too)
  #if ENABLED(FILAMENT_CHANGE_FEATURE)
    #define FILAMENT_CHANGE_X_POS 3             // X position of hotend
    #define FILAMENT_CHANGE_Y_POS 3             // Y position of hotend
    #define FILAMENT_CHANGE_Z_ADD 10            // Z addition of hotend (lift)
    #define FILAMENT_CHANGE_XY_FEEDRATE 100     // X and Y axes feedrate in mm/s (also used for delta printers Z axis)
    #define FILAMENT_CHANGE_Z_FEEDRATE 20        // Z axis feedrate in mm/s (not used for delta printers)
    #define FILAMENT_CHANGE_RETRACT_LENGTH 2    // Initial retract in mm
                                                // It is a short retract used immediately after print interrupt before move to filament exchange position
    #define FILAMENT_CHANGE_RETRACT_FEEDRATE 60 // Initial retract feedrate in mm/s
    #define FILAMENT_CHANGE_UNLOAD_LENGTH 650   // Unload filament length from hotend in mm
                                                // Longer length for bowden printers to unload filament from whole bowden tube,
                                                // shorter lenght for printers without bowden to unload filament from extruder only,
                                                // 0 to disable unloading for manual unloading
    #define FILAMENT_CHANGE_UNLOAD_FEEDRATE 60  // Unload filament feedrate in mm/s - filament unloading can be fast
    #define FILAMENT_CHANGE_LOAD_LENGTH 650       // Load filament length over hotend in mm
                                                // Longer length for bowden printers to fast load filament into whole bowden tube over the hotend,
                                                // Short or zero length for printers without bowden where loading is not used
    #define FILAMENT_CHANGE_LOAD_FEEDRATE 60    // Load filament feedrate in mm/s - filament loading into the bowden tube can be fast
    #define FILAMENT_CHANGE_EXTRUDE_LENGTH 50   // Extrude filament length in mm after filament is load over the hotend,
                                                // 0 to disable for manual extrusion
                                                // Filament can be extruded repeatedly from the filament exchange menu to fill the hotend,
                                                // or until outcoming filament color is not clear for filament color change
    #define FILAMENT_CHANGE_EXTRUDE_FEEDRATE 3  // Extrude filament feedrate in mm/s - must be slower than load feedrate
  #endif
#endif

But when i use the LCD and go to change filament, it lowers z, moves x and y, but doesn't unload. It just purges the 50mm of filament. Doesn't give me options to unload or load either so i have to do it manually. Could this be because I am trying to unload 650mm when I am only about 1 layer into a print?

wookie666, 2016-12-18:
So I have worked out that if the extruder has not moved more than the required unload/load distance, it will bypass this, Can you please make it so the extruder can go into negative numbers?

thinkyhead, 2016-12-20:
We're not fixing "RC7" any further… and are now working on "RC9." Can you test whether this problem exists in the `RCBugFix` branch? We may have fixed it already.

Bubolz, 2017-02-21:
I'm having the same problem, I tested in RC8  and last RCBugFix version and the problem remains.

closed 2017-10-31 by thinkyhead