MarlinFirmware/Marlin #5542: full graphic lcd display problem

webnamex, 2016-12-18:
Please help me, how to solve the problem in the picture

[image]
[image]

webnamex, 2016-12-18:
[image]

Tannoo, 2016-12-18:
Configuration or hardware issue? I have the same display and have run all different versions of Marlin. I have never had that issue.

Can you post your configuration files?

webnamex, 2016-12-18:
[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/659480/Configuration.txt)

webnamex, 2016-12-18:
@Tannoo

webnamex, 2016-12-18:
thanks @Tannoo

Tannoo, 2016-12-18:
ok. Try upgrading to RC8. A LOT of things have changed. See if that fixes it.

Tannoo, 2016-12-18:
@webnamex, did that help?

snoekieboe, 2016-12-22:
I actually have the exact same issue with my ramps 1.4 board and 12864LCD when running marlin.

willthiswork89, 2016-12-23:
im having this issue as well same board. osoyoo ramps 1.4 board with 12864LCD. Starts out fine, then seems like randomly the screen gets pushed on the X by 20 pixels or so and starts "looping" around just as the OP shows.

willthiswork89, 2016-12-23:
I'm on RC8

webnamex, 2016-12-23:
hi，I've fix it ，I just only  replace the lcd module ,not the lcd whole board,now it  works well ,you can try @willthiswork89 @snoekieboe

webnamex, 2016-12-23:
@Tannoo  thanks

willthiswork89, 2016-12-23:
Are you saying my LCD screen is bad? Or is there something in the code to fix this?

webnamex, 2016-12-23:
@willthiswork89 yes，just replace the lcd screen

Blue-Marlin, 2016-12-23:
Has one ore more of you defined
```
  // Enable this option and reduce the value to optimize screen updates.
  // The normal delay is 10µs. Use the lowest value that still gives a reliable display.
  //#define DOGM_SPI_DELAY_US 5
```
?

willthiswork89, 2016-12-23:
I uncommented that but its still having the issue. I'm not convinced its a bad LCD since it works fine until i do something. I have also found that is i adjust the "LCD_WIDTH" i can "simulate" the issue so im wondering if somewhere the lcd_width is being changed

WrQth, 2017-01-04:
I am having the same issues as well running RC8 & bugfix and have defined #define DOGM_SPI_DELAY_US 5. With value at 5 it would be fine at info screen if I didn't do anything but would glitch if it was printing after a few minutes or if I was moving around menus randomly. Reduced value to 3 and screen glitched immediately even at power-on. Changed value to 6 and seems to be stable while not print and doing nothing and stable when navigating menu's. It's late so haven't checked it while printing but will and provide an update.

Blue-Marlin, 2017-01-04:
@WrQth 
```
  // Enable this option and reduce the value to optimize screen updates.
  // The normal delay is 10µs. Use the lowest value that still gives a reliable display.
  //#define DOGM_SPI_DELAY_US 5
```
Does this description need improvement?
If you have problems enlarge the time or let it commented.

yhfudev, 2017-01-04:
I think there's some kind of electromagnetic interference to your LCD cable, or bad contact.

WrQth, 2017-01-04:
@Blue-Marlin I had started with it commented but upon finding this thread I un-commented it with success after changing the value to 6. Perhaps a line describing a common symptom changing this setting resolves such as reduces screen artifacts by setting a shorter delay, also include the negative impact if any of changing the setting to far. By lowest value does it mean a smaller number resulting in a shorter delay or a larger number resulting in a longer delay?

@yhfudev I will need to investigate this more, perhaps shield the other wires/cables that follow the same path or reroute the cables, but the reprap discount smart controller had no issues with the same cable and installed location.

willthiswork89, 2017-01-04:
I have verified that replacing the controller fixed the issue for me with no additional modifications. I tried all the debugging options mentioned here as well as going in and digging into the code and nothing worked. Brand new screen showed up, plugged it into same exact wires and expansion board and it worked fine. I cannot imagine that it has so many failures but... who knows.

WrQth, 2017-01-05:
@willthiswork89 Looks like I will need to replace mine as well, further testing and adjustments and it still glitches.

webnamex, 2017-01-05:
since i replace  the lcd screen, The problem is solved.

thinkyhead, 2017-02-12:
For future reference, you can also try experimenting with these delays:
```cpp
// If you want you can define your own set of delays in Configuration.h
//#define ST7920_DELAY_1 DELAY_0_NOP
//#define ST7920_DELAY_2 DELAY_0_NOP
//#define ST7920_DELAY_3 DELAY_0_NOP
```

closed 2017-10-31 by thinkyhead

rmnhsd, 2020-09-20:
Where is DELAY_0_NOP defined? I get:

'DELAY_0_NOP' was not declared in this scope

AnHardt, 2020-09-20:
Don't use ancient tips for ancient code.

Use `DELAY_NS( x )` now.