MarlinFirmware/Marlin #5543: E3D PT100

bsculley, 2016-12-18:
I am attempting to interface PT100 with mini-Rambo.  As you know there are no AUX analog pins available on mini-Rambo, so I need to use the original T0 pins.  I do not want to remove the 4k7 pullup on T0 for fear of destroying the board.

I understand that sensor type 147 was added to the firmware to allow direct connection of PT100 to the T0 pins (without amplifier).  When I tried this I got readings of -10 at room temperature.  Is there some sort of additional calibration or adjustment required?

Also, just wondering, is there any method of connecting the amplifier output to the original T0 pins that will work?

Grogyan, 2016-12-18:
The E3D PT100 uses temp_sensor 20 and wiring up to an analog input

manianac, 2016-12-18:
Its not worth trying to measure a resistance difference between 4.7k and ~180ohms (or whatever temp you're printing at).  Use the e3d amp and pick any analog input as @Grogyan stated and you'll be much happier.

bsculley, 2016-12-18:
Thanks a lot for your input here, I'll take the advice.  
The pin numbers are a bit mysterious to me.  The default pin for TEMP_0_PIN is 0 (indicating analog input 0, I'm assuming).  In order to use the XMAX pin for the PT100, would the pin assignment be: 

#define TEMP_0_PIN 30 (same as X_MAX_PIN)
or
#define TEMP_0_PIN 15 (A15)

By the way, just for the record, I replaced the calculated temperature table for sensor type 147 with a manually adjusted one and got usable (although probably inaccurate) results.  I have include the table below.  If anyone has any way of verifying (or improving) this I think it would be appreciated by everyone.

const short temptable_147[][2] PROGMEM = {
// only few values are needed as the curve is very flat  
{292,0},
{356,50},
{420,100},
{484,150},
{548,200},
{612,250},
{660,300},
{724,350},
{772,400},
{820,450},
{884,500},
{932,550},
};

thinkyhead, 2016-12-20:
> Is pin 30 really an analog input?

Analog pins only go from 0-15.
On Mega boards, the corresponding digital pin numbers are 54-69.

Bob-the-Kuhn, 2016-12-20:
Almost all the pins on the 2560 have alternate functions.  In the cases of ports F & K , they have the standard digital I/O cell attached to the physical pin plus they also have an analog input cell attached to the physical pin.  They are completely independent of each other.  You could set the pin as a digital out and do an analog read of the pin.  You wouldn't get much useable data from the analog read but it can be done and done without damaging anything.  

I agree.  The "simple" solution is to use an analog input without that pesky pullup.

I'm also scratching my head over using the limit inputs as an analog input.  The schematic I found shows that none of the limit pins go to an analog input.  In fact it shows that the only analog capable pins that go to a connector/header are the three thermistor inputs, all of which have the 4.7k pullup.  Sure would be nice if we were looking at the wrong revision of the schematic.  Just to be sure I've attached the schematic I found.

I too ran into troubles using an E3D PT100 amplifier with my Azteeg X3Ppro.  The amplifier just doesn't have the muscle to drive that 4.7K pullup on the thermistor inputs.

The custom thermistor table is probably a reasonable solution.  My unit was reading about 25C high at room temperature.  As I look at the effects of the amplifier's considerable output impedance it looks like the error actually gets smaller as the temperature increases (the current the amp has to pull from the 4.7K gets smaller as the temperature [and the voltage from the amp] increases). At 400C it looks to be about 2/3 of the error at room temperature.

If the amplifier's output impedance is the issue then the controller will be reporting a higher than actual reading.  
[Mini-Rambo.PDF](https://github.com/MarlinFirmware/Marlin/files/662647/Mini-Rambo.PDF)

bsculley, 2016-12-20:
I don’t believe pin 30 (XMAX) has any analog capabilities.  The documentation is pretty clear that there are 16 analog inputs and this is not one of them.

 

Did you come to any solution with your issue?  I have decided that the only possible solutions are either to physically remove the pull-up, or to wire up one of the unused analog inputs.  Both of these require altering the SMD’s in some way and I’m not happy about that, but I don’t see an alternative.

Grogyan, 2016-12-20:
X-MAX, Y-MAX and Z-MAX are indeed all analog inputs that can be easily
connected to on your board.
The PT100 must be connected to  an analog input without the additional
external pull ups.

manianac, 2016-12-20:
@Grogyan I'm not sure thats correct.  As @Bob-the-Kuhn stated, only ports F & K are ADC.  Port A is just a generic port that can double as external memory (page 75 @ [2560 datasheet](http://www.atmel.com/Images/Atmel-2549-8-bit-AVR-Microcontroller-ATmega640-1280-1281-2560-2561_datasheet.pdf)).

Grogyan, 2016-12-20:
@manianac Whoops, my mistake, you are correct.

@bsculley You will need to remove a pullup resistor with a soldering iron
Or
Change to another board, RAMPS or RAMBo
Or
Use a thermistor (yuck)

madhunm, 2016-12-22:
@bsculley

How did you generate the new table? I could not find the datasheet for the e3d PT100 sensor.

Bob-the-Kuhn, 2016-12-22:
Here's the [link](http://wiki.e3d-online.com/wiki/E3D_PT100_Amplifier_Documentation) that I used.

It's a table of temperature vs. output voltage of the amplifier assuming a 5V power supply.  When you convert it into analog in counts it's at most one count away from the Marlin table.

I take it you're going to be creating a custom table for your system.  An easy way is to just add the error count you see at room temperature to all the values in the Marlin table.  If  you want more accuracy then:
-  use a DVM to measure the voltage across the two input pins of the amp 
-  measure the output of the amp while it's connected to your board
-  do this for room temp and ABS temp and maybe one point in between.  It doesn't matter that the temperature isn't known exactly, all your after is the error factor vs. voltage at the input
-  The output should be 10x (+/- about 0.2%) of the input.  The output voltage should always be higher than the 10x gain would imply.
- now that you have three points you can calculate the equation for the compensation.

madhunm, 2016-12-22:
@Bob-the-Kuhn 

The problem is this: 

When I use the amp and set the temp_sensor_0 to be 20; I read 40 degrees when the ambient is 20 degrees.
When I use the PT100 directly, and I set temp_sensor_0 to be 147, I read 0 degrees when the ambient is 20 degrees.

I understand that if I desolder the 4k7 pullup and select temo_sensor_0 to be 20 and use the amp; my problems will go away. However, I am trying to do this without that.

Bob-the-Kuhn, 2016-12-22:
I don't know how good is good enough when it comes to knowing the temperature of the extruder.  I do know that I've seen 10-20C ranges given for working with ABS and that all the calibration guides I've read say that the sweet spot for the temperature will vary from machine to machine and from filament batch to filament batch.

From that I conclude that you use the temperature reading to get in the neighborhood and after that it's adjusted based on the results of the calibration.

I can't offer any first hand experience as I have not yet had a successful calibration run. I just started my very first calibration run when the Z_MIN sensor failed which caused my extruder to be driven into the hot bed which produced a bent heat break which broke when I tried to straighten it.

Give me a couple of days to do some calculations.  I may be able to predict the compensation based just on the error you see at room temperature.  It'd be better to do it based on DMM readings.

madhunm, 2016-12-22:
@Bob-the-Kuhn 

Here is some info for you:

I stuck the sensor in an ice bath. the display read 18 degrees
at an ambient of ~23 degrees, it reads 41 degrees.

Looks like there is an off set of 18 degrees over the actual temperature.

Do you want me to log temperatures at some pre-defined increments?

bsculley, 2016-12-22:
What temperature table are you using for this?  My experience was pretty opposite.  With the current 147 table (calculated, so you can’t actually see the vaues) I was reading -10 at ambient temperature of 18-20.  

 

Something isn’t lining up here…

bsculley, 2016-12-22:
Did you use the amplifier with the pull-up still in place, or on a different analog input.

 

I didn’t have any luck with amplifier / pull-up combination.

 

I also got a low reading with 147 at ambient.  Mine was below zero.

bsculley, 2016-12-22:
I found a PT100 temperature table here.  http://www.micropik.com/PDF/pt100.pdf

 

I made the bold assumption that they are all pretty much the same.

madhunm, 2016-12-22:
@Bob-the-Kuhn @bsculley 

I tried both 147 and 20. Currently, I am on 20 with the amp connected.

Odd thing is, the PT100 reads 110.6 ohms. but, the Amp output is 1.219V (which corresponds to 42 degrees)

The display is showing 42 degrees.

Maybe my amp is fried?

Thanks,
Madhu

bsculley, 2016-12-22:
That is odd.  Are you measuring the voltage just on the amplifier or on the amplifier connected to the analog input?

 

Is the pull-up still in place in the analog input?

madhunm, 2016-12-22:
> auth/AFN-M0po9vbeJzuotmVhCyk-B36ihS4-ks5rKn59gaJpZM4LQNGh> . <

Yes. I am measuring with the amp connected to the abalog input.

the 4k7 pull up is still in place.

bsculley, 2016-12-22:
Then the voltage is being affected by the potential from the pull-up, which has 5v on the other side of it.

 

Try measuring the output of the amplifier while it is not connected to the input.

 

I think maybe this configuration can be made to work, but it will probably require a different temperature table.

 

If this is feasible it would be great if one of the gurus could create another device type (247 maybe) for PT100 amplifier connected to analog input with 4k7 pull-up.

madhunm, 2016-12-22:
> M6qiDM1mpbKo5ABPYMLYqHXtpcTkks
> auth/AFN-M8DAFfhu3vxs_Qlg8tIuKbJCwIarks5rKoJwgaJpZM4LQNGh> . <

Just to eliminate a defective sensor, I am currently compiling a custom 147
thermistor table. Please stand by, I shall update shortly.

bsculley, 2016-12-22:
Great!  

 

By the way, the E3d docs on the amplifier (http://wiki.e3d-online.com/wiki/E3D_PT100_Amplifier_Documentation#Output_Characteristics

) show that a voltage of 1.219 corresponds about  25 degrees.

madhunm, 2016-12-22:
> Great!
>
>
>
> By the way, the E3d docs on the amplifier (http://wiki.e3d-online.com/
> wiki/E3D_PT100_Amplifier_Documentation#Output_Characteristics
>
> ) show that a voltage of 1.219 corresponds about 25 degrees.

Progress! I used the values for 147 as listed here:

http://shop.prusa3d.com/forum/improvements-f14/mini-rambo-1-3-and-e3d-with-pt100-t700-s40.html#p20938

at room temperature, I am now at 34 degrees!

bsculley, 2016-12-22:
That’s the table I made.  The first two entries (0 and 50 degrees) are not rational.  I fudged them to get reasonable readings.  The calculated values were 341 (0 degrees) and 406 (50 degrees).

 

Can you confirm your exact hardware/firmware setup?  I really want to make the amplifier work with the pull-up!

 

http://shop.prusa3d.com/forum/improvements-f14/mini-rambo-1-3-and-e3d-with-pt100-t700-s40.html#p20938

madhunm, 2016-12-22:
> 3-and-e3d-with-pt100-t700-s40.html#p20938
> 20%3chttps:/github.com/%0b> <https://github.com/
> MwJIKo9tsRoevFIKX5gjSFhm40jYks5rKogegaJpZM4LQNGh> . <https://github.com/
> notifications/beacon/AFN-M3Xo_efkPFNXBi1aK3WciQT6Gna2ks5rKog

My setup is the e3d PT100 kit. I would love to use the amp. Seems a and bit
to use it. But, if I can get away without resoldering the pullups.

Bob-the-Kuhn, 2016-12-22:
The problem with the PT100 amplifier is that it has about 80 Ohms of output impedance (per my calculations).  At room temperature there's about 0.8mA being sunk by the amplifier (when driving the 4.7K pullup).  0.8mA x 100 Ohms gives us about 80mV which translates to a reading 20C higher than actual.

At 200C the current drops down to 0.65mA which gives us a 65mV offset which is a 16C error.

You're seeing a 20C error at room temp.  My amp had a 25C error.

I'll generate a table based on the 20C offset you're seeing.

madhunm, 2016-12-22:
Bob,

If you're using a spreadsheet, please attach the same... Or at least, the
calculation.

Thanks!

bsculley, 2016-12-22:
Here’s the table.  

 

The first 2 columns are from the PT100 datasheet.

Volts =  (Ohms/(4700+Ohms)) * 5

ADC steps = (1024/5) * Volts

147 = ADC Steps * 16

 

Let me know if any of the calcs are not clear.

 

Temp

Ohms

Volts

ADC steps

147

0

100

0.104

21

336

50

119.4

0.124

25

400

100

138.5

0.143

29

464

150

157.33

0.162

33

528

200

175.86

0.18

37

576

250

194.1

0.198

41

640

300

212.05

0.216

44

704

350

229.72

0.233

48

752

400

247.09

0.25

51

816

450

264.18

0.266

54

864

500

280.98

0.282

58

912

550

297.49

0.298

61

976

bsculley, 2016-12-22:
Will the offset be constant or vary with the voltage?

 

Looking forward to the table!

 

Can you include a calibration adjustment for the particular PT100/amplifier readings?

bsculley, 2016-12-22:
Oh, I  guess you didn’t mean me.  Sorry.

madhunm, 2016-12-22:
@Bob-the-Kuhn 

I used the values you gave above; at room temp, the display is reading 550 degrees.

I measured the output of the amp when connected to the Rambo; it is reading 1.218 volts.

Going by the logic you have posted above, the ADC should never read more than 976 (for 550 degrees). However, the amp has a gain of 10. Doesnt this need to be incorporated somehow?

manianac, 2016-12-22:
Small recommendation, rather than using the calculations, and given you're trying to account for very small change in ADC steps, just enable SHOW_TEMP_ADC_VALUES in Configuration_adv.h and try calibrating using those.  I'm still believing the RAMBO and the mini will have the adc steps a few off than other boards.

Bob-the-Kuhn, 2016-12-22:
Here's an Excel file that'll take your temperature readings and create a column that you can paste into Marlin as is. 

   [PT100 calculator.zip](https://github.com/MarlinFirmware/Marlin/files/669475/PT100.calculator.zip)

You'll need to convert the room temperature and the temperature reported by the controller into voltages per the PT100 table mentioned a couple of times here.  That table is already in the Excel spreadsheet.

You can also play around with variations in supply voltages & resistances to see how they affect the reported temperature.

This is definitely NOT an ideal solution.  There's definitely better ways to generate the desired table.

My guess is this method will get you within 5C of actual at 200C, not including things like the PT100 sensor's accuracy and the non-linear behavior of the amplifier.

madhunm, 2016-12-22:
@Bob-the-Kuhn 

Thanks a million! this works spotlessly... Although, since I dont know if this is the right place to ask you for an explanation of the formulae, please let me know how I may contact you outside of here.

Thanks,
madhu.

madhunm, 2016-12-22:
quick one... is it fair to compare the PT100 to the bed thermistor?

The bed is reporting that it is ALWAYS 10 degrees hotter than the PT100...

thanks.

manianac, 2016-12-22:
For whats its worth, my PT100 is using the e3d amp and connected to an open analog input, no resistors.  At room temperature (21degC) my bed thermistor (after sitting overnight) reads 21.6degC, the PT100 18degC.

Point I'm trying to make is unless you have an external source to calibrate, room temp is not that great for the PT100.

madhunm, 2016-12-22:
@manianac I agree. 

I am using a Fluke thermocouple to measure against. Before @Bob-the-Kuhn posted the excel file, i was off by 20 degrees. 

hence the maniacal stubbornness. Now, I am comparing it against the heatbed; i dont know if that is right. the thermocouple and PT100 are within 3 degrees of one other.

Bob-the-Kuhn, 2016-12-22:
Here's the best I can do without spending half the day learning how to run a schematic capture program.
[PT100 equations.docx](https://github.com/MarlinFirmware/Marlin/files/669920/PT100.equations.docx)

The docx file contains a hand drawn system diagram and the equations  for it.

madhunm, 2016-12-23:
@Bob-the-Kuhn 

Sorry to be a pest; but, the doc has no hand drawn document - at least, nothing that can be seen. At the top, there is a thumbnail for img021.pdf but that cannot be opened.

Bob-the-Kuhn, 2016-12-23:
Here it is.

[system_diagram.pdf](https://github.com/MarlinFirmware/Marlin/files/671156/system_diagram.pdf)

bsculley, 2016-12-26:
Season’s Greetings!

 

Don’t want to be a freeloader, but if anybody has actually calculated a temperature table it would be great if you could share it.  I don’t trust my math skills all that well.

Bob-the-Kuhn, 2016-12-26:
Further up this chain you'll see a **PT100.calculator.zip** file.  I think that'll do what you want.

bsculley, 2016-12-27:
Got it, thanks.

 

What does anybody think about adding a new device type for this configuration?

madhunm, 2017-01-06:
@Bob-the-Kuhn

I would like to use a PT100 without the amp for the heated bed. Is that recommended? 

The reason I ask is because, the gremlins of 147 are back to haunt me. I am not able to get a sensible reading when compared to the PT100 with amp. 

If I consider the amped sensor as reference, and I mount them to the same heat block, I'm not able to get readings that are even in the same ballpark. The readings are off by 50 degrees sometimes. 

Thanks, 
Madhu.

thinkyhead, 2017-04-02:
How's it going? Any progress?

Bob-the-Kuhn, 2017-04-02:
I did a few calculations and the 147 table only has about 9 degrees C resolution.

If the internal pullup was also enabled then his readings would be off by 100C and more.

The 4.7k pullup would have to be off by 10% in order to throw the readings off by 50C at 260C.

closed 2018-02-06 by thinkyhead