MarlinFirmware/Marlin #5546: G29 result in KILLED

pettib, 2016-12-18:
I just upgraded my Prusa I3 to RC8 and got problem with G29. I also tried RC8 BugFix but same problem. I can do 28 homing and it´s working good but when i start the leveling with G29 it goes to the first point and deploy the probe and goes up, when i starts to move to next point it stops after about 1cm in X move. and the diplay says KILLED, Printer Halted. Always on the same point. 
I tried with both Liniear and Bilinear version of 3 point leveling.

Some with a guess what this can be or how I can find out what the problem is. 

Regards Peter

Grogyan, 2016-12-18:
What probe are you using?

pettib, 2016-12-18:
I´m using a servo with microswitch

logs from Repetier:
`23:03:18.511 : echo:DEBUG:INFO,ERRORS
23:03:18.511 : echo:Active Extruder: 0
23:03:32.232 : X:200.00 Y:0.00 Z:5.00 E:0.00 Count X: 64000 Y:0 Z:39940
23:03:36.913 : X:200.00 Y:200.00 Z:5.00 E:0.00 Count X: 64000 Y:32000 Z:39940
23:04:07.512 : X:78.00 Y:97.00 Z:5.00 E:0.00 Count X: 24960 Y:15520 Z:39940
23:04:26.684 : G29 Auto Bed Leveling
23:04:38.771 : Error:Printer halted. kill() called!
`

Configuration.h
`/**
 * Marlin 3D Printer Firmware
 * Copyright (C) 2016 MarlinFirmware [https://github.com/MarlinFirmware/Marlin]
 *
 * Based on Sprinter and grbl.
 * Copyright (C) 2011 Camiel Gubbels / Erik van der Zalm
 *
 * This program is free software: you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation, either version 3 of the License, or
 * (at your option) any later version.
 *
 * This program is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this program.  If not, see <http://www.gnu.org/licenses/>.
 *
 */

/**
 * Configuration.h
 *
 * Basic settings such as:
 *
 * - Type of electronics
 * - Type of temperature sensor
 * - Printer geometry
 * - Endstop configuration
 * - LCD controller
 * - Extra features
 *
 * Advanced settings can be found in Configuration_adv.h
 *
 */
#ifndef CONFIGURATION_H
#define CONFIGURATION_H

/**
 *
 *  ***********************************
 *  **  ATTENTION TO ALL DEVELOPERS  **
 *  ***********************************
 *
 * You must increment this version number for every significant change such as,
 * but not limited to: ADD, DELETE RENAME OR REPURPOSE any directive/option.
 *
 * Note: Update also Version.h !
 */
#define CONFIGURATION_H_VERSION 010100

//===========================================================================
//============================= Getting Started =============================
//===========================================================================

[... 1377 lines cut ...]
 * 34 - RAMPS1.4 - uses Analog input 5 on the AUX2 connector
 * 81 - Printrboard - Uses Analog input 2 on the Exp1 connector (version B,C,D,E)
 * 301 - Rambo  - uses Analog input 3
 * Note may require analog pins to be defined for different motherboards
 **********************************************************************/
// Uncomment below to enable
//#define FILAMENT_WIDTH_SENSOR

#define DEFAULT_NOMINAL_FILAMENT_DIA 3.00  //Enter the diameter (in mm) of the filament generally used (3.0 mm or 1.75 mm) - this is then used in the slicer software.  Used for sensor reading validation

#if ENABLED(FILAMENT_WIDTH_SENSOR)
  #define FILAMENT_SENSOR_EXTRUDER_NUM 0   //The number of the extruder that has the filament sensor (0,1,2)
  #define MEASUREMENT_DELAY_CM        14   //measurement delay in cm.  This is the distance from filament sensor to middle of barrel

  #define MEASURED_UPPER_LIMIT         3.30  //upper limit factor used for sensor reading validation in mm
  #define MEASURED_LOWER_LIMIT         1.90  //lower limit factor for sensor reading validation in mm
  #define MAX_MEASUREMENT_DELAY       20     //delay buffer size in bytes (1 byte = 1cm)- limits maximum measurement delay allowable (must be larger than MEASUREMENT_DELAY_CM  and lower number saves RAM)

  #define DEFAULT_MEASURED_FILAMENT_DIA  DEFAULT_NOMINAL_FILAMENT_DIA  //set measured to nominal initially

  //When using an LCD, uncomment the line below to display the Filament sensor data on the last line instead of status.  Status will appear for 5 sec.
  //#define FILAMENT_LCD_DISPLAY
#endif

#endif // CONFIGURATION_H`

Blue-Marlin, 2016-12-18:
How many points do you want to probe? Do you have enough RAM for that?

pettib, 2016-12-18:
3x3 points.
Running on a Ramps 1.4 on Arduino board.
I´m running the RC7 bugfix on another printer with the same board and there it´s working.

ozligia, 2016-12-19:
@pettib, just a guess. Do you have advance feature enabled in configuration_adv.h
```
//#define ADVANCE <-this line uncommented
#if ENABLED(ADVANCE)
  #define EXTRUDER_ADVANCE_K .3
  #define D_FILAMENT 1.75
#endif
```
Just ran into similar issue on my CoreXY.

soimy, 2016-12-19:
@pettib Just to remind you, if you have stepper drive set to 1/64 or 1/128 step, and/or enabling LIN_ADVANCE feature, you best option is slow down your XY movement speed to 50-100, otherwise it will kill your mega2560. It just don't powerful enough...

pettib, 2016-12-19:
Problem solved !
It was as soimy wrote. It was problem with the speed of the movement.
I reduced the speed and is now working.
Thanks everyone.

fiveangle, 2016-12-19:
Slow down XY movement speed to 50-100 when enabling LINEAR_ADV ?  Heck, it's enabling LINEAR_ADV that _allows_ me to _increase_ print speed from 50 to 100mm/s.  @pettib - What happens if you revert to the high movement speed but drop serial speed to 115200 ?

fiveangle, 2016-12-19:
@pettib - now tha I think about it, a lot of us would be very curious if you saw a different behavior when running the original settings but with PR #5481 FW ?

soimy, 2016-12-20:
> different behavior when running the original settings but with PR #5481 FW ?
This PR haven't been merged to RCBugfix branch yet, but I'm looking forward to it. Currently using LIN_ADANCE will add more pressure to Timer used by steppers.

Sebastianv650, 2016-12-20:
I think something got mixed up here. @pettib mentioned nowhere he is using ADVANCE or LIN_ADVANCE. @fiveangle slowdown the print speed in relation with high microstepping is a result of the fact that Marlin can only do up to 40.000 steps/s in theory. In reality, depending in with features or kinematics are enabled, it may be even lower. If you have a fixed upper steps/s value, the resulting max. possible speed is depending on the steps/mm value. Higher microstepping, higher steps/mm value.

fiveangle, 2016-12-28:
Yes, @Sabastionv650 i skimmed @ozgligia 's comment too quickly before.

thinkyhead, 2017-03-04:
Most likely fixed in the current `RCBugFix`. If not, start a new issue and mention this one.

closed 2017-03-04 by thinkyhead