MarlinFirmware/Marlin #5551: Bed PID tunning weird behaviour

bolsoncerrado, 2016-12-19:
I frankly dunno how to call this.... It's the third printer I install Marlin RC7 onto and when running the PID autotune for the bed I get the following weird behaviour:

https://www.dropbox.com/s/3ufajrvrjhh6flv/bedFAIL.txt?dl=0

So once I reach the temp i requested, there are 2 bounces and on the third, bed tries to go higher and higher and higher....

This never happened before on any of the other beds I calibrated :????

Any clues whats going on?

The Marlin hasnt been modified at all between them, athough the beds are different "brands".

Blue-Marlin, 2016-12-19:
Try at a slightly lower temperature to get a bit more overshoot. If hat works your bed heater could be a bit weak.
When tuning PID the time for overshoot and undershoot is compared. If one of the times is zero or too close to zero it will not work.

bolsoncerrado, 2016-12-19:
Now that was weird, I had to go as low as 40ºC for the autotune to complete hmmm...
i agree its cold as hell today in here... perhaps all togheter...

closed 2017-10-31 by thinkyhead