MarlinFirmware/Marlin #5554: Mixing Extruder woes, unsure of bug?

jwcrawley, 2016-12-19:
I am having issues with Marlin's configuration.h file, and how to set settings for a Mixing Hotend, the Diamond. 

Symptom: Setting Esteps for the 3 extruders (they are different motors) do not seem to properly hold for E2 and possibly E1. I can configure E0 correctly.

The relevant settings I have are:

#define EXTRUDERS 1
#define DISTINCT_E_FACTORS

#define MIXING_EXTRUDER
#if ENABLED(MIXING_EXTRUDER)
  #define MIXING_STEPPERS 3        // Number of steppers in your mixing extruder
  #define MIXING_VIRTUAL_TOOLS 16  // Use the Virtual Tool method with M163 and M164
  #define DIRECT_MIXING_IN_G1    // Allow ABCDHI mix factors in G1 movement commands
#endif

When I enable "allow cold extrusion" and then follow it by 

M163 S0 P1
M163 S1 P0
M163 S2 P0

I can test each eStep for the extruders (by shifting the P1 and P0's to relevent place). However, S2 does spin in the right direction, but changing the appropriate eSteps further down in the configuration.h file does *nothing* to amount of filament pushed.

S1 *might* have problems, but is within tolerance of my current measurement (my digital calipers broke). S2, there is absolutely no question this is not good.

jwcrawley, 2016-12-19:
[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/661918/Configuration.txt)

I had to set extruders at 1. If I changed it to Extruders=3, then the following error is made (in bold):

```
 #if ENABLED(MIXING_EXTRUDER)
  #if EXTRUDERS > 1
    **#error "MIXING_EXTRUDER currently only supports one extruder."**
  #endif
  #if MIXING_STEPPERS < 2
    #error "You must set MIXING_STEPPERS >= 2 for a mixing extruder."
  #endif
  #if ENABLED(FILAMENT_SENSOR)
    #error "MIXING_EXTRUDER is incompatible with FILAMENT_SENSOR. Comment out this line to use it anyway."
  #endif
#endif
```

Because my machine has 3 extruder motors feeding into the diamond, mixing, and extruding a single line of plastic. 

It would seem that I would have EXTRUDERS 3 ... yet errors out if I do that. 

Is this user error, Marlin Error, or something else?

jwcrawley, 2016-12-19:
Tl;Dr. Possible bug: Esteps are ignored for E2 and potentially E1, when using a 3 extruder Mixing hotend, like the Diamond.

Unsure if bad config, or bug.

closed 2016-12-23 by jwcrawley

thinkyhead, 2017-02-11:
Resolved with `RC8` or `RCBugFix`?