MarlinFirmware/Marlin #5555: Display empty screen for first menu item

vprheli, 2016-12-19:
Seems there is bug in the new option in MENU_HOLLOW_FRAME is defined in the ultralcd_impl_DOGM.h
By default it is defined. To see old (RC7) inverted line of selected menu item, it is necessary comment this line.
Then every click from FIRST menu line to modify parameter displayed on the third line show empty screen. Manu click from second or next lines works correctly.
Try Prepare-MoveAxis-Move 10mm-MoveX -> show empty screen OR Control-Temperature-Nozzle -> again empty screen but Control-Temperature-Bed (second menu line) item is OK. Seems that the screen cleaning area is too height.

thinkyhead, 2016-12-20:
I'll mess with this tonight.

Kaibob2, 2016-12-20:
I can not confirm!  
See https://github.com/MarlinFirmware/Marlin/issues/5194#issuecomment-268078495

vprheli, 2016-12-20:
> See #5194 (comment)

Hi, I am sorry, the bug is still there. I downloaded the ZIP file from the
https://github.com/MarlinFirmware/Marlin/tree/RCBugFix and changed only
couple of parameters in configuration files - see attachment.
Menu-Control-Temperature-Nozzle - empty screen.

Arduino IDE 1.6.12, ArduinoMEGA2560, RAMPS1.4,
REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER

and the inverted selected menu and X-Y-Z in the status screen

#if ENABLED(DOGLCD)
  // Enable to save many cycles by drawing a hollow frame on the Info Screen
  //#define XYZ_HOLLOW_FRAME
  // Enable to save many cycles by drawing a hollow frame on Menu Screens
  //#define MENU_HOLLOW_FRAME

This is only one change in the Configuration_adv.h I changed only some
parameters only for testing this bug.

For "long" time I used RC8, where this is working.

Vladimir

vprheli, 2016-12-20:
And just one comment. :o) If You test Try Prepare > Move Axis > Move 10mm > Move X, this is wrong test in the last BugFix - Move X is not on the "second line" anymore. There is additional line on the top "Move X" which scroll menu one line down.
Control-Temperature-Nozzle is using still the menu without "caption" and this failed.

Kaibob2, 2016-12-20:
This video is done with the latest RCBugFix 1a26ce1cdcebf1d6e1bbd5b05c74602cc634f6ac from today.
[20161220_181518.zip](https://github.com/MarlinFirmware/Marlin/files/664333/20161220_181518.zip)

vprheli, 2016-12-21:
:o)) Sorry, this is what I wan't to do. I don't want hollow frame in the menu.

Kaibob2, 2016-12-21:
Sorry, my fault, just verified it with
`//#define XYZ_HOLLOW_FRAME`
and
`//#define MENU_HOLLOW_FRAME`
and it shows the blank screen when Control->Temperature->Nozzle.

BUT
[this fixes it](https://github.com/MarlinFirmware/Marlin/pull/5567/files)

vprheli, 2016-12-21:
Yes, fixed. Great job. Thanks

Kaibob2, 2016-12-22:
Confirmed fixed. Please close this one then.

closed 2017-10-31 by thinkyhead