MarlinFirmware/Marlin #5564: Homing?

piesafety, 2016-12-20:
When homing the printer the motors just moves and stops before even hitting the endstop. Firmware or mechanical problem?

Fulg, 2016-12-20:
Did you check that your endstops were working with `M119`?

piesafety, 2016-12-20:
It seems to alway be in the triggered position no matter what.
even after switching firmware

jwcrawley, 2016-12-20:
Greetings,

I had a similar issue with my kossel mini Delta. The underlying problem was
that the configurations included Z minimum enabled.

Since I do not have a automatic bed Leveling Sensor, I commented out z_min
. After I did so each motor would home to 0, hit the switch rapidly, back
up and slowly approach until triggered.

I have an issue with a link video showing what happens when you have z_min
enabled. Z motor goes crazy during homing. Just look up a username to see
the issue , as I am on my phone.

Sincerely, Josh

Fulg, 2016-12-20:
If the endstops are always triggered, homing will never work. Fix that first. Check if the status changes when pressing the endstop by hand (meaning you can just reverse them in the config file). If instead nothing happens, then you have a wiring problem (use a multimeter to confirm the endstop works) or Marlin is checking the wrong pin (configuration problem).

You should attach your Configuration.h. What hardware do you have?

jwcrawley, 2016-12-20:
From what I see, the underlying confusion is having Z minimum in stop
enabled by default along with the appropriate logic.

So when one downloads and configures Marlin, they missed this. And by the
default, and in stop with nothing plugged in is triggered. That is the
problem. Nothing should be triggered if nothing is plugged in.

It may be argued that this is intended behavior. However it differs from
prior settings. I have counted at least four people who have had the
similar issue, and has had it resolved partially with my help.

I would encourage the Marlin developer Team to consider the ramifications
of having nothing plugged into extra in stocks. This would clear up a great
deal of problems.

sincerely, Josh

thinkyhead, 2017-02-12:
For deltas, we recommend starting with one of the Delta-specific `example_configurations`, which have the endstops (and much more) already set up for delta.

closed 2017-02-12 by thinkyhead