MarlinFirmware/Marlin #5566: Move Z up on print pause - feature request

playlet, 2016-12-20:
Hi!

I was looking into the repetier web configuration and saw a great feature that would be nice to have in marlin.

It would be great if we could include a command to move Z up a couple of mm when print pause is activated. If you click pause, nozzle stays on the object and burns a little crater in that place which is later visible if it stops on the perimeter...

Best regards,

Ivan

Kaibob2, 2016-12-20:
This is an often reported request. See #5539
I think thinkyhead has it on his TODO list :)

manianac, 2016-12-20:
I may check this out, are you pausing with M25 or LCD?

playlet, 2016-12-20:
LCD. For me, it's not practical to keep computer connected to the printer. :)

Kaibob2, 2016-12-20:
It doesn't matter if you pause by M25 via serial or via Pause on LCD. The effect of the nozzle staying in place is the same for both scenarios.

akaJes, 2016-12-21:
Can i ask you what is a goal of this command in your case?
because i found for myself M600 command (filament changing) and use it as Pause
and i got two in one and better (M600 doing a retract then get a nozzle up then move to desired XY )
because in my case extrusion do not stop when extruder not working already and filament fall out on the model

playlet, 2016-12-21:
I actually use pause function to change filament (insert different colour). I don't want to use M600 because then i can't use filament runout protection. They don't work simoultaneously. 

Anyhow, Z up after pause would be very useful for any application.

manianac, 2016-12-21:
I was looking into providing a nozzle park or even heater shutoff / resume when SD pause/resume.  But if M600 already would work you should try that.

But you other statement about M600 has me wondering, why won't it work with filament runout protection?  The entire procedure I thought was self-contained and it resets the runout flag at the end before resuming normal operation?

playlet, 2016-12-21:
You can't compile marlin with both features turned on. It stops and shows the warning.

manianac, 2016-12-21:
Can you post the warning it shows?  I just compiled Marlin with both turned on without an issue.

Hell the default runout_script is to run an `M600`, it'd be odd if the two didn't play nice.

playlet, 2016-12-21:
SanityCheck.h:255: error: #error "EXTRUDER_RUNOUT_PREVENT is incompatible with FILAMENT_CHANGE_FEATURE."

   #error "EXTRUDER_RUNOUT_PREVENT is incompatible with FILAMENT_CHANGE_FEATURE."

    ^

exit status 1
#error "EXTRUDER_RUNOUT_PREVENT is incompatible with FILAMENT_CHANGE_FEATURE."

manianac, 2016-12-21:
That is not filament runout prevention, that is extruder runout.  All that feature does is extrude filament every couple of seconds to avoid an empty hotend (since it all oozed out).

Given your problem was ooze ruining the print anyway, why not disable that feature?  You prime the extruder when doing an `M600` anyway.

playlet, 2016-12-21:
Sorry, my bad, I forgot these are not the same as I don't have filament runout sensor. :)

manianac, 2016-12-21:
Its all good, but if I were you I'd weigh which feature is more important to you ;).  A longer skirt makes up for an empty extruder anyway.

playlet, 2016-12-21:
Yeah, I gotta try that. Still, Z up could be a nice optional addition. :) Thanks for looking into this.

manianac, 2016-12-21:
I'm using M600 as a template, but I see no reason not to have a configuration option when pausing SD prints to park them at a coordinate.  Even powering off heaters and re-enabling them on resume seems do-able.

manianac, 2016-12-22:
Alright @playlet, its pretty hacky, but I figured you'd be interested in trying out [this branch of RCBugFix](https://github.com/manianac/Marlin/tree/RCBugFix-ParkWhenPaused).

Whenever you initiate a Pause from the LCD, it will retract and move to X/Y stated in Configuration_adv.h.  Look for `SDPAUSE_PARK`.

This code is pretty ugly, and not worth a pull request into Marlin yet.

playlet, 2016-12-22:
I LIKE IT! :)
Although I recommend to leave retract at 0mm and set only a small recover length, say 0.1mm, because plastic oozes out after a few seconds of waiting and recover mm make a huge blob on resume. :)

Thank you sir, you are very kind for taking the time to create this code for me! :)

thinkyhead, 2017-04-02:
This has now been added. However, it requires that you also enable `M600` Filament Change.

closed 2017-04-02 by thinkyhead

playlet, 2017-04-02:
nice, thank you! :)