MarlinFirmware/Marlin #5568: Inconsistent [X][Y][Z]_PROBE_OFFSET_FROM_EXTRUDER treatment

yonkiman, 2016-12-20:
In `Configuration.h`, just noticed that non-integer values for `X_PROBE_OFFSET_FROM_EXTRUDER` and `Y_PROBE_OFFSET_FROM_EXTRUDER` cause 'Configuration.h:656: error: floating constant in preprocessor expression' errors at compilation.  However `Z_PROBE_OFFSET_FROM_EXTRUDER 0.2` compiles fine.  Shouldn't they all be floating point?

On a related note, compiled with `Z_PROBE_OFFSET_FROM_EXTRUDER 0.2`, the once running, executed `M502` then `M501`, which returned `M851 Z0.10`.  Shouldn't it have been 0.2 as specified by `Z_PROBE_OFFSET_FROM_EXTRUDER`?

manianac, 2016-12-20:
For the second note, `M501` just reloaded what you had saved in the EEPROM.  Do a `M502` and an `M500` to overwrite the EEPROM with the default settings.

Or just do a `M851 Z0.2` and `M500` to achieve the same outcome.

yonkiman, 2016-12-20:
Yep.  Appreciate the quick response, sorry for the distraction.

closed 2016-12-20 by yonkiman

Blue-Marlin, 2016-12-20:
> // Z Probe to nozzle (X,Y) offset, relative to (0, 0).
// __X and Y offsets must be integers.__
//
// In the following example the X and Y offsets are both positive:
// #define X_PROBE_OFFSET_FROM_EXTRUDER 10
// #define Y_PROBE_OFFSET_FROM_EXTRUDER 10
//
//    +-- BACK ---+
//    |           |
//  L |    (+) P  | R <-- probe (20,20)
//  E |           | I
//  F | (-) N (+) | G <-- nozzle (10,10)
//  T |           | H
//    |    (-)    | T
//    |           |
//    O-- FRONT --+
//  (0,0)
#define X_PROBE_OFFSET_FROM_EXTRUDER 10  // X offset: -left  +right  [of the nozzle]
#define Y_PROBE_OFFSET_FROM_EXTRUDER 10  // Y offset: -front +behind [the nozzle]
#define Z_PROBE_OFFSET_FROM_EXTRUDER 0   // Z offset: -below +above  [the nozzle]

Is clear, isn't it?
The preprocessor can't calculate with floats. With x and y we do that but not with z.
Resolution in mm is completely sufficient for x and y, but not for z.

yonkiman, 2016-12-21:
Totally clear - very embarrassed I didn't RTFM particularly since TFM was literally right above the code.

Again, sorry I wasted everyone's time.  Will be more careful in the future.