MarlinFirmware/Marlin #5569: PR: kill boards that are no longer available?

Bob-the-Kuhn, 2016-12-20:
I’ve been going through the pins_XXXX.h files and it’s obvious that a lot of the non mega1280/2560 boards won’t compile the latest Marlin software (and have a useful image).  That implies that the boards are no longer available.

My proposal is to drop support for the boards that are no longer available.  Specifically we’d delete the pins_XXX.h files and remove them from the boards.h file.

I’ll update the ones that are currently available.

As part of the update I’ll also include pointers to currently available extensions to the Arduino IDE that support the board.  Teensyduino is the only one I know of that works with Arduino 1.6.5 & later and has a pin map that matches FASTIO.h. Pointers to other useable extensions would be welcome.

Below is a list of all the currently supported AT90USB646/1286 and ATmega644/1284 boards.  Please let me know what boards in the “available?” section are still available.

available?..pins.h...................CPU..................
............pins_5DPRINT.h...........AT90USB1286..........
............pins_BRAINWAVE_PRO.h.....AT90USB1286..........
............pins_PRINTRBOARD.h.......AT90USB1286..........
............pins_PRINTRBOARD_REVF.h..AT90USB1286..........
............pins_SAV_MKI.h...........AT90USB1286..........
............pins_BRAINWAVE.h.........AT90USB646 ..........
............pins_LEAPFROG.h..........ATmega1280/2560......
............pins_MELZI.h.............ATmega1284P..........
............pins_MELZI_MAKR3D.h......ATmega1284P..........
............pins_STB_11.h............ATmega1284p .........
............pins_GEN6_DELUXE.h.......Atmega1284P?.........
............pins_OMCA.h..............ATmega644............
............pins_OMCA_A.h............ATmega644............
............pins_GEN6.h..............ATmega644A...........
............pins_GEN7_12.h...........ATmega644A...........
............pins_GEN7_13.h...........ATmega644A...........
............pins_GEN3_MONOLITHIC.h...ATmega644P...........
............pins_GEN7_14.h...........ATmega644P...........
............pins_GEN7_CUSTOM.h.......ATmega644P...........
............pins_SANGUINOLOLU_11.h...ATmega644P...........
............pins_GEN3_PLUS.h.........ATmega644P /1284P....
............pins_SETHI.h.............ATmega644P /1284P....

keep......................................................
............pins_TEENSY2.h...........AT90USB1286..........
............pins_TEENSYLU.h..........AT90USB1286..........
............pins_SANGUINOLOLU_12.h...ATmega1284P..........
............pins_AZTEEG_X1.h.........V1 -644P, V2 - 1284..

kill......................................................
............pins_99.h................??...................

Blue-Marlin, 2016-12-20:
Availability is not a valid criteria.
There may be enough old boards out there to justify there existence in Marlin.

manianac, 2016-12-21:
I would vote against removing support for a board because its not available.

Now removing support because it won't compile is a different thing.  An issue should probably be made if it won't compile and support removed based on if the issue can be fixed.

thinkyhead, 2016-12-23:
I think it's good to keep around as many known boards we can, for historical and informational reasons. Even for those small boards where now Marlin might not easily fit, someone may still be able to whittle it down to make it work.

boelle, 2016-12-23:
pins_SAV_MKI.h

http://reprap.org/wiki/File:SAV_MK-I_RevE.pdf

and

http://www.electrofunltd.com/p/buy.html

just out of stock... of course it is and who would restock if they are making a new rev

and yes even boards with a 644 should stay... who knows if marlin will fit someday

gege2b, 2016-12-23:
Hi
I own an old prusa mendel with a sanguinololu 1.3 board (644p)

I will try RCx/BugFix as soon as I got some time to recover my lost configuration.h, but as far as I tested, Marlin should compile and fit on the 644p (until I keep very basic features only, so no ABL, no LCD/SD, etc... but it's not a problem for me)

Bob-the-Kuhn, 2016-12-24:
@gege2b  - what do you use to compile for your board?  

I couldn't find anything for the current Arduino DE so I modified the MightyBoard system.

gege2b, 2016-12-24:
arduino IDE under linux :)
and  https://github.com/Lauszus/sanguino to include the board into the IDE

I didn't uploaded the code on the board yet , just compiled it for now. That's why I'm not completely sure it will work

Bob-the-Kuhn, 2016-12-26:
All - thanks for the input.  I'll be including all the boards when I update the pinsXXXX.h files.

@gege2b - thanks for the pointer to the Sanguino IDE package.  It's actively being maintained and has agreed to a few minor changes to support our 644/1284 boards.  I'll be pointing these boards to that package.

closed 2016-12-26 by Bob-the-Kuhn