MarlinFirmware/Marlin #5570: Request: Be able to set timeout on LCD screen

jwcrawley, 2016-12-20:
When I am working with the controls on my lcd/knob and jogging and preheating bed, it stays on that screen for only a few seconds, and then defaults back to the home screen. 

Is there a setting I can set, either in configuration.h , configuration_adv.h , or the source file that would allow extending this timeout to something I would like, say 30 seconds?

Is there a reason why this is not done?

Sincerely,
Josh Conway

jwcrawley, 2016-12-20:
Ok, a quick grep, and I might have found it. I'm away from my printer, so I cannot verify, but is this the setting to change?

```
<In Conditionals_LCD.h line 277-279 >
  #ifndef BOOTSCREEN_TIMEOUT
    #define BOOTSCREEN_TIMEOUT 2500
  #endif
```

And that would be in milliseconds, changed to the timeout I want, correct?

Kaibob2, 2016-12-20:
What you want to edit is [this value here in ultralcd.h](https://github.com/MarlinFirmware/Marlin/blob/RCBugFix/Marlin/ultralcd.h#L67) 

Any screen, except for Babystepping, will return to the Info screen after 15000ms on default. You can change to your demand.

jwcrawley, 2016-12-20:
Awesome. Thank you much, @Kaibob2

Kaibob2, 2016-12-21:
Please close here if you have no further questions regarding this.

closed 2016-12-21 by jwcrawley

ruggb, 2017-02-28:
there is a problem here.
1st - there is the setting in ultralcd.h to set the timeout - default = 15000 or 15sec. 
2nd - This function DID work for babystepping in RC7. It does not in RC8 so I think it is a bug that needs fixing. What about you??? @Roxy-3D has verified it - will it get fixed b4 release??????????

Kaibob2, 2017-02-28:
I always thought it was because of 
[defer_return_to_status](https://github.com/MarlinFirmware/Marlin/blob/7bea5e5e5701de0b90b6c422c954337ce860bb0f/Marlin/ultralcd.cpp#L690) and so...intended.

I can remember, that approx. 6 month ago somebody requested that the timeout should be ignored if you're on the babystepping screen. This was then implemented into RC7 BugFix but i don't know how. The original RC7 was respecting the timeout also for the babystepping screen.

ruggb, 2017-02-28:
Since it went away, I assumed it was an error, but if it is intentional then ignore my #2 request.
I can live with it that way, and it may be marginally better to stay there till clicked off.
But I think the LCD timeout define should be moved to a config file.

Kaibob2, 2017-02-28:
I think this was the moment the babystepping screen behaviour changed back then:
https://github.com/MarlinFirmware/Marlin/issues/5031#issuecomment-255520081

Roxy-3D, 2017-03-01:
I think it is better not returning to the status screen after 15 seconds.   I know a lot of people were having trouble with the timeout causing them to leave BabyStepping before they wanted to leave.   I used to just twitch the encoder knob one way and then the other to keep the time out from kicking me out.

And certainly...  It is easy enough to do one click when you want the status screen back.   After all, your fingers are already on the Encoder Wheel!

ruggb, 2017-03-01:
@Roxy-3D I agree. I only brought it up b/c it went away without warning so I thought it was an error.

Tannoo, 2017-03-02:
How about a short note at the bottom of the screen to tell the user `Click to return`? Not everyone is as keen on Marlin, nor active on git.

jgwinner, 2020-05-03:
I know this is closed, but I just ran into an issue where I couldn't do manual bed leveling as the LCD would go to the main screen before I had adjusted the bed - and I'm not that slow.
Oddly, I never had LCD_TIMEOUT_TO_STATUS set at all! 

I changed it to 100000 and now I can level the bed.

The weird thing is - it just happened recently and as near as I can tell nothing I changed should have affected the LCD timeout. I'd tried setting the Babystep Z double click, and then unset it; my other changes were about UBL. Not quite sure what I did to make the timeout so short!

Maybe this will help someone else.