MarlinFirmware/Marlin #5574: Menu Option to init SD Card

bvenneker, 2016-12-21:
Hi,
The menu option to init the sd card is available for boards that lack the sd_detect pin.
My ramps board has this pin but I use an SD extender (dummy sd card -> flat cable -> SD reader) which makes the sd_detect pin useless because the dummy sd card is never removed. Could we have an option in config.h to force the availability of the manual sd_init menu options?
For now I hacked the firmware a bit to make it work but an option in config.h would be nice.
Thanks for this great project!
Bart

manianac, 2016-12-22:
Think about it this way, you've essentially disabled your sd-detect pin, so rather than make a config option, just comment out `#define SD_DETECT` in your pins_ file.

Same effect, and no need to re-code anything.

bvenneker, 2016-12-23:
Thanks, yes, that makes it even easier. But still I would like to keep all changes in the config and config_adv so I don't need to keep track of changes I made to other files. 
For now, I am happy to leave it at this, but if more people like this idea then maybe it is something for the future.
Thanks Again,
Bart

manianac, 2016-12-24:
You should be able to put a `#undef SD_DETECT` statement in your config files for the same effect if you prefer.

thinkyhead, 2017-04-02:
Well, `#define SD_DETECT_PIN -1` actually….

I suppose we could have the "No SD card" menu item try to init SD when you click on it….

closed 2017-10-31 by thinkyhead