MarlinFirmware/Marlin #5576: error when compile 'millis' was not declared in this scope

MOglubalu, 2016-12-21:
anyone can help me?

when compile of firmware arduino send me this error message 

'millis' was not declared in this scope

how i can  solved this problem

thanks

Blue-Marlin, 2016-12-21:
No. Not without more information.

Marlin version/branch from?
Arduino IDE version?
Exact error - with file and line number?

MOglubalu, 2016-12-21:
`Atención: platform.txt del núcleo 'Arduino AVR Boards' contiene recipe.preproc.macros="{compiler.path}{compiler.cpp.cmd}" {compiler.cpp.flags} {preproc.macros.flags} -mmcu={build.mcu} -DF_CPU={build.f_cpu} -DARDUINO={runtime.ide.version} -DARDUINO_{build.board} -DARDUINO_ARCH_{build.arch} {compiler.cpp.extra_flags} {build.extra_flags} {includes} "{source_file}" desfasada, automáticamente convertida a recipe.preproc.macros="{compiler.path}{compiler.cpp.cmd}" {compiler.cpp.flags} {preproc.macros.flags} -mmcu={build.mcu} -DF_CPU={build.f_cpu} -DARDUINO={runtime.ide.version} -DARDUINO_{build.board} -DARDUINO_ARCH_{build.arch} {compiler.cpp.extra_flags} {build.extra_flags} {includes} "{source_file}" -o "{preprocessed_file_path}". Considera el actualizar este núcleo.
In file included from sketch\language.h:278:0,

                 from sketch\printcounter.h:27,

                 from sketch\Marlin.h:58,

                 from sketch\M100_Free_Mem_Chk.cpp:43:

sketch\language_es.h:182:0: warning: "MSG_INFO_PRINT_COUNT" redefined [enabled by default]

 #define MSG_INFO_PRINT_COUNT                "Impresiones"

 ^

sketch\language_es.h:177:0: note: this is the location of the previous definition

 #define MSG_INFO_PRINT_COUNT                "Conteo de impresion"

 ^

sketch\language_es.h:184:0: warning: "MSG_INFO_PRINT_TIME" redefined [enabled by default]

 #define MSG_INFO_PRINT_TIME                 "Total"

 ^

sketch\language_es.h:179:0: note: this is the location of the previous definition

 #define MSG_INFO_PRINT_TIME                 "Tiempo total de imp."

 ^

sketch\language_es.h:185:0: warning: "MSG_INFO_PRINT_LONGEST" redefined [enabled by default]

 #define MSG_INFO_PRINT_LONGEST              "Mas larga"

 ^

sketch\language_es.h:180:0: note: this is the location of the previous definition

 #define MSG_INFO_PRINT_LONGEST              "Impresion mas larga"

 ^

sketch\language_es.h:186:0: warning: "MSG_INFO_PRINT_FILAMENT" redefined [enabled by default]

 #define MSG_INFO_PRINT_FILAMENT             "Extrusion"

 ^

sketch\language_es.h:181:0: note: this is the location of the previous definition

 #define MSG_INFO_PRINT_FILAMENT             "Total de Extrusion"

 ^

In file included from sketch\language.h:278:0,

                 from sketch\printcounter.h:27,

                 from sketch\Marlin.h:58,

                 from sketch\thermistortables.h:26,

                 from sketch\Configuration.h:1483,

                 from sketch\MarlinConfig.h:30,

                 from C:\Users\EdyDaniel\Downloads\Marlin-RC8\Marlin\Marlin.ino:37:
[... 128 lines cut ...]

                                                              ^

In file included from sketch\MarlinSerial.cpp:33:0:

sketch\stepper.h: In static member function 'static short unsigned int Stepper::calc_timer(short unsigned int)':

stepper.h:315: error: 'MYSERIAL' was not declared in this scope

         MYSERIAL.print(MSG_STEPPER_TOO_HIGH);

         ^

exit status 1
'millis' was not declared in this scope`

Is the rc8 version
actually
Arduino ide 1.6.12

I have to say that I have not changed anything, it simply gives the error

I have tried with newer versions of the ide and older but do not compile with any

Yesterday I compiled without problems with 1.6.9 but now with none

MOglubalu, 2016-12-21:
Sorry i closed it by mistake... Simeone who can help me?

thinkyhead, 2016-12-23:
@MOglubalu Try `RCBugFix`. At the very least, the Spanish language file in `RC8` is messed up.

closed 2017-10-31 by thinkyhead