MarlinFirmware/Marlin #5578: HELP: Bed leveling too high on one side, too low the other

FancyChancie, 2016-12-22:
I have a Prusa i3 variant with inductive auto leveling. The print is not level on bed when I run G28, G29 then print. The issue is that 1st layer (and all following layers) are too high on the left and too low on the right (like not sticking on the left and digging into the bed on the right). Suspecting an issue with auto leveling firmware, I'm running Marlin 1.1.0-RC8. My other D-Bot printer is running 1.1.0-RC8 w/ auto leveling and has no issues. I have made sure the aluminum bed is perfectly flat what up to temp using a machinist straight edge and bed is perfectly level (it prints great when not using auto leveling). Suspecting the right side Z motor was skipping steps, I swapped drivers, pins in firmware, motor connectors to the board and even replaced the stepper motor with a brand new one (and same with the left side when the replacement made no change). I have also replaced both X and Y axis linear rods with new ones to remove any possibility of the rods being bent having effect on the issue. As well as attempted the 4 point (2x2) vs the 9 point (3x3) auto leveling grid.

- link to my current firmware version [**Link**](https://www.dropbox.com/sh/a26sbdzay9ztcd8/AACjTC_4hE5ceVdfkzkpG6T_a?dl=0)

- pics of printer [**Link**](http://imgur.com/a/ysdkW)

- pics of print with auto leveling and without auto leveling  (the pics don't really do it justice) [**Link**](http://imgur.com/a/is6BW)

brainscan, 2016-12-22:
Have you used the fade mesh option before with success? I tried switching it on and it destroyed one of my nozzles and gouged a line in my bed. Switching it off all worked fine. I've been so busy I haven't been able to try it again to see if it's a bug.

landodragon141, 2016-12-22:
I know this may be a stupid question but, have you verified that the X axis is level by running it all the way to the top before you run calibration?

FancyChancie, 2016-12-22:
@brainscan I haven't used mesh leveling yet, just started looking into it last night after I posted the issue. But from your experience, I'm a little worried. Lol. I'll do some research on it and see if its something I want to set up, however I'd most like to get this issue with my auto leveling figured out 1st.

@landodragon141 Yeah, I have made sure the X axis are perfectly level with the bed. I use a depth gauge to verify they are absolutely perfect. Here are some pics of how it works. [**Link**](http://imgur.com/a/Vipyz)

manianac, 2016-12-22:
I saw your post on how you level your X axis, but I'm just curious if the probe sees the bed as flat or if maybe the inductive probe engages sooner on another part of the bed.  
Something I always do to check my X axis is: 
```
G28
G1 X20 Y[max/2] F9000
G30
G1 X[max] Y[max/2] F9000
G30
```
and compare the two Z readings.

Other than that, if you do a `M420 V` on the latest RCBugFix, we can see your ABL grid.  Otherwise copy/paste your grid after a `G29`.

Blue-Marlin, 2016-12-23:
I almost all cases when Marlin is compensating z to the wrong direction:
* The printers axis don't build a 'right handed coordinate system'
* or the x axis is twisted so that the probe is lowed, or raised at the one or the other side, depending on if the probe is before or behind of the twisting point of the x axis and the direction of twist.

brainscan, 2016-12-23:
Sorry my mistake I must have misread your config, I'll try again.

NiklasOlofsson, 2016-12-23:
I have the same problem. Too low on the left side makes the extruder motor loose steps because too close to the print bed (both in RC8 and RCBugFix-branch). I can't for my life make it work. It does somewhat work if rotating the left stepper motor by hand after ABL is done. Using BLTouch. Yesterday I measured all the rods to make sure everything is correct. I had one driver for my two Z-motors but changed to 2 drivers yesterday. Is it possible to add or subtract movement for one of the motors? ;-)

manianac, 2016-12-24:
@FancyChancie is not using dual endstops, just dual Z drivers.  What about you @NiklasOlofsson?

NiklasOlofsson, 2016-12-24:
I'm back to one driver, two motors. My Z endstop is the probe. I have fiddled with the config for hours now. It's better but not good enough. I use heated glass, could it be warped? Only having problem on bigger prints.

`03:12:46.375 : Bilinear Leveling Grid:
03:12:46.375 : 0     1     2     3
03:12:46.375 : 0 +0.58 +0.52 +0.73 +0.76
03:12:46.375 : 1 +0.18 +0.20 +0.42 +0.43
03:12:46.375 : 2 -0.15 -0.13 +0.10 +0.15
03:12:46.375 : 3 -0.53 -0.48 -0.17 -0.10
`

My "bed level correction matrix" is close to others I find when I Google it. Hm.

manianac, 2016-12-24:
I wouldn't say warped, just uneven.  But since you're using one driver, I doubt Marlin could've had any effect on this, your's simply looks like a hardware issue, such as what @Blue-Marlin explained or just simply a tilted bed.

FancyChancie, 2016-12-24:
@NiklasOlofsson I can get mine to print perfectly as well if I pause the print and adjust one of the motors by hand.

@manianac Here is the results of my G29
          Bed Level Correction Matrix:
          +0.999998 +0.000000 -0.002028 
          +0.000001 +1.000000 +0.000426 
          +0.002028 -0.000426 +0.999998

manianac, 2016-12-24:
I assume you're using 3 point or linear leveling. Have you tried bilinear yet to see if maybe your bed isn't flat? Linear handles tilting,  but not much else.

FancyChancie, 2016-12-24:
I haven't yet, I'll set that up right now and see what I get then post it.

FancyChancie, 2016-12-24:
Okay here's the results:
**Bilinear Leveling Grid:**
         _0        1          2
0  -0.04  -0.07  -0.33
1  +0.02  +0.01  -0.29
2  +0.04  -0.00  -0.27_

manianac, 2016-12-24:
Ok that's weird. Probably not enough data to state for certain,  but the whole side opposite your xmin endstop is reading lower. Increase the able max grid size,  5 or 6 should be enough,  and see what happens then. More data helps

FancyChancie, 2016-12-24:
I had thought that maybe I should increase the grid. Oh well. Here is a larger gird.

**READ: Bilinear Leveling Grid:**
_0     1     2     3     4     5
0 -0.04 -0.06 -0.05 -0.10 -0.20 -0.36
1 -0.00 -0.02 -0.03 -0.09 -0.21 -0.33
2 +0.01 -0.01 +0.00 -0.03 -0.15 -0.31
3 +0.02 +0.01 -0.00 -0.07 -0.18 -0.30
4 +0.02 +0.01 +0.02 -0.02 -0.13 -0.29
5 +0.03 +0.02 +0.02 -0.04 -0.16 -0.27_

manianac, 2016-12-24:
Yea it's like @blue-marlin said. Either your bed actually has a bend in it,  or your x axis is twisting and moving the probe height. Given how it's a steady rise after column 3, I'd investigate your x axis. Definitely not a firmware issue.

FancyChancie, 2016-12-24:
Okay. Thanks for taking a look at everything I have going on here @manianac 
This might be a stupid question, but how would I go about checking if the X axis is "twisting"? What could be causing X axis twisting?

manianac, 2016-12-24:
Bent hardware. Try the old paper under the nozzle at different points on your bed. If your probe is changing height when moving on the x axis,  I'd assume your nozzle is too

FancyChancie, 2016-12-25:
I've made some adjustments that have helped. Once I've got it 100% (in the next few days), I'll post a complete summary of my resolution then close this issue (in the future it may be of use to someone with similar issues).

NiklasOlofsson, 2016-12-26:
If I set ABL_GRID_MAX_POINTS_X to 4, the probing will start in the right bottom corner instead of the left front as with max points set to 3. Debugging and looking at the output I can see that the grid is inverted.

manianac, 2016-12-26:
ABL is set up to always end opposite of (0,0) if i recall right.  I believe what you're describing is normal.

NiklasOlofsson, 2016-12-26:
It does end in the right place. The starting point changes.

NiklasOlofsson, 2016-12-26:
But still looks inverted(?). Here I made the right front corner higher:

> Bilinear Leveling Grid:
> 0     1     2
> 0 -0.45 -0.31 +3.33
> 1 -0.27 -0.10 +0.15
> 2 -0.08 +0.19 +0.53

manianac, 2016-12-26:
When you say right front, where is that in relation to your home?

NiklasOlofsson, 2016-12-26:
Home at X0 Y0 is left front.

manianac, 2016-12-26:
So then it is working.  (0,0) is home, you lifted the end opposite of home on the X axis.  Looking at your grid (2,0) is lifted.

NiklasOlofsson, 2016-12-26:
This is what I did:

[image]

manianac, 2016-12-26:
Ok.  You said home is at the Left Front.  You put a spacer on the Right Front.  It measured that and reported it to you.

I'm sorry, I'm not understanding your concern.

NiklasOlofsson, 2016-12-26:
It was more like a question. I'm really new to this 3D printing world. So, it looks right then. ;-)

manianac, 2016-12-26:
Ah alright.  Yes its working fine, the table's upper left (0,0) is your home, then follow it out from there.

NiklasOlofsson, 2016-12-26:
Ok, thanks!

How about this one, how does it look:

> 04:58:57.421 : Bed Height Topography:
> 04:58:57.421 : +--- BACK --+
> 04:58:57.421 : |           |
> 04:58:57.421 : L |    (+)    | R
> 04:58:57.421 : E |           | I
> 04:58:57.421 : F | (-) N (+) | G
> 04:58:57.421 : T |           | H
> 04:58:57.421 : |    (-)    | T
> 04:58:57.421 : |           |
> 04:58:57.421 : O-- FRONT --+
> 04:58:57.421 : (0,0)
> 04:58:57.421 : -0.35082 -0.17309 +0.02535
> 04:58:57.421 : -0.51176 -0.37387 -0.23364
> 04:58:57.421 : -0.74457 -0.66059 +3.02300
> 04:58:57.421 : Corrected Bed Height vs. Bed Topology:
> 04:58:57.422 : +1.75214 +1.19292 +0.65440
> 04:58:57.422 : +1.23849 +0.63943 +0.04271
> 04:58:57.422 : +0.65297 +0.00000 +2.94664
> 04:58:57.422 : Bed Level Correction Matrix:
> 04:58:57.422 : +0.999969 +0.000000 +0.007924
> 04:58:57.422 : +0.000040 +0.999987 -0.005039
> 04:58:57.422 : -0.007924 +0.005039 +0.999956
>

NiklasOlofsson, 2016-12-26:
Btw, G29 V4 T does not give me the topology report or bed level correction matrix anymore. Just getting the "Bilinear Leveling Grid" output. What have I messed with?

manianac, 2016-12-26:
@NiklasOlofsson, your Topology report is correct, opposite of 0,0 is the +3.023 raised portion you put in.
You should read the config file on the differences between linear and bilinear.  

Topology report and the correction matrix are for the linear grid only, as you're trying to tilt a solid plane.

Bilinear doesn't have these because the topology report _**is**_ the bilinear leveling grid.  And there is no correction matrix because you are not just tilting the bed, you are adjusting for any irregularities in the flatness.

Unless you have a really good reason (milled bed with no surface irregularities, or printing bare glass with nothing on it), just use bilinear and you should be well off.

djuniversal, 2016-12-28:
I'm having the same issue on RC8. When I issue a G29 command prior to printing, I have the nozzle height to high on the right side and to low on the left

djuniversal, 2016-12-28:
Also, when I do G29 V4 T I do not get any topology report.

FancyChancie, 2017-01-04:
So the main issue that I was having with mine was the X axis was twisted and not at a perfect 90°. Using a digital angle gauge I found that the side that was digging into the bed (the right side) was at 88.6°. After a lot of tweaking and making sure everything that I could possibly square up was square. It prints fine now.

### Here is my before adjusting the X axis angles.
**READ: Bilinear Leveling Grid:**
_0 1 2 3 4 5
0 -0.04 -0.06 -0.05 -0.10 -0.20 -0.36
1 -0.00 -0.02 -0.03 -0.09 -0.21 -0.33
2 +0.01 -0.01 +0.00 -0.03 -0.15 -0.31
3 +0.02 +0.01 -0.00 -0.07 -0.18 -0.30
4 +0.02 +0.01 +0.02 -0.02 -0.13 -0.29
5 +0.03 +0.02 +0.02 -0.04 -0.16 -0.27_

### Here is my after adjusting the X axis angles.
**READ: Bilinear Leveling Grid:**
_0 1 2 3 4 5
0 -0.01 +0.01 +0.07 +0.09 +0.04 -0.07
1 +0.01 +0.05 +0.10 +0.10 +0.02 -0.06
2 +0.03 +0.04 +0.12 +0.12 +0.10 -0.05
3 +0.01 +0.06 +0.11 +0.11 +0.05 -0.05
4 +0.00 +0.04 +0.12 +0.12 +0.10 -0.03
5 +0.01 +0.05 +0.11 +0.10 +0.09 -0.00_

I do still have a slight bow in the center of the bed, but with Bilinear leveling, it seems to be doing just fine.

I am REALLY appreciative of @manianac for helping me so much, your good deed does not go unnoticed nor should your skill and time be uncompensated for, hit me up on [FB](https://www.facebook.com/fancychancie) or shoot me an [email](mailto:chance1616@gmail.com) I will send you something via PayPal. Thank you again, you revived my 3D printing spirit.

manianac, 2017-01-06:
Rock on man, thanks for sharing your fix; glad it worked.  If you don't mind, don't forget to close the issue.

closed 2017-01-06 by FancyChancie

eduardoscamargo, 2017-09-11:
Hey @FancyChancie, I am having the same problem. What did you mean when you said your X axis was twisted. Did you mean one side of the X axis was closer to the bed than the other one?

FancyChancie, 2017-09-11:
@eduardoscamargo I have actually gotten a few direct e-mails with this question. I have made a video descriptively explaining what I meant by "twisted" and have uploaded it [here](https://youtu.be/E1NUSAsvPdA) for anyone who may be having the same struggle I was. [CLICK HERE TO GO TO EXPLANATION VIDEO](https://youtu.be/E1NUSAsvPdA)

eduardoscamargo, 2017-09-11:
@FancyChancie thank you!

adamsd5, 2017-09-16:
Excellent video, btw.  I'm going to check my printer today.  I have one quadrant I must avoid on my bed as the prints go too high in that corner.  I can't seem to adjust anything to get it right.  I use AUTO_BED_LEVELING_BILINEAR.  I'm confident my bed is not perfectly square to the axes (it is likely tilted).  I thought that's what auto-leveling is supposed to do.  Maybe a twist in the X axis would cause the leveling readings to be off for that corner (or worst in that corner).

I had another thought today.  Is there a way I can hard-code an adjustment factor in the configuration settings?  If there are 9 touch points, can I add an artificial Z-correction for each point, and tune it that way?  (Sort of like the Z-offset setting in my slicer software, but for each of the 9 points.)  Does anyone think this is a good idea?  Does it already exist?

adamsd5, 2017-09-16:
Just learned a lot about UBL... trying it out.  My next question... once I have it dialed in, what should I use to start each print.  Currently I have a G28 followed by a G29.  Should I use G29 P1 instead?

Roxy-3D, 2017-09-16:
> Currently I have a G28 followed by a G29. Should I use G29 P1 instead?

The answer depends on whether you choose to have the printer power up with UBL enabled or not.   I activate UBL and then save the mesh and do a M500.   At that point, every time the printer powers up, UBL automatically loads the last saved mesh and is enabled.   I do not have a G29 in the startup GCode of my prints.

But if you want to be more traditional, you can have UBL disabled when the printer powers up.  You can then add a few lines to your Slicer's start up GCode:
- G28
- G29 L 1 ;to load the mesh from EEPROM slot 1
- G29 A ;activate UBL