MarlinFirmware/Marlin #5584: Auto bed leveling with 0-4v analog sensor

silentninja1, 2016-12-23:
I came across a cmos laser sensor from a scrapped industrial machine with a 0-4v analog output for 20-45mm range. 2um repeatability, .1% linearity that operates at 10-30vdc. I figure this is a perfect non-contact any surface z-probe if I can read an analog value and set a window for home pos.

Sensor is a Keyence IA-030 with an IA-1000 amplifier.

Has anyone tried to use an analog sensor and value comparison before? I suppose the only other option with this would be to obtain a reference voltage and a comparitor, but thatd be difficult to tune with a .2mv analog repeatability.

Blue-Marlin, 2016-12-23:
You could try it, but the ADCs are only checked ~ 5 times per second - asynchronous to the stepper interrupts.
So if your probing speed is faster then ~2.5 steps per second you will lose precision. That's extremely slow!

silentninja1, 2016-12-23:
That is slow... I guess it would work best with some code changes to move quickly until the analog input value changes (eg we are now within the sensor range, in my case 45mm) then either use a calculable scaling factor to translate the analog input value to a position and set it as my z height, Ive done that in PLC projects with offset calculations detecting parts and focus adjustments for laser etching.

Barring that, a slowdown of the probe speed if value is between X and Y would be the next best bet, or a define switch to increase the scan time of the analog inputs. This sensor can be set between a sampling of 0.33-5ms. 
Are any of these feature / functionality changes that would be of value to the base project rather than a one off?

manianac, 2016-12-23:
Holy crap thats a nice sensor.  You may want to consider using something like what [David Crocker did with his differential infrared sensor](https://miscsolutions.wordpress.com/mini-height-sensor-board), and use something else to switch the endstop input for Marlin (He used an ATTINY25).  That way you wouldn't have to rely on the slow ADC checks, and could set the threshold of when it switches.

thinkyhead, 2017-03-04:
Any luck?

closed 2017-03-04 by thinkyhead