MarlinFirmware/Marlin #5586: Mixing Extruder woes: E2 (3rd extruder) is giving out incorrect steps.

jwcrawley, 2016-12-23:
When you enable Mixing Extruder, with 1 extruder, distinct e factors, 3 mixing steppers and the appropriate e factors for the extra steppers, what happens is the following:

Advance 5mm: E0 goes 5mm. E1 goes 5mm. **E2 goes 30mm**. <facepalm>

Hardware: 3 of the exact same motors. Switching them around between E0 through E2 shows the motors are correctly wired, and give correct steps IF in E0 or E1. 
_________________

I have tried DISTINCT_E_FACTORS=ON and OFF. I have tried permutations of including and excluding E1 and E2 settings further down in the config. All permutations I have tried to change E2 have done nothing to solve said issue.

E1 appears to follow the esteps set, as does E0. E1 **appears** to take from E0's settings if distinct_e_factors is NOT enabled. E2 does its own merry thing.

As of right now, Mixing extruders are unusable.

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/671272/Marlin.zip)
Included configuration, configuration_adv and pins_ramps.

jwcrawley, 2016-12-28:
Is there anything else I need to supply to have someone look at this? I would be willing to assist someone with what's going on here.

thinkyhead, 2017-04-02:
Apparently it's just me looking through the issues, and when I'm away… nothing but crickets.

Have you tested a recent `RCBugFix` and found it working, or is this still unresolved?

baghtal, 2017-05-11:
I looked though, but all my ideas were self-defeating... Are your drivers all the same?

Bob-the-Kuhn, 2017-05-11:
Please try 1.1.0 or bugfix -1.1x.  I know they've done some corrections on the distinct e-factors and extruder stepper issues in the April timeframe.

closed 2018-02-06 by thinkyhead