MarlinFirmware/Marlin #5590: Z Stop homing issue

NickLangstone, 2016-12-24:
Hi
I have just started using marlin so am very new to it and 3d printing - but have read the comment on on several other tickets including #5564  which appears similar but I am not able to determine if there is a resolution.
20:34:27.207 : FIRMWARE_NAME:Marlin 1.1.0-RC8 (Github) SOURCE_CODE_URL:https://github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:Tarantula EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff
20:34:27.221 : echo:DEBUG:INFO,ERRORS
20:34:27.222 : echo:Active Extruder: 0

When I issue an M119 command I get the following output
20:26:49.912 : Reporting endstop status
20:26:49.913 : x_min: open
20:26:49.913 : y_min: open
20:26:49.913 : z_min: TRIGGERED

But the Z min switch is not triggered - (non are ) - if I manually depress Z-Min and run M119 - it reports
20:35:55.529 : Reporting endstop status
20:35:55.529 : x_min: open
20:35:55.529 : y_min: open
20:35:55.529 : z_min: open

So this tells me the switch is working correctly - but is being interpreted incorrectly

I have tried altering the settings of   Z_MIN_ENDSTOP_INVERTING   to both true and false - but observe NO CHANGE IN BEHAVIOR. This confuses me.
// Mechanical endstop with COM to ground and NC to Signal uses "false" here (most common setup).
#define X_MIN_ENDSTOP_INVERTING true // set to true to invert the logic of the endstop.
#define Y_MIN_ENDSTOP_INVERTING true // set to true to invert the logic of the endstop.
#define Z_MIN_ENDSTOP_INVERTING false // set to true to invert the logic of the endstop.
#define X_MAX_ENDSTOP_INVERTING true // set to true to invert the logic of the endstop.
#define Y_MAX_ENDSTOP_INVERTING true // set to true to invert the logic of the endstop.
#define Z_MAX_ENDSTOP_INVERTING false // set to true to invert the logic of the endstop.

Because I do not have a probe (yet - need to print a bracket first .... but need z home to work ....)
It appears the prob config is messing with the z_min_endstop config

How can I force the config to invert the Z Endstop ??
I have looked at the code and there are a lot of different configs that seem to overload the z_stop in different ways for different probes ....

any help would be appreciated

[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/671689/Configuration.txt)
Config file attached)

Thanks

Nick

brainscan, 2016-12-24:
I would re check the way you have it wired first. If you're using mechanical endstops on x and y and you've had to invert the logic on those so they work then that isn't what's recommended. Try switching the z endstop for one of the others and have them all wired exactly the same and with the same logic in Marlin. You also have bltouch defined but you've said you don't have a probe, I'm not sure if that would cause a problem but maybe best to comment it out until you have/use one.

mperdue, 2016-12-24:
With bltouch defined Z_MIN_ENDSTOP_INVERTING will be set to false as this is the default setting for the bltouch probe. Comment out the bltouch define and give it a shot.

NickLangstone, 2016-12-26:
Thanks Mqerdue - yes the bltouch was the problem - once commented this out the endstop behaved correctly
Thanks

closed 2016-12-26 by NickLangstone