MarlinFirmware/Marlin #5592: Mesh Bed Leveling - RCBugFix (24/12/2016)

psavva, 2016-12-24:
When going to Prepare - Level Bed.
The Z lifts, and you expect it to start homing.

The LCD says (Homing XYZ), and it just stays there.
No movement, No Homing.

The revision of RCBugFix is 922c67f17d5052f499f331f9702559799c6a1b0c

psavva, 2016-12-24:
Configuration: http://pastebin.com/V9kpnShH

epatel, 2016-12-24:
Sorry to hear that. 
When looking though the commit history (https://github.com/MarlinFirmware/Marlin/commits/RCBugFix) my first guess would be this commit https://github.com/MarlinFirmware/Marlin/commit/d0e24e0876c845c764e921287afce521966ce6f0
So maybe try to narrow down which commit that caused this new behaviour. Does that commit experience the same behaviour and the commit before that https://github.com/MarlinFirmware/Marlin/commit/2dbd38ce166257ac8fb61b4ff048dfc482c1f68d does not it is a clue where a bug got introduced.
Another method to find a bug that got introduced is to do a binary search, i.e. go back say about 120 commits and see if the bug is there. If its not, select the commit 60 steps back and then in a binary search way narrow down to the commit that introduced the bug. 7 tests should be sufficient to find a commit in about 120 commits (7 bits -> 128).
Good luck and Merry Christmas! :)

brainscan, 2016-12-24:
Does mesh levelling work with a bltouch?

manianac, 2016-12-24:
@brainscan, Mesh leveling is meant for when you don't have a sensor, and you adjust each point manually (checking nozzle height).

If you have a sensor such as a BLTouch, use auto bed leveling instead.  Bilinear leveling is what you're looking for to get the same results as mesh bed leveling.

psavva, 2016-12-24:
@maniac,
Actually, mesh bed leveling is supposed to compensate for an uneven bed on multiple planes.

My bed is 400x380, and I'm using normal 6mm glass sheet, not borosilicate. (Extremely expensive in my country)

So attempt 1 for me is to use a not so perfect glass window, which costs 5 EUR. 

The BLTouch is used to probe the z height.  It was working before on RC 8, but with a different bug...

manianac, 2016-12-24:
Yes @psavva, but Bilinear auto bed leveling can do the same thing using your BLTouch.  Mesh bed leveling is just doing the process manually.  Read the configuration notes to see the differences.

@psavva I would recommend you use auto bed leveling with your BLTouch, but if you still choose to use mesh bed leveling, then enable `DEBUG_LEVELING_FEATURE` in Configuration.h, and issue a `M111 S35 to output debugging over the serial connection.  You will need to hook your machine up to a computer of course.

brainscan, 2016-12-24:
Yeah I've been using it since it was first added I know how it works. I was asking if things had changed and if it could be used with a probe now as I thought that was in Roxy's UBL but not in RCBUGFIX yet.

psavva, 2016-12-24:
They is great, I'll enable Bilinear Bed Leveling then.
However, still need to find the issue so we can fix it.

I'll have to test it after Christmas and see if I can find the root cause.

manianac, 2016-12-25:
Sounds good.  Enable debugging to help that process.

Fixx, 2016-12-25:
I experience the same issue with RCBugfix dated 15/12/2016.

Tannoo, 2016-12-27:
Same issue here today. I'm not sure why it's waiting or what it's waiting for, yet.

Tannoo, 2016-12-27:
This is not an issue? Looks like it will just wait forever.

`while (!axis_homed[X_AXIS] || !axis_homed[Y_AXIS] || !axis_homed[Z_AXIS]) idle();`

Tannoo, 2016-12-27:
If I do this:

`  inline void lcd_wait_for_homing() {
    no_reentrance = true;
    while (!axis_homed[X_AXIS] || !axis_homed[Y_AXIS] || !axis_homed[Z_AXIS]); // idle();
    no_reentrance = true;
  }`

It doesn't wait forever....but Marlin resets 5 secs or so after raising z. Although, nothing homes..

Tannoo, 2016-12-27:
In this function:

`    void _lcd_level_bed_continue() {
      defer_return_to_status = true;
      axis_homed[X_AXIS] = axis_homed[Y_AXIS] = axis_homed[Z_AXIS] = false;
      mbl.reset();
      enqueue_and_echo_commands_P(PSTR("G28"));
      lcd_goto_screen(_lcd_level_bed_homing);
    }`

The serial monitor shows: echo:enqueueing "G28"
And because of the 'idle()' command previously commented out, there is one other "blip" on the serial monitor.
It displays: 

start

and the system is restarted.

Tannoo, 2016-12-27:
Here is a snipplet from my serial monitor:

`echo:Marlin 1.1.0-RCBugFix

echo: Last Updated: 2016-12-06 12:00 | Author: (Tanno, RCbugFix-DHT config)
Compiled: Dec 27 2016
echo: Free Memory: 3348  PlannerBufferBytes: 1232
echo:Hardcoded Default Settings Loaded
echo:Steps per unit:
echo:  M92 X80.00 Y80.00 Z3840.00 E90.00
echo:Maximum feedrates (mm/s):
echo:  M203 X250.00 Y250.00 Z2.00 E22.00
echo:Maximum Acceleration (mm/s2):
echo:  M201 X1000 Y1000 Z5 E1000
echo:Accelerations: P=printing, R=retract and T=travel
echo:  M204 P500.00 R500.00 T500.00
echo:Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
echo:  M205 S0.00 T0.00 B20000 X15.00 Y15.00 Z0.40 E5.00
echo:Home offset (mm)
echo:  M206 X0.00 Y0.00 Z0.00
Mesh Bed Leveling:
echo:  M420 S0
echo:  G29 S3 X1 Y1 Z0.00000
echo:  G29 S3 X2 Y1 Z0.00000
echo:  G29 S3 X3 Y1 Z0.00000
echo:  G29 S3 X1 Y2 Z0.00000
echo:  G29 S3 X2 Y2 Z0.00000
echo:  G29 S3 X3 Y2 Z0.00000
echo:  G29 S3 X1 Y3 Z0.00000
echo:  G29 S3 X2 Y3 Z0.00000
echo:  G29 S3 X3 Y3 Z0.00000
echo:Material heatup parameters:
echo:  M145 S0 H195 B50 F0
  M145 S1 H220 B110 F0
echo:PID settings:
echo:  M301 P18.64 I1.02 D85.18
echo:  M304 P358.06 I49.23 D651.00
echo:Filament settings: Disabled
echo:  M200 D3.00
echo:  M200 D0
echo:SD card ok
echo:SD card ok
echo:enqueueing "G28"
start
echo:Marlin 1.1.0-RCBugFix

echo: Last Updated: 2016-12-06 12:00 | Author: (Tanno, RCbugFix-DHT config)
Compiled: Dec 27 2016
echo: Free Memory: 3348  PlannerBufferBytes: 1232
echo:Hardcoded Default Settings Loaded
echo:Steps per unit:
echo:  M92 X80.00 Y80.00 Z3840.00 E90.00
echo:Maximum feedrates (mm/s):
echo:  M203 X250.00 Y250.00 Z2.00 E22.00
echo:Maximum Acceleration (mm/s2):
echo:  M201 X1000 Y1000 Z5 E1000
echo:Accelerations: P=printing, R=retract and T=travel
echo:  M204 P500.00 R500.00 T500.00
echo:Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
echo:  M205 S0.00 T0.00 B20000 X15.00 Y15.00 Z0.40 E5.00
echo:Home offset (mm)
echo:  M206 X0.00 Y0.00 Z0.00
Mesh Bed Leveling:
echo:  M420 S0
echo:  G29 S3 X1 Y1 Z0.00000
echo:  G29 S3 X2 Y1 Z0.00000
echo:  G29 S3 X3 Y1 Z0.00000
echo:  G29 S3 X1 Y2 Z0.00000
echo:  G29 S3 X2 Y2 Z0.00000
echo:  G29 S3 X3 Y2 Z0.00000
echo:  G29 S3 X1 Y3 Z0.00000
echo:  G29 S3 X2 Y3 Z0.00000
echo:  G29 S3 X3 Y3 Z0.00000
echo:Material heatup parameters:
echo:  M145 S0 H195 B50 F0
  M145 S1 H220 B110 F0
echo:PID settings:
echo:  M301 P18.64 I1.02 D85.18
echo:  M304 P358.06 I49.23 D651.00
echo:Filament settings: Disabled
echo:  M200 D3.00
echo:  M200 D0
echo:SD card ok
echo:SD card ok
`

Tannoo, 2016-12-27:
I comment out this line:

`      //lcd_wait_for_homing();
`
Added this line from RC:

`      if (axis_homed[X_AXIS] && axis_homed[Y_AXIS] && axis_homed[Z_AXIS])
`

And I have bed leveling again!! WooHoo!!

Tannoo, 2016-12-27:
I don't know what the REAL fix is for this is but, this fixed it for me.

Tannoo, 2016-12-27:
`    void _lcd_level_bed_homing() {
      if (lcdDrawUpdate) lcd_implementation_drawedit(PSTR(MSG_LEVEL_BED_HOMING), NULL);
      lcdDrawUpdate =
        #if ENABLED(DOGLCD)
          LCDVIEW_CALL_REDRAW_NEXT
        #else
          LCDVIEW_CALL_NO_REDRAW
        #endif
      ;
      if (no_reentrance) return;
      //lcd_wait_for_homing();
      if (axis_homed[X_AXIS] && axis_homed[Y_AXIS] && axis_homed[Z_AXIS])
      lcd_goto_screen(_lcd_level_bed_homing_done);
    }`

Tannoo, 2016-12-27:
This is also going to be in my PR for the DHT sensor. Only because it works.

psavva, 2016-12-27:
@Thinkyhead, can you review and comment on the suggested fix?

Tannoo, 2016-12-27:
I will say that the LCD responsiveness is much faster than the RC.

Blue-Marlin, 2016-12-27:
The real fix is to home the machine before doing any moves.

Tannoo, 2016-12-27:
Raising Z before homing is working. The problem was that it was not homing any axis. It would raise Z and stop.

Tannoo, 2016-12-27:
I do have this issue now..

      if (axis_homed[X_AXIS] && axis_homed[Y_AXIS] && axis_homed[Z_AXIS])
      lcd_goto_screen(_lcd_level_bed_homing_done);

This doesn't update the lcd until I move the encoder or click the button.

manianac, 2016-12-27:
I'd say the real fix is looking into lcd_wait_for_homing and determine why
it's not working

epatel, 2016-12-28:
Interesting findings. Sry I can't test myself. Well, according to the commit history the lines you have changed recently was introduced into this commit https://github.com/MarlinFirmware/Marlin/commit/b4dbf4d18a03029f44217e73b7dbe8c7b5123a91
So, a test to do would to try the commit just before that, this commit https://github.com/MarlinFirmware/Marlin/commit/7b625e0048c7041ca1b9cbcb62c4f98c64a49ce6

Tannoo, 2016-12-28:
@manianac 
> I'd say the real fix is looking into lcd_wait_for_homing and determine why
it's not working

Well, here's the function:

`  inline void lcd_wait_for_homing() {
    no_reentrance = true;
    while (!axis_homed[X_AXIS] || !axis_homed[Y_AXIS] || !axis_homed[Z_AXIS]) idle();
    no_reentrance = true;
  }`

I only see it sitting forever doing nothing `idle()` if any of the axis are not homed `!axis_homed`.

Tannoo, 2017-01-01:
I put in a PR. (#5634)

closed 2017-02-06 by psavva