MarlinFirmware/Marlin #5595: HELP!!! Nothing will upload to Mega 2560

MH15, 2016-12-25:
So I've had a pretty good time uploading firmware to the arduino so far but today when I was planning on changing the steps per mm on the extruder, I never got that far cause I could never upload. Here is the error arduino gav me:

Roxy-3D, 2016-12-25:
It may be you have some other task that has already opened the serial port.   For example, if you use PronterFace, and have been using it to talk to the AVR board.   Only one program can have the serial port open at a time.  And which ever program tries to open the serial port when another program has it open will fail.

Tannoo, 2016-12-25:
Selected wrong serial port? Plugged into a different USB port on the PC? Also, what Roxy said.

psavva, 2016-12-28:
Restart your PC and open the Arduino IDE and try again. 

Don't open other programs such as Cura that might open the com port.

closed 2017-02-12 by thinkyhead

matskat, 2019-08-25:
I have the same issue. I bet it's Cura running thats the issue. But Cura is ALWAYS running LOL! Guess I need to use a different PC or wait until this print is complete before trying to set up my ATMEGA2560 as an ISP. Good times.

MH15, 2019-08-30:
Close Cura (from the Task Manager if necessary) then restart. You may have to turn off Cura's autostart ability. You can do this in Task Manager as well.

If you have a print running then oof not sure what to do. This is one of the reasons why it's often better to print standalone.