MarlinFirmware/Marlin #5597: Dual-X-Carriage, G29, T1-Z-Offset Problem

gsart84, 2016-12-25:
My Custom Dual-X-Carriage printer has a BL-Touch mounted on T0.
The M851 Z-Offset is Z-1.51
The Z-offset for T1 M218 Z+0.5 .
The Nozzles are cant be adjustet in the Z position. so the T1 nozzle are 0.5mm higher then t0 

if i home with G28 and print with T0 or T1, all works perfect.
if i do G28 and then G29, T0 works Perfect but T1 are liftet up 0.5mm.
to me it seems G29 do not work on T1.

has anyone else this problem?

or is there anything i am doing wrong?

greets

thinkyhead, 2017-04-02:
Sorry no one has attended to this issue. What is the state?

christianh17, 2017-04-02:
This bug was solved, around 2 weeks ago. There were two patches for it, the last one: 421468a (it had another improvement, too). So I think this issue can be closed.

closed 2018-02-06 by thinkyhead

braga-dev, 2018-06-15:
@christianh17 the commit you referenced is not the solution for this since it solves the problem for the switching extruder but not for the dual x_carriage. I have been working in a dual x-carriage printer but im having problems using G29 since que behaviour of the second carriage is strange. G29 starts with the last point of probing on the right side of the bed using the second carriage.

thinkyhead, 2018-06-15:
Sounds like we should patch `G29` so it always starts with `tool_change(0)` on Dual X Carriage.

braga-dev, 2018-06-20:
@thinkyhead i just tried 3bb332b4405a86a28621b77b10739af111b57dee and tool 1 is still probing before tool 0, and is not probing on the RIGHT_PROBE_BED_POSITION defined in configuration file.

thinkyhead, 2018-06-21:
Weird. So it doesn't do the tool change back to 0 right at the start of `G29`?

braga-dev, 2018-06-22:
Pretty weird! No, it doesnt it just probes with tool0 in the home position and then it moves to the tool 1 and starts probing in the wrong position

thinkyhead, 2018-06-28:
Hmm, will need to keep experimenting as time allows…