MarlinFirmware/Marlin #5599: Assistance Request - Bilinear ABL detecting bed levels but no Z-Axis movement during print

sstjean, 2016-12-25:
Hi,

I'm having an issue with Bilinear ABL on my printer and I was hoping that someone here would be kind enough to tell me where I'm going wrong.  Lots of other folks seem to be able to get this working and I've been trying for a few days now but I can't seem to find the right mix of configuration settings.  

Any help would be greatly appreciated.  Thanks in advance and please let me know if you need any more information or picture (or video).

I'm currently working with an[ Alunar M508](http://www.alunar.net/product/diy-3d-printer-kit-m508/) (Prusa i3) Cartesian printer and have tried to get ABL running with both the RC8 and RCBugFix branches with no success.  All of the configuration and attached data are from the RCBugFix branch as of 25-Dec-2016.  I have an inductive sensor on the Z axis and have Auto Bed Leveling tuned on. I have selected Bilinear ABL.  My printer is configured so that homing puts the head at the front-left corner of the print bed.  

I have intentionally skewed the bed upward at the front-right corner to test whether the z-axis is moving during printing.

After uploading the firmware (RCBugFix) I connected with Pronterface and sent a G28 to home the unit and then a G29.  I have DEBUG_LEVELING_FEATURE turned on and received the following table:
```
Bilinear Leveling Grid:
      0     1     2     3     4
 0 -0.00 +0.24 +0.57 +1.00 +1.45
 1 -0.20 +0.04 +0.39 +0.81 +1.25
 2 -0.40 -0.15 +0.19 +0.59 +1.01
 3 -0.61 -0.36 -0.04 +0.39 +0.80
 4 -0.85 -0.60 -0.27 +0.12 +0.53 
```
To test this I sent a print of a 20mmx20mm test cube with skirt to my printer from Pronterface, Cura 15.04.4 and from the printer's SD card.  I placed the cube near the front-right corner (x[max], y0) which is the high spot. 
[image]

The GCode has a G28 and G29.  When the print starts I noticed that the Z-axis motors are not compensating for the differences in height of the bed and the extruder motor is skipping on the front-right of the print but not on the back-left.  You can also see from this picture that there is a marked difference in the thickness of the filament that is extruded.
[image]

Here are my configuration.h and configuration_adv.h files.

```
/**   RCBugFix
 * Marlin 3D Printer Firmware
 * Copyright (C) 2016 MarlinFirmware [https://github.com/MarlinFirmware/Marlin]
 *
 * Based on Sprinter and grbl.
 * Copyright (C) 2011 Camiel Gubbels / Erik van der Zalm
 *
 * This program is free software: you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation, either version 3 of the License, or
 * (at your option) any later version.
 *
 * This program is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this program.  If not, see <http://www.gnu.org/licenses/>.
 *
 */

/**
 * Configuration.h
 *
 * Basic settings such as:
 *
 * - Type of electronics
 * - Type of temperature sensor
 * - Printer geometry
 * - Endstop configuration
 * - LCD controller
 * - Extra features
 *
 * Advanced settings can be found in Configuration_adv.h
 *
 */
#ifndef CONFIGURATION_H
#define CONFIGURATION_H

/**
[... 2479 lines cut ...]
 * echo:i2c-reply: from:99 bytes:5 data:hello
 */

// @section i2cbus

//#define EXPERIMENTAL_I2CBUS
#define I2C_SLAVE_ADDRESS  0 // Set a value from 8 to 127 to act as a slave

/**
 * Add M43 command for pins info and testing
 */
//#define PINS_DEBUGGING

/**
 * Auto-report temperatures with M155 S<seconds>
 */
//#define AUTO_REPORT_TEMPERATURES

/**
 * Include capabilities in M115 output
 */
//#define EXTENDED_CAPABILITIES_REPORT

#endif // CONFIGURATION_ADV_H
```

Tannoo, 2017-01-19:
Try it with the `devel-ubl` branch.

thinkyhead, 2017-03-04:
Sounds like your `DEFAULT_MAX_ACCELERATION` or `DEFAULT_MAX_FEEDRATE` for Z may be set too high. Try reducing these settings and see if it improves the behavior.

closed 2017-10-31 by thinkyhead