MarlinFirmware/Marlin #5600: Wait time/wait event for M401 and M402 manually deployed probes?

JohnOCFII, 2016-12-25:
Greetings,

The configuration.h code in a modified version of RC3 that I'm using does not define Z_PROBE_ALLEN_KEY, so I assume at the time RC3 was released point, that it was equivalent to the current FIX_MOUNTED_PROBE.

That modified RC3 version also relied on modifications in Marlin_main.cpp to check to see if the probe had been deployed, and if not, wait, using M401.  It also had a modified M402 to wait for the probe to be manually stowed after G29 was complete.   The current RC8 does not have this code in M401 or M402 in Marlin_main.cpp.  Are there provisions or recommendations for waiting for a manually deployed/stowed probe?  It appears the current M401 and M402 are stub routines that don't do anything.

While I will move to a BLTouch, I'd like to create the best possible configuration for those still using the manually deployed probe on the G2S Pro.  Suggestions welcome.

Thank you!

I've included a code snippet below from the modified version or RC3 I was using. It appears there was never a pull request to have that code become part of official Marlin.  I'm not sure if something like this has been considered (or is already there, and I missed it)?

Thanks,

John

```#if ENABLED(AUTO_BED_LEVELING_FEATURE) || ( DISABLED(Z_PROBE_SLED) && (HAS_SERVO_ENDSTOPS || ENABLED(Z_PROBE_ALLEN_KEY)))

  /**
   * M401: Engage Z Servo endstop if available
   */
  inline void gcode_M401() {
    #if HAS_SERVO_ENDSTOPS
      raise_z_for_servo();
    #endif
    deploy_z_probe();

        // and block if probe is not deployed
    #define MSG_WAIT_PROBE_DEPLOYED "Wait probe deployment"
     boolean probestate=true,probe_msg=true;
  
     while(probestate){
      #if HAS_Z_MIN
        probestate=READ(Z_MIN_PIN)^Z_MIN_ENDSTOP_INVERTING;
      #endif
      #ifdef Z_PROBE_ENDSTOP
        probestate=READ(Z_PROBE_PIN)^Z_PROBE_ENDSTOP_INVERTING;
      #endif
  
      if (probestate && probe_msg){      // send waiting message once                        
        LCD_MESSAGEPGM(MSG_WAIT_PROBE_DEPLOYED);
        SERIAL_ECHO_START;
        SERIAL_ECHOLNPGM(MSG_WAIT_PROBE_DEPLOYED);
        probe_msg=false;
      }
      idle();
     }
     SERIAL_ECHOLNPGM(MSG_ZPROBE_OUT);
     LCD_MESSAGEPGM(MSG_ZPROBE_OUT);

  }

  /**
   * M402: Retract Z Servo endstop if enabled
   */
  inline void gcode_M402() {
    #if HAS_SERVO_ENDSTOPS
      raise_z_for_servo();
    #endif
    stow_z_probe(false);

   // and block if probe is not stowed
    #define MSG_WAIT_PROBE_DEPLOYED "Wait probe stowed"
     boolean probestate=false,probe_msg=true;
  
     while( ! probestate){
      #if HAS_Z_MIN
        probestate=READ(Z_MIN_PIN)^Z_MIN_ENDSTOP_INVERTING;
      #endif
      #ifdef Z_PROBE_ENDSTOP
        probestate=READ(Z_PROBE_PIN)^Z_PROBE_ENDSTOP_INVERTING;
      #endif
  
      if (! probestate && probe_msg){      // send waiting message once                        
        LCD_MESSAGEPGM(MSG_WAIT_PROBE_DEPLOYED);
        SERIAL_ECHO_START;
        SERIAL_ECHOLNPGM(MSG_WAIT_PROBE_DEPLOYED);
        probe_msg=false;
      }
      idle();
     }
     // add a waiting tiile to secure the probe
     if (! probe_msg){
      millis_t codenum = 3500;
      codenum += millis();  // keep track of when we started waiting

     while (millis() < codenum) idle();
      ;
     }
     SERIAL_ECHOLNPGM(WELCOME_MSG);
     LCD_MESSAGEPGM(WELCOME_MSG);
    
  }

#endif // AUTO_BED_LEVELING_FEATURE && (HAS_SERVO_ENDSTOPS || Z_PROBE_ALLEN_KEY) && !Z_PROBE_SLED
```

thinkyhead, 2017-04-02:
An option for manual deployment / stowage is still a good idea. So we'll keep this open.

boelle, 2019-02-18:
has this been added?