MarlinFirmware/Marlin #5601: RC8 G29 Delta Probing area incorrect

JohnOCFII, 2016-12-25:
Greetings,

I'm working to migrate to RC8 from RC3, and the G29 probe seems to start the Bilinear probing in the middle of the Delta bed and continue to the rear and to the right.  I have to stop the probe before it goes off the bed at the rear.  It also appears the Z-height between probing increases throughout the probing process.

I've attached a zipped configuration.h and debug output from  M111 S32 before issuing G28 and G29 V4.

I did a scan on open and closed issues for G29 in the last 45 days and didn't see anything similar.

Thanks,

John

Blue-Marlin, 2016-12-26:
Please make M119 in mid air. Be sure no endstop/probe is triggered - in case of a triggered when stowed probe. like an Allen key probe, the probe may be triggered. Then activate the probe with M401. Check again with M119. Now even a triggered when stowed probe must be untriggered.

JohnOCFII, 2016-12-26:
Thank you for the suggestion.  I will try that tomorrow.  The steps I will follow will be:

G28 ; home all axes 
; manually lower probe
M401; activate probe 
G29 ; auto-levelling
G1 Z40 F5000 ; lift nozzles
; manually stow probe
M402; deactivate probe

I will issue M119 commands at various points while the hot-end is moving and not touching the bed and report back while is reported.

Thank you!

John

JohnOCFII, 2016-12-27:
@Blue-Marlin 

I lowered the carriages off of the endstops and did some tests:

With probe extended:

SENT: M119
READ: Reporting endstop status
READ: x_max: open
READ: y_max: open
READ: z_min: open
READ: z_max: open

With probe retracted:

SENT: M119
READ: Reporting endstop status
READ: x_max: open
READ: y_max: open
READ: z_min: TRIGGERED
READ: z_max: open

These were conducted after issuing M401 and again after M402.  I did not see any impact from either M401 or M402.

Any suggestions?  Do I need to invert the Z_MIN?   If the Z_MIN probe (using a micro-switch) only has two states (open or TRIGGERED) is the use of M401 and M402 supposed to bring in a third state?  Otherwise, I would think that mechanically and electrically, I'd get the same result if the probe were stowed or actually triggered by hitting the bed.

Thanks,

John

Blue-Marlin, 2016-12-27:
Actually a __probe__ has two additional states - "check me" when active and "don't check me" when stowed.
The above reports show me - your probe is connected to z_min but not configured.
Your Configuration shows me you have a FIX_MOUNTED_PROBE configured - i'm confused.

Please tell more about your probe.

JohnOCFII, 2016-12-27:
The probe is a manually deployed "Allen-key" type probe.  I did not select "Allen-Key" in the configuration as that refers to pressing against a belt.  When the probe is up/stowed, the micro-switch is open.  When the probe is down, the microswitch is compressed until it is lifted by touching the bed.  I've added two photos showing the probe both up and down.

Thank you for your help!

John
[image]
[image]

JohnOCFII, 2016-12-28:
Any additional ideas, @Blue-Marlin?

Thanks,

John

Blue-Marlin, 2016-12-29:
Try with manually defing
`    #define PROBE_IS_TRIGGERED_WHEN_STOWED_TEST`

Additionaly you could experiment with this while loop.
```
      // If endstop is already false, the Z probe is deployed
      if (_TRIGGERED_WHEN_STOWED_TEST == deploy) {     // closed after the probe specific actions.
                                                       // Would a goto be less ugly?
        //while (!_TRIGGERED_WHEN_STOWED_TEST) idle(); // would offer the opportunity
                                                       // for a triggered when stowed manual probe.

        if (!deploy) endstops.enable_z_probe(false); // Switch off triggered when stowed probes early
                                                     // otherwise an Allen-Key probe can't be stowed.
    #endif
```

JohnOCFII, 2017-01-01:
No change in behavior by changing from FIX_MOUNTED_PROBE to Z_PROBE_ALLEN_KEY.

I also manually defined PROBE_IS_TRIGGERED_WHEN_STOWED_TEST - no change with that either.

I think there might be two separate issues here.  One is related to deploying/stowing the probe, and the use of M401 M402, and the other is having G29 probe the appropriate sections of the bed.  I can live with manually dropping the probe and running M401 before the G29, and M402 after G29, but I can't get anywhere with testing auto-bed-leveling when I have to stop the probe manually before it runs into the rear (Z) tower.

Here's a link to a video of the probing behavior:  https://youtu.be/-gRSZ40JF5Q

Thanks for any additional advice folks can provide!

Blue-Marlin, 2017-01-01:
Is the z-cariage going downward at any time during probing?

JohnOCFII, 2017-01-01:
> Is the z-cariage going downward at any time during probing?

Yes.  All three carriages go up and down (varying amounts) during the probing.

You can sort of see that in the belt motion in the video, but I just re-ran M401 followed by G29 with the probe extended and confirmed.

Blue-Marlin, 2017-01-02:
> //#define Z_MIN_PROBE_ENDSTOP // commmented out 12/20 for testing - osh

I'm quite sure this must be active. You never need a z-min-endstop at a DELTA.

```
// Mechanical endstop with COM to ground and NC to Signal uses "false" here (most common setup).
#define X_MIN_ENDSTOP_INVERTING false  // set to true to invert the logic of the endstop.
#define Y_MIN_ENDSTOP_INVERTING false  // set to true to invert the logic of the endstop.
#define Z_MIN_ENDSTOP_INVERTING true  // set to true to invert the logic of the endstop.
#define X_MAX_ENDSTOP_INVERTING false  // set to true to invert the logic of the endstop.
#define Y_MAX_ENDSTOP_INVERTING false  // set to true to invert the logic of the endstop.
#define Z_MAX_ENDSTOP_INVERTING false  // set to true to invert the logic of the endstop.
#define Z_MIN_PROBE_ENDSTOP_INVERTING false  // set to true to invert the logic of the endstop.
```
Polarity of endstop and probe must be the same.

JohnOCFII, 2017-01-02:
I did some testing:

If I have this line active:
```#define Z_MIN_PROBE_ENDSTOP```
instead of this line:
```//#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN```

M119 reports this status from a fresh-reset:

Reporting endstop status
x_max: open
y_max: open
z_min: TRIGGERED
z_probe: open

And G28 fails (note the absense of z_max in the list above).

The way I had it before:
```#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN```
and the line below commented out:
```//#define Z_MIN_PROBE_ENDSTOP```

M119 reports this status from a fresh-reset:
Reporting endstop status
x_max: open
y_max: open
z_min: TRIGGERED
z_max: open

And G28 succeeds.

I also tried changing the line below from false to true (using the configuration above that succeeded) and there was no change in M119 report, and no change in the G29 behavior where the probing starts in the middle of the bed and will run into the Z-axis rods if not stopped.

```#define Z_MIN_PROBE_ENDSTOP_INVERTING true  // set to true to invert the logic of the endstop. - changed from false to true 1/2/2017 - osh```

Perhaps I misinterpreted what you suggested I try.  I would have expected that when I initiate G29, when the spider lowers it would start to move towards the front of the bed to start the scan pattern.  I'm not sure the endstop would be involved in that logic.

JohnOCFII, 2017-01-03:
In looking at the ALLEN_KEY options, I'm currently using the defaults from the Delta example/configuration.h.  I'm pretty sure I need to comment out some of this, even though they are all commented in in the delta example configuration.h

I'll review tomorrow, but this might well be related to my issues:

#if ENABLED(Z_PROBE_ALLEN_KEY)
  // 2 or 3 sets of coordinates for deploying and retracting the spring loaded touch probe on G29,
  // if servo actuated touch probe is not defined. Uncomment as appropriate for your printer/probe.

  #define Z_PROBE_ALLEN_KEY_DEPLOY_1_X 30.0
  #define Z_PROBE_ALLEN_KEY_DEPLOY_1_Y DELTA_PRINTABLE_RADIUS
  #define Z_PROBE_ALLEN_KEY_DEPLOY_1_Z 100.0
  #define Z_PROBE_ALLEN_KEY_DEPLOY_1_FEEDRATE XY_PROBE_SPEED

  #define Z_PROBE_ALLEN_KEY_DEPLOY_2_X 0.0
  #define Z_PROBE_ALLEN_KEY_DEPLOY_2_Y DELTA_PRINTABLE_RADIUS
  #define Z_PROBE_ALLEN_KEY_DEPLOY_2_Z 100.0
  #define Z_PROBE_ALLEN_KEY_DEPLOY_2_FEEDRATE (XY_PROBE_SPEED)/10

  #define Z_PROBE_ALLEN_KEY_DEPLOY_3_X Z_PROBE_ALLEN_KEY_DEPLOY_2_X * 0.75
  #define Z_PROBE_ALLEN_KEY_DEPLOY_3_Y Z_PROBE_ALLEN_KEY_DEPLOY_2_Y * 0.75
  #define Z_PROBE_ALLEN_KEY_DEPLOY_3_Z Z_PROBE_ALLEN_KEY_DEPLOY_2_Z
  #define Z_PROBE_ALLEN_KEY_DEPLOY_3_FEEDRATE XY_PROBE_SPEED

  #define Z_PROBE_ALLEN_KEY_STOW_1_X -64.0 // Move the probe into position
  #define Z_PROBE_ALLEN_KEY_STOW_1_Y 56.0
  #define Z_PROBE_ALLEN_KEY_STOW_1_Z 23.0
  #define Z_PROBE_ALLEN_KEY_STOW_1_FEEDRATE XY_PROBE_SPEED

  #define Z_PROBE_ALLEN_KEY_STOW_2_X -64.0 // Push it down
  #define Z_PROBE_ALLEN_KEY_STOW_2_Y 56.0
  #define Z_PROBE_ALLEN_KEY_STOW_2_Z 3.0
  #define Z_PROBE_ALLEN_KEY_STOW_2_FEEDRATE (XY_PROBE_SPEED)/10

  #define Z_PROBE_ALLEN_KEY_STOW_3_X -64.0 // Move it up to clear
  #define Z_PROBE_ALLEN_KEY_STOW_3_Y 56.0
  #define Z_PROBE_ALLEN_KEY_STOW_3_Z 50.0
  #define Z_PROBE_ALLEN_KEY_STOW_3_FEEDRATE XY_PROBE_SPEED

  #define Z_PROBE_ALLEN_KEY_STOW_4_X 0.0
  #define Z_PROBE_ALLEN_KEY_STOW_4_Y 0.0
  #define Z_PROBE_ALLEN_KEY_STOW_4_Z Z_PROBE_ALLEN_KEY_STOW_3_Z
  #define Z_PROBE_ALLEN_KEY_STOW_4_FEEDRATE XY_PROBE_SPEED

#endif // Z_PROBE_ALLEN_KEY

JohnOCFII, 2017-01-05:
Commenting out the Z_PROBE_ALLEN_KEY options 2, 3, and 4 had no impact on my bed probing starting in the middle.  I'm sure there is something simple I'm missing.  I'm not sure what.

Looking through the configuration.h, it appears the following #defines are related to the probing area:
```
DELTA_PRINTABLE_RADIUS 		92.0
DELTA_PROBEABLE_RADIUS 		60.0
LEFT_PROBE_BED_POSITION 	-60.0
RIGHT_PROBE_BED_POSITION 	 60.0
FRONT_PROBE_BED_POSITION	-60.0
BACK_PROBE_BED_POSITION		 60.0
```

These all look reasonable.  Not sure how the code in Marlin_main.cpp gets from here to just probing the back half of the bed.  Also - it looks like there are checks in SanityCheck.h to see that the points are reasonable, so I'm not sure why it is running off the bed if I don't manually stop it.

Others seem be be running G29 without issue.  Not sure if any of the other positive reports are running a Delta.

I welcome any additional clues.  Perhaps I should try RCBugfix instead of RC8 -- I've been watching and didn't key off of any fix in this area, but maybe I was not searching for the correct term.

Thanks,

John

DavidTre07, 2017-01-11:
Hi @JohnOCFII ,
I faced the same problem with a delta (Kossel Mini) and the RC8.
I solved the problem by changing X_PROBE_OFFSET_FROM_EXTRUDER and Y_PROBE_OFFSET_FROM_EXTRUDER to integers.
For example, if you have a value like 1.7 change it to 2
I was migrating from RC3 so I made theses changes:
//#define X_PROBE_OFFSET_FROM_EXTRUDER 1.7     // X offset: -left  +right  [of the nozzle]
#define X_PROBE_OFFSET_FROM_EXTRUDER **2**
//#define Y_PROBE_OFFSET_FROM_EXTRUDER 16.4   // Y offset: -front +behind [the nozzle]
#define Y_PROBE_OFFSET_FROM_EXTRUDER **16**
#define Z_PROBE_OFFSET_FROM_EXTRUDER -6.7  // Z offset: -below +above  [the nozzle]

JohnOCFII, 2017-01-11:
You gave me hope for a second there, @DavidTre07 - but my X and Y are already integers:

```
#define X_PROBE_OFFSET_FROM_EXTRUDER -19    // X offset: -left  +right  [of the nozzle]
#define Y_PROBE_OFFSET_FROM_EXTRUDER +11  // Y offset: -front +behind [the nozzle] 
#define Z_PROBE_OFFSET_FROM_EXTRUDER -1.90  // Z offset: -below +above  [the nozzle]
```

I appreciate you taking the time to offer up a suggestion!

If your G29 is working I'd love a copy of your configuration.h so I could compare to see if something jumps out.  What type of probe do you have?

John

DavidTre07, 2017-01-11:
Sorry for that, so perhaps I solve my problem with a different parameter...
When I got this bug I changed a lot of things (too much parameters).

Could I copy my config directly on this thread ? Or it's preferable to use an external system ?

DavidTre07, 2017-01-11:
Sorry, I don't fully answer to your questions:
Yes my G29 is working and I'm using an opto switch

JohnOCFII, 2017-01-11:
Thanks, @DavidTre07!  I'll look at this more closely later tonight.

John

DavidTre07, 2017-01-14:
Hi @JohnOCFII ,
did you find out the problem ?
David.

JohnOCFII, 2017-01-15:
@DavidTre07 

> Hi @JohnOCFII ,
> did you find out the problem ?

It has been a busy week.  Tomorrow will be my first chance to work with my printer.  

Thanks for asking.

JohnOCFII, 2017-01-15:
@DavidTre07 I finally had the time to compare your configuration.h to mine.  Other than the expected differences in dimensions, etc., nothing looked obvious.  

How are you invoking the bed leveling?  Do you issue an M401 or M420 before issuing the G29 command?

My probe needs to be manually lowered.  Here are my steps:

G28 ; Home (works correctly)
M401; hot-end effector moves part way down and to the rear and stops, and then I receive an error message on the display:  Err: ZPROBE
At this point I lower the probe manually, and continue with 
G29 - the bed is probed - the area seems mostly correct. After probe is complete, the hot-end effector moves up and left, and stops.  If I attempt to enter a movement command such as G0 X0 Y0 I see the following error message in the terminal window:

```
Printer stopped due to errors. Fix the error and use M999 to restart. (Temperature is reset.  Reset it after restarting)
```

I can enter G28 to re-home at this point, but the message Err: ZPROBE is still on the LCD.

If I skip the M401, and issue a G29 directly after G28, I get the issues described earlier in this thread where the probing starts in the middle of the bed and moves to the rear.  I have to manually stop it at that point otherwise it will hit the Z rods.

Here is my current configuration.h

[Configuration-h.txt](https://github.com/MarlinFirmware/Marlin/files/706868/Configuration-h.txt)

DavidTre07, 2017-01-15:
Hi @JohnOCFII ,
I'm not using M401, I'm using G28 and directly G29.
could you try with DELTA_HOME_TO_SAFE_ZONE activated ?

This new functionality do some strange things...
I'll look deeper on your conf file is this not the solution.
David.

JohnOCFII, 2017-01-15:
Interesting suggestion, @DavidTre07.  

With DELTA_HOME_TO_SAFE_ZONE defined, I now get bed probed similar to above (ie, it covers most of the bed successfully) whether or not I issue M401 after I issue G28.  Unfortunately, it still fails in the same way after the probing.

I had elected not to have that feature enabled, as I lose 60mm from my 200mm build height with it activated.  I understand I can't go to full diameter at a higher position but most tall items that I print get narrow towards the top anyway.

Thanks again for helping!

John

DavidTre07, 2017-01-15:
@JohnOCFII , I think I don't understand somethings...
You wrote in the last message: "it still fails in the same way after the probing".
Your problem is not that the probe function is not probing the whole glass surface and Z-height between probing increases ?
I think that DELTA_HOME_TO_SAFE_ZONE solve this problem, no ?

I agree with you, I don't like the DELTA_HOME_TO_SAFE_ZONE option too but... without it I'm not able to autoprobe...
Regards,
David.

JohnOCFII, 2017-01-15:
@DavidTre07 
My problem has changed since I defined DELTA_HOME_TO_SAFE_ZONE.

Now, when I issue G29, the bed is probed (approximately 75 touches).  After the last touch, the hot-end rises from the bed, moves to the left, then stops.  The message: Err: ZPROBE is displayed on the LCD.

After that, I can not issue any movement commands other than G28.

Maybe I should try RCBugFix.

DavidTre07, 2017-01-16:
@JohnOCFII 
Are you sure if Z_PROBE_ALLEN_KEY_STOW_1_X is well defined ?
In your conf file Z_PROBE_ALLEN_KEY_STOW_2_X and Z_PROBE_ALLEN_KEY_STOW_3_X are commented.

In my conf file I have 3 values to be sure that it store the allen key (and the endstop change it's state).

I'm not sure but if the autoprove endstop remain in the same state, do the firmware report an error ?
David.

JohnOCFII, 2017-01-16:
@DavidTre07 I will look more closely at that.  I had been using fixed probe earlier, so had not paid close attention.  Then, I looked at t multiple ALLEN_KEY options, and commented out all by the _1_ options, as I thought maybe having more than one was confusing the situation.

The comments around the definition for Z_ALLEN_KEY_PROBE is confusing as it refers to pressure on a belt.  I don't have anything like that -- I have something that LOOKS like an Allen key that touches the bed and makes contact with a microswitch.

Thanks again.

JohnOCFII, 2017-01-17:
@DavidTre07 

Some positive progress to report!

I switched the bed leveling from ALLEN_KEY to FIXED_PROBE and the G29 probe completes and I can issue movement commands after it is complete!

Here are the results of that first probe.  The first point was off the bed, so I just used my finger to activate the probe for that point.

 Bilinear Leveling Grid:
      0     1     2     3     4     5     6     7     8
 0 +6.17 -0.41 -0.27 -0.13 -0.01 +0.14 +0.20 +0.28 +0.45
 1 -0.37 -0.29 -0.20 -0.11 -0.04 +0.03 +0.14 +0.28 +0.44
 2 -0.35 -0.24 -0.18 -0.08 -0.01 +0.06 +0.18 +0.20 +0.29
 3 -0.27 -0.23 -0.17 -0.11 -0.05 -0.03 +0.03 +0.06 +0.05
 4 -0.30 -0.23 -0.18 -0.08 -0.12 -0.08 +0.04 -0.01 +0.05
 5 -0.33 -0.28 -0.23 -0.23 -0.16 -0.18 -0.12 -0.16 -0.19
 6 -0.34 -0.29 -0.29 -0.23 -0.21 -0.21 -0.21 -0.23 -0.28
 7 -0.42 -0.34 -0.30 -0.30 -0.24 -0.30 -0.25 -0.36 -0.41
 8 -0.43 -0.39 -0.33 -0.26 -0.27 -0.29 -0.34 -0.40 -0.50

I need to move the coordinates back 10 to 20mm for the probe area, but this is much further along that I've been!

Thanks for your help.

John

JohnOCFII, 2017-01-17:
@DavidTre07 

I moved to RCBugFix (as of today).  I can now comment out DELTA_HOME_TO_SAFE_ZONE and the G29 bed probing still completes (successfully?).  I still need to tweak the bed parameters as one of the probe points is still off of the bed.

I still have issues/questions on the bed leveling process.  

If I manually lower the probe using Pronterface after doing the G29 bed leveling, I don't believe it is reflecting any "leveling."  Points that could lightly grasp a piece of paper at .2 above the bed are still grasping paper at .2 above the bed.  I would have thought that it would provide leveling such that the light grasp of paper would be at 0.0 or at worst, 0.1.

Thoughts?

Thanks,

John

```Bilinear Leveling Grid:
      0     1     2     3     4     5     6     7     8
 0 -1.74 -0.54 -0.34 -0.15 -0.03 +0.05 +0.29 +0.64 +0.89
 1 -2.12 -0.43 -0.30 -0.16 -0.02 +0.07 +0.38 +0.53 +0.51
 2 -0.42 -0.30 -0.20 -0.07 -0.00 +0.12 +0.18 +0.31 +0.22
 3 -0.34 -0.28 -0.16 -0.10 -0.03 +0.00 +0.08 +0.07 +0.13
 4 -0.33 -0.25 -0.20 -0.13 -0.08 -0.01 +0.09 +0.03 +0.09
 5 -0.16 -0.10 -0.04 -0.00 +0.04 +0.05 -0.12 -0.14 -0.18
 6 -0.20 -0.11 -0.10 -0.09 -0.02 +0.07 +0.03 -0.01 +0.06
 7 -0.26 -0.20 -0.15 -0.13 -0.11 -0.05 -0.10 -0.11 -0.21
 8 -0.27 -0.21 -0.37 -0.30 -0.31 -0.30 -0.33 -0.36 -0.38
```

DavidTre07, 2017-01-17:
Hi @JohnOCFII ,
good to read that, so I will make a try to the RCBugFix.

Your question about G29 Leveling Grid is really out off my skills...
And it's really a good question (as I have the same ;) ).
Here my G29 result:
`Bilinear Leveling Grid:
      0     1     2     3     4     5     6     7     8
 0 +0.84 +1.04 +0.93 +0.56 +0.27 +0.23 +0.42 +0.62 +0.67
 1 +0.31 +0.57 +0.75 +0.71 +0.42 +0.49 +0.58 +0.57 +0.52
 2 -0.03 +0.27 +0.43 +0.53 +0.54 +0.56 +0.50 +0.50 +0.49
 3 -0.30 +0.09 +0.23 +0.39 +0.46 +0.48 +0.48 +0.46 +0.50
 4 -0.34 -0.11 +0.11 +0.28 +0.36 +0.41 +0.42 +0.44 +0.51
 5 -0.45 -0.29 -0.06 +0.13 +0.25 +0.34 +0.35 +0.39 +0.49
 6 -0.39 -0.47 -0.26 -0.04 +0.10 +0.22 +0.28 +0.30 +0.40
 7 -0.66 -0.61 -0.45 -0.20 -0.02 +0.11 +0.21 +0.27 +0.36
 8 -0.94 -0.75 -0.56 -0.33 -0.13 +0.03 +0.17 +0.23 +0.30`

Someone could help us to understand these numbers ?
Thanks,
David.

JohnOCFII, 2017-01-18:
@DavidTre07 

Did you enable the M48 probe test?  I was pretty impressed with my physical probe results:  

M48 test with single tap: (centered on bed)
```
Mean: 1.682513  Min: 1.673  Max: 1.698  Range: 0.025
Standard Deviation: 0.009352
```

I've been reading some of the other open issues related to auto-level and I don't think we are alone in our confusion.  I need to do some more reading to see if there is something we need to do between the G29 and the beginning of a print to have the probing data actually be used.

John

JohnOCFII, 2017-01-18:
I'm going to close this issue and start a new one as I've crossed through many different issues throughout this thread.  If I don't find an answer to our question, I'll open a new issue.

closed 2017-01-18 by JohnOCFII