MarlinFirmware/Marlin #5605: Delta bed leveling strange behaviour

nzinov, 2016-12-26:
I am running delta printer with RCBugFix (2e904cb4). Today I tried test print with autoleveling on. Here goes my calibration matrix:

```
Bilinear Leveling Grid:
      0     1     2     3     4     5     6     7     8
 0 +0.50 +0.51 -0.07 +0.01 -0.14 -0.29 -0.43 0.00 0.00
 1 +0.10 -0.03 -0.19 -0.26 -0.31 -0.41 -0.45 -0.60 -0.61
 2 -0.40 -0.53 -0.44 -0.35 -0.30 -0.42 -0.53 -0.68 -0.70
 3 -0.68 -0.46 -0.31 -0.21 -0.27 -0.37 -0.49 -0.72 -0.85
 4 -0.52 -0.41 -0.29 -0.23 -0.24 -0.36 -0.51 -0.73 -0.94
 5 -0.39 -0.23 -0.23 -0.13 -0.20 -0.26 -0.44 -0.71 -0.91
 6 -0.31 -0.29 -0.24 -0.22 -0.19 -0.27 -0.44 -0.61 -0.80
 7 -0.21 -0.19 -0.24 -0.24 -0.25 -0.32 -0.42 -0.62 -0.91
 8 -0.26 -0.34 -0.43 -0.45 -0.46 -0.55 -0.60 -0.69 -0.82
```
Print went smooth and first level was correct (it was spanning only 7x7 cm though). But transition from first layers to upper layers was strange. I attached a pic of the vertical surface. In my vision (I don't know leveling internals) it was smoothing in wrong direction. The vertical surface had two levels where there was visible shift: one at about 10'th layer and another at approx 30 layer.

manianac, 2016-12-26:
I can't tell from the picture but it looked like it shifted parallel the layers.  Given the ABL code shouldn't ever affect anything but all the axis motors at once, it might not be related.  Can you do the same print with ABL enabled?

nzinov, 2017-01-02:
Turns out I calibrated endstops wrong. Tested again & can confirm bilinear leveling works as expected

closed 2017-01-02 by nzinov