MarlinFirmware/Marlin #5611: Getting an Error while trying to upload.

Rlewisrlou666, 2016-12-27:
```
`Arduino: 1.8.0 (Windows 10), Board: "Sanguino, ATmega1284 or ATmega1284P (8 MHz)"

In file included from sketch\MarlinSerial.cpp:33:0:

sketch\stepper.h: In static member function 'static short unsigned int Stepper::calc_timer(short unsigned int)':

stepper.h:301: error: 'speed_lookuptable_fast' was not declared in this scope

         unsigned short table_address = (unsigned short)&speed_lookuptable_fast[(unsigned char)(step_rate >> 8)][0];

                                                         ^

stepper.h:308: error: 'speed_lookuptable_slow' was not declared in this scope

         unsigned short table_address = (unsigned short)&speed_lookuptable_slow[0][0];

                                                         ^

exit status 1
'speed_lookuptable_fast' was not declared in this scope

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
`
```

Blue-Marlin, 2016-12-27:
> `Arduino: 1.8.0 (Windows 10), Board: "Sanguino, ATmega1284 or ATmega1284P (8 MHz)"

8MHz ??? That can't be right. The lookup table depends on the processors frequency.
If you really want to compile for 8MHz you have to make a matching one. (createSpeedLookupTable.py)

Blue-Marlin, 2016-12-27:
> `Arduino: 1.8.0 (Windows 10), Board: "Sanguino, ATmega1284 or ATmega1284P (8 MHz)"

Arduino: 1.8.0 ??? Is completely untested. Seems to be the reunion of Arduino.cc with Arduino.org. So may have more changes then we'd like to see.

Bob-the-Kuhn, 2016-12-28:
@Rlewisrlou666 - what Arduino IDE extension are you using for the Sanguino support?

Bob-the-Kuhn, 2016-12-30:
I ran into that error yesterday.  I believe it was caused by the wrong CPU frequency being selected.

If that doesn't fix it then take the Arduino IDE back to zero by deleting the "Arduino15" directory.  The path to my  "Arduino15" directory was: C:\Users\Bob\AppData\Local

Also, some board packages don't play well with others.  Try loading/adding/installing just the board package for Sanguino.

closed 2017-02-26 by thinkyhead