MarlinFirmware/Marlin #5615: unable to find a register to spill in class 'POINTER_REGS'

Firesped, 2016-12-27:
I hate for this to be a regular thing BUT I may have found something closer to what is causing it. I am working on configuring RC8 for the raise3d N-SERIES. Everything was compiling fine last night but when I got home from work, It started doing this and I could not figure out that is causing it.

I thought I might have damaged the code elsewhere. The touchscreen on the N-Series is looking for very specific parse data and I have had to guess what it is. I've managed to get touchscreen to let the firmware communicate now but that is another issue.

I've got 2 extruders set. If I only set one, it works. set 2 and it doesn't. I started to configure another build with the same settings. What I found is when I turned on the z_probe, it stopped work, turned off and it worked. so it must be some issue with the z_probe. I was using #define FIX_MOUNTED_PROBE

just checked the other build I was working on before. commented out the #define FIX_MOUNTED_PROBE and it compiled.

Linking everything together...
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-gcc" -w -Os -g -flto -fuse-linker-plugin -Wl,--gc-sections,--relax -mmcu=atmega2560  -o "C:\Users\Michael\AppData\Local\Temp\arduino_build_380057/Marlin.ino.elf" "C:\Users\Michael\AppData\Local\Temp\arduino_build_380057\sketch\M100_Free_Mem_Chk.cpp.o" "C:\Users\Michael\AppData\Local\Temp\arduino_build_380057\sketch\Marlin.ino.cpp.o" "C:\Users\Michael\AppData\Local\Temp\arduino_build_380057\sketch\MarlinSerial.cpp.o" "C:\Users\Michael\AppData\Local\Temp\arduino_build_380057\sketch\Marlin_main.cpp.o" "C:\Users\Michael\AppData\Local\Temp\arduino_build_380057\sketch\Sd2Card.cpp.o" "C:\Users\Michael\AppData\Local\Temp\arduino_build_380057\sketch\SdBaseFile.cpp.o" "C:\Users\Michael\AppData\Local\Temp\arduino_build_380057\sketch\SdFatUtil.cpp.o" "C:\Users\Michael\AppData\Local\Temp\arduino_build_380057\sketch\SdFile.cpp.o" "C:\Users\Michael\AppData\Local\Temp\arduino_build_380057\sketch\SdVolume.cpp.o" "C:\Users\Michael\AppData\Local\Temp\arduino_build_380057\sketch\blinkm.cpp.o" "C:\Users\Michael\AppData\Local\Temp\arduino_build_380057\sketch\cardreader.cpp.o" "C:\Users\Michael\AppData\Local\Temp\arduino_build_380057\sketch\configuration_store.cpp.o" "C:\Users\Michael\AppData\Local\Temp\arduino_build_380057\sketch\dac_mcp4728.cpp.o" "C:\Users\Michael\AppData\Local\Temp\arduino_build_380057\sketch\digipot_mcp4451.cpp.o" "C:\Users\Michael\AppData\Local\Temp\arduino_build_380057\sketch\endstops.cpp.o" "C:\Users\Michael\AppData\Local\Temp\arduino_build_380057\sketch\mesh_bed_leveling.cpp.o" "C:\Users\Michael\AppData\Local\Temp\arduino_build_380057\sketch\planner.cpp.o" "C:\Users\Michael\AppData\Local\Temp\arduino_build_380057\sketch\planner_bezier.cpp.o" "C:\Users\Michael\AppData\Local\Temp\arduino_build_380057\sketch\printcounter.cpp.o" "C:\Users\Michael\AppData\Local\Temp\arduino_build_380057\sketch\qr_solve.cpp.o" "C:\Users\Michael\AppData\Local\Temp\arduino_build_380057\sketch\servo.cpp.o" "C:\Users\Michael\AppData\Local\Temp\arduino_build_380057\sketch\stepper.cpp.o" "C:\Users\Michael\AppData\Local\Temp\arduino_build_380057\sketch\stepper_dac.cpp.o" "C:\Users\Michael\AppData\Local\Temp\arduino_build_380057\sketch\stepper_indirection.cpp.o" "C:\Users\Michael\AppData\Local\Temp\arduino_build_380057\sketch\stopwatch.cpp.o" "C:\Users\Michael\AppData\Local\Temp\arduino_build_380057\sketch\temperature.cpp.o" "C:\Users\Michael\AppData\Local\Temp\arduino_build_380057\sketch\twibus.cpp.o" "C:\Users\Michael\AppData\Local\Temp\arduino_build_380057\sketch\ultralcd.cpp.o" "C:\Users\Michael\AppData\Local\Temp\arduino_build_380057\sketch\utility.cpp.o" "C:\Users\Michael\AppData\Local\Temp\arduino_build_380057\sketch\vector_3.cpp.o" "C:\Users\Michael\AppData\Local\Temp\arduino_build_380057\sketch\watchdog.cpp.o" "C:\Users\Michael\AppData\Local\Temp\arduino_build_380057/core\core.a" "-LC:\Users\Michael\AppData\Local\Temp\arduino_build_380057" -lm
C:\Users\Michael\AppData\Local\Temp\arduino_build_380057\sketch\temperature.cpp: In function 'manage_heater':

temperature.cpp:797: error: unable to find a register to spill in class 'POINTER_REGS'

 }

 ^

temperature.cpp:797: error: this is the insn:

(insn 109 106 111 5 (set (reg:SF 110 [ D.2332 ])

        (mem:SF (post_inc:HI (reg:HI 111 [ ivtmp.126 ])) [7 MEM[base: _141, offset: 0B]+0 S4 A8])) C:\Users\Michael\AppData\Local\Temp\arduino_build_380057\sketch\temperature.cpp:684 100 {*movsf}

     (expr_list:REG_INC (reg:HI 111 [ ivtmp.126 ])

        (nil)))

C:\Users\Michael\AppData\Local\Temp\arduino_build_380057\sketch\temperature.cpp:797: confused by earlier errors, bailing out

lto-wrapper: C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-gcc returned 1 exit status

c:/program files (x86)/arduino/hardware/tools/avr/bin/../lib/gcc/avr/4.9.2/../../../../avr/bin/ld.exe: lto-wrapper failed

collect2.exe: error: ld returned 1 exit status

exit status 1
unable to find a register to spill in class 'POINTER_REGS'

bobc, 2016-12-27:
The "cause" seems to be a compiler bug which appeared in avr-gcc 4.9.2, so first thing is to use Arduino AVR boards version 1.6.8.

Firesped, 2016-12-28:
I'm not sure what is going on. I don't know if my printer has a z-probe. It has an z-screw attached to the bed and endstop it hits. They have z-probe offsets defined .

This is from the current firmware on the printer.

#define CUSTOM_M_CODES
#ifdef CUSTOM_M_CODES
  #define CUSTOM_M_CODE_SET_Z_PROBE_OFFSET 851
  #define Z_PROBE_OFFSET_RANGE_MIN -15
  #define Z_PROBE_OFFSET_RANGE_MAX -5
#endif

  // Offsets to the probe relative to the extruder tip (Hotend - Probe)
  // X and Y offsets must be integers
  #define X_PROBE_OFFSET_FROM_EXTRUDER -25     // Probe on: -left  +right
  #define Y_PROBE_OFFSET_FROM_EXTRUDER -29     // Probe on: -front +behind
  #define Z_PROBE_OFFSET_FROM_EXTRUDER -12.35  // -below (always!)

  #define Z_RAISE_BEFORE_HOMING 4       // (in mm) Raise Z before homing (G28) for Probe Clearance.
                                        // Be sure you have this distance over your Z_MAX_POS in case

  #define XY_TRAVEL_SPEED 8000         // X and Y axis travel speed between probes, in mm/min

  #define Z_RAISE_BEFORE_PROBING 15    //How much the extruder will be raised before traveling to the first probing point.
  #define Z_RAISE_BETWEEN_PROBINGS 5  //How much the extruder will be raised when traveling from between next probing points
  #define Z_RAISE_AFTER_PROBING 15    //How much the extruder will be raised after the last probing point.

Roxy-3D, 2016-12-28:
Have you resolved the compiler warning by installing Arduino v. 1.6.8 or 1.6.9 ?

Firesped, 2016-12-29:
installing arduino 1.6.8 threw out all kinds of other issues. using bltouch z probe does  not generate the issue, only with a fix_mounted_probe, so I must not have it configured right.

Firesped, 2016-12-30:
it seems to have something to do with the custom code I need for the reporting to my touchscreen. the error seems to relate to a frame pointer being eliminated but other things were dependent on that pointer.

for m105, I have it modified to shot it like this

> T:11.8 /0.0 B:13.8 /0.0 T0:11.8 /0.0 F0:100 T1:12.7 /0.0 F1:100 S:0 P:100

```
#if HAS_TEMP_HOTEND || HAS_TEMP_BED

  void print_heaterstates() {
    #ifdef N_SERIES_PROTOCLE
      #if HOTENDS > 1
        HOTEND_LOOP() {
          SERIAL_PROTOCOLPAIR(" T", e);
          SERIAL_PROTOCOLCHAR(':');
          SERIAL_PROTOCOL_F(thermalManager.degHotend(e), 1);
          SERIAL_PROTOCOLPGM(" /");
          SERIAL_PROTOCOL_F(thermalManager.degTargetHotend(e), 1);
          SERIAL_PROTOCOLPAIR(" F", e);
          SERIAL_PROTOCOLCHAR(":");
          SERIAL_PROTOCOL(flow_percentage[e]);
        }
      #endif
      SERIAL_PROTOCOLPGM(" S:");
      SERIAL_PROTOCOL(fanSpeeds[0]);
      SERIAL_PROTOCOLPGM(" P:");
      SERIAL_PROTOCOL(feedrate_percentage);
    #else    
      #if HAS_TEMP_HOTEND
        SERIAL_PROTOCOLPGM(" T:");
        SERIAL_PROTOCOL_F(thermalManager.degHotend(target_extruder), 1);
        SERIAL_PROTOCOLPGM(" /");
        SERIAL_PROTOCOL_F(thermalManager.degTargetHotend(target_extruder), 1);
        #if ENABLED(SHOW_TEMP_ADC_VALUES)
          SERIAL_PROTOCOLPAIR(" (", thermalManager.current_temperature_raw[target_extruder] / OVERSAMPLENR);
          SERIAL_CHAR(')');
        #endif
      #endif
      #if HAS_TEMP_BED
        SERIAL_PROTOCOLPGM(" B:");
        SERIAL_PROTOCOL_F(thermalManager.degBed(), 1);
        SERIAL_PROTOCOLPGM(" /");
        SERIAL_PROTOCOL_F(thermalManager.degTargetBed(), 1);
        #if ENABLED(SHOW_TEMP_ADC_VALUES)
          SERIAL_PROTOCOLPAIR(" (", thermalManager.current_temperature_bed_raw / OVERSAMPLENR);
          SERIAL_CHAR(')');
        #endif
      #endif
      #if HOTENDS > 1
        HOTEND_LOOP() {
          SERIAL_PROTOCOLPAIR(" T", e);
          SERIAL_PROTOCOLCHAR(':');
          SERIAL_PROTOCOL_F(thermalManager.degHotend(e), 1);
          SERIAL_PROTOCOLPGM(" /");
          SERIAL_PROTOCOL_F(thermalManager.degTargetHotend(e), 1);
          #if ENABLED(SHOW_TEMP_ADC_VALUES)
            SERIAL_PROTOCOLPAIR(" (", thermalManager.current_temperature_raw[e] / OVERSAMPLENR);
            SERIAL_CHAR(')');
          #endif
        }
      #endif
    
      SERIAL_PROTOCOLPGM(" @:");
      SERIAL_PROTOCOL(thermalManager.getHeaterPower(target_extruder));
      #if HAS_TEMP_BED
        SERIAL_PROTOCOLPGM(" B@:");
        SERIAL_PROTOCOL(thermalManager.getHeaterPower(-1));
      #endif
        #if HOTENDS > 1
        HOTEND_LOOP() {
          SERIAL_PROTOCOLPAIR(" @", e);
          SERIAL_PROTOCOLCHAR(':');
          SERIAL_PROTOCOL(thermalManager.getHeaterPower(e));
        }
      #endif
    #endif
  }
#endif

/**
 * M105: Read hot end and bed temperature
 */
inline void gcode_M105() {
  if (get_target_extruder_from_command(105)) return;

  #if HAS_TEMP_HOTEND || HAS_TEMP_BED
    SERIAL_PROTOCOLPGM(MSG_OK);
    #ifdef N_SERIES_PROTOCLE
      #if HAS_TEMP_HOTEND
        SERIAL_PROTOCOLPGM(" T:");
        SERIAL_PROTOCOL_F(thermalManager.degHotend(target_extruder), 1);
        SERIAL_PROTOCOLPGM(" /");
        SERIAL_PROTOCOL_F(thermalManager.degTargetHotend(target_extruder), 1);
      #endif
      #if HAS_TEMP_BED
        SERIAL_PROTOCOLPGM(" B:");
        SERIAL_PROTOCOL_F(thermalManager.degBed(), 1);
        SERIAL_PROTOCOLPGM(" /");
        SERIAL_PROTOCOL_F(thermalManager.degTargetBed(), 1);
      #endif
    #endif
    print_heaterstates();
  #else // !HAS_TEMP_HOTEND && !HAS_TEMP_BED
    SERIAL_ERROR_START;
    SERIAL_ERRORLNPGM(MSG_ERR_NO_THERMISTORS);
  #endif

  SERIAL_EOL;
}

```

This function is working.

Firesped, 2016-12-30:
in my custom code for m109, it is coming up with the pointer issue. It's like it doesn't want to exceed a certain number of lines of code.

```
/**
 * M109: Sxxx Wait for extruder(s) to reach temperature. Waits only when heating.
 *       Rxxx Wait for extruder(s) to reach temperature. Waits when heating and cooling.
 */
inline void gcode_M109() {

  if (get_target_extruder_from_command(109)) return;
  if (DEBUGGING(DRYRUN)) return;

  #if ENABLED(SINGLENOZZLE)
    if (target_extruder != active_extruder) return;
  #endif

  bool no_wait_for_cooling = code_seen('S');
  if (no_wait_for_cooling || code_seen('R')) {
    thermalManager.setTargetHotend(code_value_temp_abs(), target_extruder);
    #if ENABLED(DUAL_X_CARRIAGE)
      if (dual_x_carriage_mode == DXC_DUPLICATION_MODE && target_extruder == 0)
        thermalManager.setTargetHotend(code_value_temp_abs() == 0.0 ? 0.0 : code_value_temp_abs() + duplicate_extruder_temp_offset, 1);
    #endif

    #if ENABLED(PRINTJOB_TIMER_AUTOSTART)
      /**
       * We use half EXTRUDE_MINTEMP here to allow nozzles to be put into hot
       * stand by mode, for instance in a dual extruder setup, without affecting
       * the running print timer.
       */
      if (code_value_temp_abs() <= (EXTRUDE_MINTEMP)/2) {
        print_job_timer.stop();
        LCD_MESSAGEPGM(WELCOME_MSG);
      }
      /**
       * We do not check if the timer is already running because this check will
       * be done for us inside the Stopwatch::start() method thus a running timer
       * will not restart.
       */
      else print_job_timer.start();
    #endif

    if (thermalManager.isHeatingHotend(target_extruder)) LCD_MESSAGEPGM(MSG_HEATING);
  }

  #if ENABLED(AUTOTEMP)
    planner.autotemp_M104_M109();
  #endif

  #if TEMP_RESIDENCY_TIME > 0
    millis_t residency_start_ms = 0;
    // Loop until the temperature has stabilized
    #define TEMP_CONDITIONS (!residency_start_ms || PENDING(now, residency_start_ms + (TEMP_RESIDENCY_TIME) * 1000UL))
  #else
    // Loop until the temperature is very close target
    #define TEMP_CONDITIONS (wants_to_cool ? thermalManager.isCoolingHotend(target_extruder) : thermalManager.isHeatingHotend(target_extruder))
  #endif //TEMP_RESIDENCY_TIME > 0

  float theTarget = -1.0, old_temp = 9999.0;
  bool wants_to_cool = false;
  wait_for_heatup = true;
  millis_t now, next_temp_ms = 0, next_cool_check_ms = 0;

  KEEPALIVE_STATE(NOT_BUSY);

  do {
    // Target temperature might be changed during the loop
    if (theTarget != thermalManager.degTargetHotend(target_extruder)) {
      wants_to_cool = thermalManager.isCoolingHotend(target_extruder);
      theTarget = thermalManager.degTargetHotend(target_extruder);

      // Exit if S<lower>, continue if S<higher>, R<lower>, or R<higher>
      if (no_wait_for_cooling && wants_to_cool) break;
    }

    now = millis();
    if (ELAPSED(now, next_temp_ms)) { //Print temp & remaining time every 1s while waiting
      next_temp_ms = now + 1000UL;
      #ifdef N_SERIES_PROTOCLE
          SERIAL_PROTOCOLPGM(" T:");
      //    SERIAL_PROTOCOL_F(thermalManager.degHotend(target_extruder), 1);
      //    SERIAL_PROTOCOLPGM(" E:");
      //    SERIAL_PROTOCOL((int)target_extruder);
      #else
        print_heaterstates();
      #endif
      #if TEMP_RESIDENCY_TIME > 0
        SERIAL_PROTOCOLPGM(" W:");
        if (residency_start_ms) {
          long rem = (((TEMP_RESIDENCY_TIME) * 1000UL) - (now - residency_start_ms)) / 1000UL;
          SERIAL_PROTOCOLLN(rem);
        }
        else {
          #ifdef N_SERIES_PROTOCLE
            SERIAL_PROTOCOL( "?" );
          #else
            SERIAL_PROTOCOLLNPGM("?");
          #endif
        }
      #else
        SERIAL_EOL;
      #endif
      /*#ifdef N_SERIES_PROTOCLE
        SERIAL_PROTOCOLPGM(" D:");
        SERIAL_PROTOCOL_F(thermalManager.degTargetHotend(target_extruder), 1);
        #if HAS_TEMP_BED
          SERIAL_PROTOCOLPGM(" B:");
          SERIAL_PROTOCOL_F(thermalManager.degBed(), 1);
          SERIAL_PROTOCOLPGM(" /");
          SERIAL_PROTOCOL_F(thermalManager.degTargetBed(), 1);
        #endif
        print_heaterstates();
        SERIAL_EOL;
      #endif*/
    }

    idle();
    refresh_cmd_timeout(); // to prevent stepper_inactive_time from running out

    float temp = thermalManager.degHotend(target_extruder);

    #if TEMP_RESIDENCY_TIME > 0

      float temp_diff = fabs(theTarget - temp);

      if (!residency_start_ms) {
        // Start the TEMP_RESIDENCY_TIME timer when we reach target temp for the first time.
        if (temp_diff < TEMP_WINDOW) residency_start_ms = now;
      }
      else if (temp_diff > TEMP_HYSTERESIS) {
        // Restart the timer whenever the temperature falls outside the hysteresis.
        residency_start_ms = now;
      }

    #endif //TEMP_RESIDENCY_TIME > 0

    // Prevent a wait-forever situation if R is misused i.e. M109 R0
    if (wants_to_cool) {
      // break after MIN_COOLING_SLOPE_TIME seconds
      // if the temperature did not drop at least MIN_COOLING_SLOPE_DEG
      if (!next_cool_check_ms || ELAPSED(now, next_cool_check_ms)) {
        if (old_temp - temp < MIN_COOLING_SLOPE_DEG) break;
        next_cool_check_ms = now + 1000UL * MIN_COOLING_SLOPE_TIME;
        old_temp = temp;
      }
    }

  } while (wait_for_heatup && TEMP_CONDITIONS);

  if (wait_for_heatup) LCD_MESSAGEPGM(MSG_HEATING_COMPLETE);

  KEEPALIVE_STATE(IN_HANDLER);
}
```

Firesped, 2016-12-30:
ok, turned on warnings debugging and I started seeing stuff.  Seems to be throwing warnings related to axisEnum

this isn't working

    static FORCE_INLINE void set_e_position_mm(const float &e) {
      set_position_mm((AxisEnum)E_AXIS
        #if ENABLED(DISTINCT_E_FACTORS)
          + active_extruder
        #endif
      , e);
    }

```
C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries G:\Users\Michael\Documents\Arduino\libraries -fqbn=arduino:avr:mega:cpu=atmega2560 -ide-version=10800 -build-path C:\Users\Michael\AppData\Local\Temp\arduino_build_656676 -warnings=default -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.arduinoOTA.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avrdude.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -verbose G:\Users\Michael\Documents\Arduino\Marlin-RC1.1.0 RC8 - N-Series\Marlin-RC\Marlin\Marlin.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries G:\Users\Michael\Documents\Arduino\libraries -fqbn=arduino:avr:mega:cpu=atmega2560 -ide-version=10800 -build-path C:\Users\Michael\AppData\Local\Temp\arduino_build_656676 -warnings=default -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.arduinoOTA.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avrdude.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -verbose G:\Users\Michael\Documents\Arduino\Marlin-RC1.1.0 RC8 - N-Series\Marlin-RC\Marlin\Marlin.ino
Using board 'mega' from platform in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr
Using core 'arduino' from platform in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr
Detecting libraries used...
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10800 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\mega" "C:\Users\Michael\AppData\Local\Temp\arduino_build_656676\sketch\Marlin.ino.cpp" -o "nul"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10800 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\mega" "C:\Users\Michael\AppData\Local\Temp\arduino_build_656676\sketch\M100_Free_Mem_Chk.cpp" -o "nul"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10800 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\mega" "C:\Users\Michael\AppData\Local\Temp\arduino_build_656676\sketch\MarlinSerial.cpp" -o "nul"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10800 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\mega" "C:\Users\Michael\AppData\Local\Temp\arduino_build_656676\sketch\Marlin_main.cpp" -o "nul"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10800 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\mega" "C:\Users\Michael\AppData\Local\Temp\arduino_build_656676\sketch\Sd2Card.cpp" -o "nul"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10800 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\mega" "C:\Users\Michael\AppData\Local\Temp\arduino_build_656676\sketch\SdBaseFile [... cut ...]
     (expr_list:REG_INC (reg:HI 111 [ ivtmp.126 ])

        (nil)))

C:\Users\Michael\AppData\Local\Temp\arduino_build_656676\sketch\temperature.cpp:797: confused by earlier errors, bailing out

lto-wrapper: C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-gcc returned 1 exit status

c:/program files (x86)/arduino/hardware/tools/avr/bin/../lib/gcc/avr/4.9.2/../../../../avr/bin/ld.exe: lto-wrapper failed

collect2.exe: error: ld returned 1 exit status

exit status 1
unable to find a register to spill in class 'POINTER_REGS'

```

Firesped, 2016-12-30:
moving to RCbugfix fixed the issues. closing.

closed 2016-12-30 by Firesped