MarlinFirmware/Marlin #5616: A few bugs to report - important!

playlet, 2016-12-28:
Hi!

I'm talking about the latest RCBugFix branch version available to download.

1. When not printing but the nozzle and bed is preheated, Prepare menu is blank. Only the first item ("go back to previous menu") is visible. Menu scrolls (visible on the first line) but nothing is accessible. - big problem.

2. Manual mesh bed leveling - impossible to do rendering all probeless printers useless on the latest version. I don't know if this was already fixed or not. - major problem for people who don't have a probe.

3. While printing, if you change steps/mm for extruder, value is stored but the menu doesn't close. It flashes once like it wants to exit and then goes back to editing steps/mm screen. After timeout, screen goes to normal again.

l-leukert, 2016-12-28:
Hi, can confirm No.1 in RCBugFix.
Prepare -> Preheat -> Prepare : only first Line 'Main' visible 
(Works in RC branch)

FHeilmann, 2016-12-28:
I second No1 as well.

Tannoo, 2016-12-29:
I found that you can get the prepare menu back if you set the nozzle to 0 degs from the control menu/temperature.

Tannoo, 2016-12-30:
I have found that if this line (1406) in `ultralcd.cpp`:

HOTEND_LOOP() if (thermalManager.target_temperature[e]) { has_heat = true; break; }

is commented out, the prepare menu works just fine while the hotend has a set point higher than 0.

Tannoo, 2016-12-30:
I had an unapproved fix for 2.
I don't know about 3.
I don't have that issue with 4.
I don't have that issue with 5.

playlet, 2016-12-30:
Thanks for the no. 1 fix, I can use my printer without pronterface again. Thank you, sir!!! :)

Tannoo, 2017-01-01:
I put in a PR. (#5632)

manianac, 2017-01-01:
Nothing against @Tannoo, but I didn't like the PR fix.  All that was needed was removing the break statement or having it on multi-hotends only.  See #5647

buzzkc, 2017-01-14:
I can confirm #5 also, having same exact issue.

lordjg, 2017-01-15:
I also confirm #5. In my case the lower corner is the front left and the higher is the front right

playlet, 2017-01-15:
I have noticed some more bugs that I'm also adding to the list in the first post.

6. When adjusting Motion - Z offset - value tends to avoid numbers such as: -0.30, -0, 40, -0,50 and so on. It always skips even number, example: skips from -0.29 to -0.31

7. Vertical lines on printed parts. I tried lowering jerk values and acceleration values and it helps but not all the way. If I try to print the same file on RC7 with higher jerk and accel values, there are no vertical lines on the part.

playlet, 2017-02-01:
Is anyone working on the problems number 4 & 5 from the first post?

It's annoying that you have to mess with babystepping to lower the nozzle after every print, while the next is just starting. It just gets higher and higher above the print bed after every print if you don't touch anything.

Also, number 5 is pretty bad with large objects. It almost always results in a warped piece.

Please fix this. It worked great in RC7. I'd go back to RC7 but I really like linear advance these days. :)

Roxy-3D, 2017-02-01:
ThinkyHead will be back very soon, and I suspect he will take care of this very quickly.

playlet, 2017-02-02:
Ok, thanks for the info :)

FATALHERMIT, 2017-02-12:
@playlet  for problem number 4, try disabling leveling fade height in configuration.h , since this feature was implemented to gradually reduce leveling correction until the movement is level to the machine's X and Y plane

dragonnn, 2017-02-15:
Is number 2 fixed? I tried Manual Bed Leveling on RCBugFix and it doesn't work, on RC8 it works fine.

Tannoo, 2017-02-15:
#5634 is the "fix" I'm using for MBL until @thinkyhead gets in a proper fix.

dragonnn, 2017-02-15:
Thanks, will try it today.

thinkyhead, 2017-02-15:
Number 7 is interesting. We have new jerk code (from the Prusa MK2 branch), but I'm not sure if this is the main cause. @playlet — Could you post some pictures of the vertical lines so we can see what features are producing them the most?

thinkyhead, 2017-02-15:
> @playlet for problem number 4, try disabling leveling fade height in configuration.h , since this feature was implemented to gradually reduce leveling correction until the movement is level to the machine's X and Y plane

Does this help? The feature should have no effect unless Marlin gets the `M420 Znn` command.

playlet, 2017-02-15:
Number 7 problem - you can disregard this. I didn't see that new code makes steppers start faster than before. I reduced my jerk value to half (15 to 7) and now it't fine. All good! :) Linear advance does the rest and all works great (k=43 for me on MK8 with32 uStep).

As for the fade height thing, I will test this in the next few days when I gaet my hands on the printer again.

Thanks for taking the time to look into this.

playlet, 2017-02-17:
Left to right and forward to back leveling seems a lot better now although not entirely consistent.

There's a bigger problem here that troubles me, and that is number 4 - after each print z axis is higher and higher above the bed. And when I leave a printer off for an hour or so and start printing again, now it is too low (scratching the bed) and then gets higher and higher again.

There is not even 1 print in a row that can be done without messing with z babystepping.

Roxy-3D, 2017-02-17:
> There's a bigger problem here that troubles me, and that is number 4 - after each print z axis is higher and higher above the bed. And when I leave a printer off for an hour or so and start printing again, now it is too low (scratching the bed) and then gets higher and higher again.

Pretty much, every print does a G28 and the position gets reset with a G28.   If each print goes higher and higher above the bed, there are very limited number of things that could do that.   It might make sense to make a special debug version of the code that prints out all of the constant type parameters and see which one is changing.   For example, print out these types of things (if they are compiled into the code):
- pre_home_z
- bilinear_z_offset
- mbl.z_offset
- zprobe_zoffset

Something is changing...   If we knew what was changing it probably would not be that hard to figure out the root cause.

playlet, 2017-02-18:
I'm willing to test everything if someone makes the changes :)

playlet, 2017-02-24:
I've been testing the repeatability in relation to the probe speed (z homing speed) and bed heating and have discovered a strange thing.

If the bed is cold, repeatability is in the range of 0.003-0.009. Also, if probed faster (slow speed/2), it gives less error (than let's say slow_speed/5)!

If the bed is heated, repeatability is in the range of 0.080-0.135. That is almost a whole 1 layer height. The funny thing here is that when the probe is measuring and the heater is on, error increases in relation to when it's off (in the on/off heat the bed loop).

So, I guess there might be no problem with marlin but with the probe sensing power through the bed and throwing the leveling grid off. This is really annoying. I use capacitive LJC18A3-B-Z/BX NPN NO probe.

Is there a way I can make probes more accurate?

Blue-Marlin, 2017-02-24:
Could be a problem with VCC for the probe.
When the heaters are on the voltage could drop significant when wires are too long, too thin or the power supply is too weak.
A different temperature of the probe could be an other reason, but usually is not.

Roxy-3D, 2017-02-24:
It sounds like you need to power the probe off of a clean source.  You might be able to put a LM7805 on the +12 volts used to power the bed.  And that might be able to give you a constant +5v even if the +12 v is pulsing due to the bed load.

lrpirlet, 2017-02-24:
@playlet.. Well, I am surprised that someone tries to tram with a capacitive probe while heating the bed...

Physic tells me that heat is a good way of MODIFYING the capacitance (the dielectric becomes less insulating at higher temperature). By heating the bed you heat the air and that CHANGES the permittivity (note, beside heat, the humidity, the atmospheric pressure and maybe some other environment factor will have a direct incidence on the repeatability)... this is why I, **PERSONALLY**, do NOT believe that a capacitive probe is better than a good old switch taken from a broken mouse :-)

So, as you have observed, make sure to probe a bed at a very stable temperature (I would strongly suggest to probe at ambient temperature...)

Blue-Marlin, 2017-02-24:
LJC18A3-B-Z/BX NPN NO usually needs at least 6V. 
The idea with an extra regulator is not bad, but use a higher voltage type.

playlet, 2017-02-24:
@Blue-Marlin Might be VCC, I'll test this too. I connected the probe to the rightmost connector on the ramps 1.4 12V input.

@Roxy-3D I can't use 5V as the probe is not accurate enough on such low voltage.

@lrpirlet Cold bed is not good for probing, in my opinion, as the bed warps slightly when fully heated. That's why I went with the method described below. :)

My workaround for hassle free printing is this:

In marlin, setup 4*60 z speed and 9000 probe speed, homing and probe slow_speed/2 for both. This topic (https://github.com/MarlinFirmware/Marlin/issues/5868) claims to have better probing accuracy on faster speed and I have confirmed that!

Procedure:

preheat the bed (me: abs 98°C, with annyer extra strong blue hair spray - part sticks even too much!),
turn off bed heating,
g28,
g29,
preheat bed again,
print.

This makes repeatability results in the range of 0.015 which is good enough in my book. Fast probe and homing speed is important here because bed looses heat fast. Results are very good indeed.

Thanks for your help

playlet, 2017-02-24:
I just tested the probe voltage. It is 12.16V when bed is off and 12.07V when bed is heating. Is this too much for a 30A 12V LED power supply or did I connect the probe to a wrong connector (rightmost 12V power input on ramps 1.4)?

Blue-Marlin, 2017-02-24:
0.06 V difference seems to be ok.

playlet, 2017-02-26:
More debugging done today. Today I reverted back to RC7 and tested the z offset problem. I calibrated z offset, made it perfect and started printing the cube over and over again with probe measuring at the same time bed is being heated!!

I take back what I sad about marlin not being a culprit here - IT IS! Something is wrong with rcbugfix code what makes z offset reduce after each print leaving the nozzle in next higher and higher above the bed.
RC7 does this perfectly without any problem. You can print 100 cubes one after another and z offset would always be perfect.

Current Marlin has a bug somewhere. Pleeeeeeease fix this. For a month now, I've been waiting for this to be fixed but noone is looking into it, looks like. I see less important stuff being fixed weekly and this crucial feature is not. Please fix this... :)

Roxy-3D, 2017-02-27:
> @Roxy-3D I love your double click the encoder to enter babystepping shortcut :)

Yes!   I agree it is a very nice feature.   And only 15 lines of code.

playlet, 2017-03-06:
I'm so happy that this is now included in standard rcbugfix :)

@Roxy-3D Would you mind moving this option higher in the configuration_adv. This is a very useful feature and it's on the bottom of the file where people don't usually look, if they update often as rcbugfix does :)

Tannoo, 2017-03-06:
I think there should be a list options available in the configuration files that are available in the given files.
Simular to the G, M, and T codes listed in `marlin_main.cpp`.

Tannoo, 2017-03-06:
I might compile that list and submit a PR.

playlet, 2017-03-08:
Reporting back some observations.

Bed leveling works fine if printing inside probed area. If outside, it's not that good.
How does marlin's code determine z offset outside of probed area?

Thanks :)

playlet, 2017-08-05:
I'm still experiencing number 4 in the latest version (tested on August 2nd). After every print, in the beginning of the next, I need to lower the nozzle about 0.125 mm more. If I reset power and then print again, no need to do that. Please look into this. Ramps 1.4 board on prusa i3 diy clone.

closed 2018-02-06 by thinkyhead