MarlinFirmware/Marlin #5617: [QUESTION] Using 2nd extruder stepper for another purpose?

infernusdomoso, 2016-12-28:
I have custom modified 1.1.0 RC2 quite heavily - I have installed a Z probe on a sled, that can auto-level itself against a known point, so that when I change tool heads, I don't need to manually re-level anything - the sled docks, and a stepper on the sled raises and lowers the z probe to calibrate it. I did this by hacking the heck out of the code. I added a 2nd extruder, but did a bunch of overrides that allowed the 2nd extruder to act like a Z axis. Obviously, this wasn't very clean, code-wise.

I'd like to upgrade to RC8, to use the bilinear mesh bed leveling, as well as some of the other newer features. Going over the code and trying to manually patch the files has been daunting, but as I've been looking I'm seeing a ton of new config options in the files, such as E_MANUAL and the fact that EXTRUDERS and HOTENDS have been split apart to take into account things like 1 stepper for multiple extruders etc.

Is there a way in the current RC8 version that would allow me to more easily use the extra unused stepper driver for another purpose - in other words - to raise and lower a Z probe? I could convert it to a servo, but I prefer the much finer grained control I can get by using a stepper and a fine threaded rod. I'd rather not have to patch the heck out of RC8 and hack it to work. (My current running code does things like setting the active_extruder=1, then basically issuing forward and back commands to the extruder, then setting it back to active_extruder=0. This required all sorts of extra code such as ignoring cold extrustion checks, making the Z endstop trigger when the extruder was moving, etc).

Any guidance would be helpful. Also, I didn't see a forum that was best suited for this question, so my apologies for opening a question here if there was a better place to ask this.

infernusdomoso, 2016-12-28:
I edited the title - basically, the simplest way to word the question is I'd like to add another axis, and use the E1 stepper driver to drive it. All code for controlling it, I'd be adding, but is there any support for that extra axis? Or is faking a second extruder the more simple way to do it, as that's already supported? (Adding another axis would be a ton of code it seems - hacking a second extruder seems like less work?)

Roxy-3D, 2016-12-28:
There has been some interest in having extra stepper motors on a given axis for this exact purpose.   There has also been some discussion about algorithms to level an axis.   I think it would be a worth while venture to allow extra steppers to be assigned on a per axis basis and to have the code be smart enough to use them to level what can be handled.

infernusdomoso, 2016-12-28:
I'm actually using it as an entirely separate axis. The process is as follows: The head grabs the carriage, and moves to an area to the far left and behind the print bed - the head is not over the bed at all. 

Then the axis homes Z, with the tip of the print head touching a limit switch mounted to a stationary plate thats mounted to the bed carriage (using a TAZ5 for this) - note it's mounted to the plate the bed is on - bed height and alignment has no bearing on this. Now that we know where Z is at this arbitrary zero, it moves a bit so the z probe is over another area of this plate. The extra axis then rotates a threaded rod, lowering the probe until it contacts the base plate.

At this point, the Z probe tip is now exactly 2.6mm higher than the print head. (The 2.6mm is the measured difference between the limit switch the tip hits, and the plate the z probe adjusts to).

I can then run a normal probe level by grabbing the sled and leveling.

The whole purpose of this is twofold:
1) I never have to manually level a darn thing
and 2) If I change print heads or nozzles (which I do, switching sometimes between a normal E3D nozzle and a Volcano, and when switching nozzles they aren't always 100% the same height from the bed) I simply run a custom command and the tip of the nozzle is always 2.6mm from the Z probe.

Obviously this took a ton of custom coding to do. Basically just trying to see if in the newer versions of the software changes were made that allow easier control over the unused stepper/E1 driver without having to hack it into thinking it's NOT an actual extruder. (I ran into issues with cold extrusion prevention, firmware retraction was being applied to it, had to write custom code to "home" the E axis, etc)

jerryerry, 2017-01-06:
I'm also trying to do this. 
I want to use both extruders on ramps as steppers to drive lead screws. The only change I need is to have some gcode that can use endstops to sense it. 

Any ideas on how to detect endstops when extruders are being used? 
Only idea I have is to rather use dc motors and detect endstop using M226

jerryerry, 2017-01-06:
I have an additional ramps&mega board - would I be able to use i2c to run the extra steppers on the other ramps board?

jerryerry, 2017-01-06:
Would actually work if M600 was modified to not pause the print?

jerryerry, 2017-01-13:
couldn't figure out how to make m600 run immediately as the switch as triggered. I just rewrote over g12 (since i had no use for it) to do an external method:

  while ((digitalRead(3) == HIGH))      //using pin 3 for endstop
  {
    destination[E_AXIS] += 0.001;
    prepare_move_to_destination();         // set_current_to_destination
    stepper.synchronize();
    refresh_cmd_timeout();
  }

it works but it isn't smooth

RobWks, 2017-02-14:
Hey everyone, I'm also looking at using both extruder axis to drive lead screws, would be even better if I could use endstops with them. What modifications did you make to control them simply as auxiliary axis without the limitations of marlin thinking it's an extruder axis?

thinkyhead, 2017-02-14:
> I have an additional ramps&mega board - would I be able to use i2c to run the extra steppers on the other ramps board?

I have implemented this and it works well.

thinkyhead, 2017-02-14:
> I'm also looking at using both extruder axes to drive lead screws

Could be quite a project. Marlin currently always assumes 3 movement axes and 1 extruder axis. You basically would need to drop the E axes and implement 2 more movement axes. It's not a job for the faint of heart.

closed 2017-10-31 by thinkyhead